moveit/moveit #3260: bug about getCurrentPose in  moveit::planning_interface::MoveGroupInterface

dbdxnuliba, 2022-11-01:
### Description

Overview of your issue here.

### Your environment
* ROS Distro:[Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? source
* If binary, which release version?
* If source, which branch? **moveit 1.1.10**
*
when the dual arm  robot keep static ,
I  use   group.getCurrentPose("left_arm_link6")  to get left_arm_link6 pose reltv world frame,
but I got different value between twice, 
[image]
and  when I check the left_arm_link6 tf value reltv world frame  in rviz ,it was shown in the following png
[image]

we can see that the second time value is same with the tf value in rviz,but the value got by the first time is fully wrong ,
this is the code 
[image]

this is the tf tree
[image]

this is the model in rviz
[image]

Question:
1. is this a bug caused by moveit or something
2. how can I fix the issue 
Thanks

simonschmeisser, 2022-11-01:
It seems the joint positions have not yet arrived at the Move Group Interface. Therefore they are all at the default value of 0.

You can find the relevant code here: https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp 

you could try placing a `cam_arm_group.startStateMonitor();` before your sleep

dbdxnuliba, 2022-11-02:
@simonschmeisser ,your solution is nice ,
this is the output after  placing a cam_arm_group.startStateMonitor(); before  sleep
[image]

> It seems the joint positions have not yet arrived at the Move Group Interface. Therefore they are all at the default value of 0.
>
> You can find the relevant code here: [...]

simonschmeisser, 2022-11-02:
I don't know, maybe @v4hn knows why `startStateMonitor()` is not called in the constructor?

simonschmeisser, 2022-11-02:
You can also open a pull request (PR) and we can see if somebody objects

v4hn, 2022-11-02:
> but why don't you set move_group.startStateMonitor() is called by default when creating group in moveit source code.

Because doing so means your node becomes a subscriber to the /joint_states topic (with the related overhead) when it is not necessary in the most common use case of the MoveGroupInterface (`setGoal(...)`, `move()`). That's enough justification for me, but other maintainers might disagree.

I would propose to add a warning to the `getCurrentState` method when the monitor was not started yet and tell users to call `startStateMonitor` after construction. Also I would like to merge a patch that makes the related `getCurrent*` methods *throw or return false* when there is no complete state yet after waiting. Returning invalid values is plain broken. It would be great if you could take the time to do so @dbdxnuliba !

That being said I'm somewhat surprised that one second is not enough on your system to subscribe to the `/joint_state` topic and get messages. Either your state update rate is awfully slow or your system takes too much time setting up the connection for some reason.

dbdxnuliba, 2022-11-03:
@v4hn , 

The /joint_state topic was published by  three nodes:
1 ./tor_lift_node  publish /joint_states  only contains torso_lift_joint  with 30hz
[image]

2.left_arm_node publish  /joint_states only contains left_arm_joint1 ~left_arm_joint6 with 40hz 

3.right_arm_node publish  /joint_states only contains right_arm_joint1 ~right_arm_joint6 with 40hz

and I'm looking foward the patch you mentoined ,is it convinet for you please  to notify me ,when you accompish the patch