moveit/moveit #3265: "ABORTED: Solution found but controller failed during execution" massage.

chaimash, 2022-11-10:
### Description
"ABORTED: Solution found but controller failed during execution" massage while planning path for Baxter the robot
For instance, the arm going up, and deny to go back to the first point
Overview of your issue here.
### Your environment
* ROS Distro: [Noetic Ninjemys]
* OS Version: 20.04 (Focal)
* Source or Binary build?
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2022-11-10:
This happens if the robot controller cannot track the desired trajectory with the given precision. Try to tune your controller parameters or relax tracking precision.

ans0ld2, 2022-11-10:
> This happens if the robot controller cannot track the desired trajectory with the given precision. Try to tune your controller parameters or relax tracking precision.

I tried to reduce the velocity and precision, but it did not help