moveit/moveit #3267: how can we change end_effectors of arm dynamiclly  in  moveit ROS1, is there method  or services,thanks a lot

dbdxnuliba, 2022-11-15:
how can we change end_effectors of arm dynamiclly  in  moveit ROS1, is there method  or services,thanks a lot

rhaschke, 2022-11-15:
If you just want to switch between different end-effector frames defined for your robot, you can call the methods
`setEndEffectorLink()` or `setEndEffector()` of `MoveGroupInterface`.
However, this will not change the actual tool mounted to an arm. This is not easily possible as this would require changing the robot's URDF on the fly, which is not supported throughout ROS1.

dbdxnuliba, 2022-11-16:
can we do it in the following way :
`    def attach_end_effector(
        self,
        name: str,
        filename: str = "/dev_ws/src/trajectory_tools/end_effector/ur_ee.stl",
    ) -> bool:

        pose = PoseStamped()
        pose.header = Header()
        pose.header.frame_id = self.move_group.get_end_effector_link()
        pose.pose = Pose(position=Point(0, 0, 0), orientation=Quaternion(0, 0, 0, 1))

        ac = AttachedCollisionObject()
        ac.object = make_mesh(name, pose, filename=filename)

        ac.link_name = self.move_group.get_end_effector_link()
        ac.object.subframe_names = ["tcp"]
        ori = tf.transformations.quaternion_from_euler(0, pi / 2, 0)
        ac.object.subframe_poses = [
            Pose(
                position=Point(0.184935, 0.0, 0.06),
                orientation=Quaternion(ori[0], ori[1], ori[2], ori[3]),
            )
        ]

        self.scene.attach_object(ac)

        return self.wait_for_state_update(name, object_is_attached=True)

from https://github.com/vinceHuyghe/trajectory_tools/blob/dev/src/trajectory_tools/trajectory_handler.py#:~:text=def%20attach_end_effector(,%3DTrue)
`

rhaschke, 2022-11-16:
Sure, if your end effector is a rigid tool, you can just add it as an attached collision object.
However, most end effector are actuated, e.g. involving finger joints. Changing these would require modifying the URDF, which is not possible.

dbdxnuliba, 2022-11-21:
> Sure, if your end effector is a rigid tool, you can just add it as an attached collision object.
> However, most end effector are actuated, e.g. involving finger joints. Changing these would require modifying the URDF, which is not possible. 

OK，I see, but for now,attach the tools can meet my jobs

rhaschke, 2022-11-21:
If this resolves your issue, please close it.

dbdxnuliba, 2022-11-24:
ok

rhaschke, 2022-11-24:
> how can I close the issue, I can't find the option to close it

Just pushing the corresponding button right below the comment field:
[image]

closed 2022-11-24 by rhaschke

dbdxnuliba, 2022-11-27:
[image]
It seems that there is no such buttion

rhaschke, 2022-11-27:
Sure, because I closed it already!

dbdxnuliba, 2022-11-28:
OK