moveit/moveit #3268: collision_detect is   not valid  in moveit  when jog robot  with teachpad tool

dbdxnuliba, 2022-11-15:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?  source moveit [1.1.10]

when I jog in z- direction the robot_arm , it collision detection is not valid when the collision is happened ,like the gif:
[image]

but when I drag the interactive marker in rviz ,the collision detection is valid ,like that:
[image]

Question:
how can I set  moveit collison to be valid when I move the robot with the api provided by  robot_driver  which could change the robot jnt states in rviz , it we can do that, it's useful to keep the safe  of the robot,
Any helps will be pleasured!

rhaschke, 2022-11-15:
Your robot_driver doesn't use MoveIt (doesn't even know about it). Thus, if you want collision checking, you need to go through MoveIt and replace your robot's teach pad with something using MoveIt.

dbdxnuliba, 2022-11-16:
can we open a thread or node in some way using moveit to open collision detect  with the input by listening to /joint_states topic

dbdxnuliba, 2022-11-27:
![col1](https://user-images.githubusercontent.com/13350158/204117424-127d7b73-4c25-4618-94ef-6bda0034e541.gif
I open a ros_node to decect collision ,here is the code : 
`
static const std::string PLANNING_GROUP = "arms_and_torso";

class CollisionDectMonitor
{
public:
    CollisionDectMonitor() = default;
    ~CollisionDectMonitor() = default;

public:
    void start()
    {
        // boost::function0< void> f =  boost::bind(&CollisionDectMonitor::colsionMonitorFun,this);
        // boost::thread thrd( f );
        boost::thread thrd( boost::bind(&CollisionDectMonitor::colsionMonitorFun,this) );    
        thrd.join();
    }

private:
    void jointStateCallback(const sensor_msgs::JointStateConstPtr &msg_in)
    {
        
        jnt_num_ = msg_in->position.size();
        jnt_position_.resize(jnt_num_);

        mutex.lock();
        // std::cout<<"jnt_names:"<<std::endl;
        // for (int i = 0; i < jnt_num_; i++)
        // {
        //     jnt_position_[i] = msg_in->position[i];
        //     std::cout<<msg_in->name[i]<<"\t";
        // }
        // std::cout<<""<<std::endl;
        mutex.unlock();
        has_got_jnt_pos_ = true;
    }

    void colsionMonitorFun()
    {
        ros::NodeHandle n;
        ros::Rate r(20);
        ros::Subscriber sub_position = n.subscribe("/joint_states", 10, &CollisionDectMonitor::jointStateCallback,this); // buffer for data to callback,
        planning_scene_monitor::PlanningSceneMonitorPtr planning_scene_monitor_;
        planning_scene_monitor_.reset(new planning_scene_monitor::PlanningSceneMonitor("robot_description"));
        if (planning_scene_monitor_->getPlanningScene())
        {
            //   planning_scene_monitor_->startPublishingPlanningScene(planning_scene_monitor::PlanningSceneMonitor::UPDATE_SCENE,
            //                                                         "grasping_planning_scene");
            //   planning_scene_monitor_->getPlanningScene()->setName("grasping_planning_scene");
        }
        else
        {
            ROS_ERROR_STREAM_NAMED("test", "Planning scene not configured");
        }

        const robot_model::RobotModelConstPtr robot_model = planning_scene_monitor_->getRobotModel();

        planning_scene::PlanningScenePtr ps(new planning_scene::PlanningScene(robot_model));
        while (n.ok())
        {
            robot_state::RobotState current_state = ps->getCurrentState();
            if(!has_got_jnt_pos_)
            {
                r.sleep();
                continue;
            }

            mutex.lock();
            current_state.setVariablePositions(jnt_position_);
            mutex.unlock();

            ROS_ERROR("002\n");

            // const double * postion = current_state.getJointPositions("shoulder_pan_joint");
            if (ps->isStateColliding(current_state, PLANNING_GROUP))
            {
                //   ROS_ERROR_STREAM('self collision occur ' << ps->isStateValid(current_state, PLANNING_GROUP));
                // ROS_ERROR_STREAM('collision occur ' << ps->isStateValid(current_state, PLANNING_GROUP));

                std::string info = "collision occur" + ps->isStateValid(current_state, PLANNING_GROUP);
                ROS_ERROR(info.c_str());
                std::cout << "collision jnt postion deg is:" << std::endl;
                  for (int i = 0; i < jnt_num_; i++)
                  {
                    std::cout << current_state.getJointPositions("left_arm_joint1")[i] / M_PI * 180.0 << " ";
                  }
                std::cout << " " << std::endl;
            }
            else
            {
                ROS_INFO_STREAM("state ok");
            }
            r.sleep();
        }
    }

private:
    bool has_got_jnt_pos_{false};
    std::vector<double> jnt_position_;
    boost::mutex mutex;
    int jnt_num_{0};
};

int main(int argc, char **argv)
{
    ros::init(argc, argv, "collision_detect_test");
    ros::NodeHandle nh;

    ros::AsyncSpinner spin(2);
    spin.start();

    CollisionDectMonitor monitor;

    monitor.start();

    // wait_cmd_finish();
    ros::waitForShutdown();

    return 0;
}`

simonschmeisser, 2022-11-27:
Please use tripple backticks on a separate line to format source code

you can either have the `PlanningSceneMonitor` update the `RobotState` (and `PlanningScene`). Have a look for `LockedPlanningSceneRO`

or you can update the `RobotState` yourself (like you commented out in the callback function) but note that you should still check collisions with a `PlanningScene` updated by `PlanningSceneMonitor` if you want to know about collisions with objects not defined via URDF

dbdxnuliba, 2022-11-28:
> 

I have try to use LockedPlanningSceneRO
,but we get wrong result about collision check:

"'
//
// Created by ubuntu on 22-8-25.
//
#include <ros/ros.h>
#include <rm_msgs/MoveJ_P.h>
#include <rm_msgs/Plan_State.h>
#include <std_msgs/Empty.h>
#include <rm_msgs/Force_Position_Move_Pose.h>
#include <rm_msgs/MoveL.h>
#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit_msgs/AttachedCollisionObject.h>
#include <moveit_msgs/CollisionObject.h>
#include <rm_msgs/Arm_Current_State.h>
#include <rm_msgs/GetArmState_Command.h>
#include <rm_msgs/Force_Position_State.h>
#include <std_msgs/Bool.h>
#include <rm_msgs/Current_Work_Frame.h>
#include <tf2_ros/transform_listener.h>

#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>

#include <moveit_msgs/DisplayRobotState.h>
#include <moveit_msgs/DisplayTrajectory.h>

#include <moveit_msgs/AttachedCollisionObject.h>
#include <moveit_msgs/CollisionObject.h>

#include <moveit/robot_trajectory/robot_trajectory.h>
#include <moveit/trajectory_processing/iterative_time_parameterization.h>
#include <moveit_msgs/OrientationConstraint.h>
#include <moveit_visual_tools/moveit_visual_tools.h>
#include <moveit_visual_tools/moveit_visual_tools.h>
#include <chrono>
#include <string>
#include <vector>
#include <algorithm>

// MoveIt
#include <moveit/robot_model_loader/robot_model_loader.h>
#include <moveit/planning_scene/planning_scene.h>

#include <moveit/kinematic_constraints/utils.h>

static const std::string PLANNING_GROUP = "arms_and_torso";
const std::string ROBOT_DESCRIPTION = "robot_description";

class CollisionDectMonitor
{
public:
    CollisionDectMonitor() = default;
    ~CollisionDectMonitor() = default;

public:
    void start()
    {
        boost::thread thrd(boost::bind(&CollisionDectMonitor::colsionMonitorFun, this));
        thrd.join();
    }

private:
    void jointStateCallback(const sensor_msgs::JointStateConstPtr &msg_in)
    {
[... 86 lines cut ...]
    bool has_got_jnt_pos_{false};
    std::vector<double> jnt_position_;
    boost::mutex mutex;
    int jnt_num_{0};
};

int main(int argc, char **argv)
{
    ros::init(argc, argv, "collision_detect_test");
    ros::NodeHandle nh;

    ros::AsyncSpinner spin(2);
    spin.start();

    CollisionDectMonitor monitor;

    monitor.start();

    // wait_cmd_finish();
    ros::waitForShutdown();

    return 0;
}
 
"'