moveit/moveit #3272: Robotic Arm smooth motion while object tracking

Akumar201, 2022-11-19:
I would like to request for a potential solution for my current problem. Currently I am tracking an aruco marker and moving the arm to that point, but the motion is not continuous (it's kind of jerky), is there methods that can be used to make the motion smoother. I am using MoveIt python API. Ultimate goal is to track object and later pick them from moving conveyor belts without using encoders.

rhaschke, 2022-11-19:
Did you have a look at [MoveIt servo](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html)?

Akumar201, 2022-11-21:
@rhaschke  Yes, thank you for the suggestion, As I am new to MoveIt, I have two followup questions 
1. Is it available in python API?
2. I want to use it with real robot is there any examples which I can see and learn from. I have already learning from official MoveIt website - https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html

rhaschke, 2022-11-21:
@AndyZe, can you please answer these questions regarding moveit_servo?

AndyZe, 2022-11-21:
My quick response is, there's no Python API to move to a pose smoothly with MoveIt Servo.

What you can do is publish a TwistStamped message to the servo node, which tells the robot which direction to move. You can look at this file for a simple example of how to do that. This is a C++ file but it should be very simple to translate to Python.

https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/src/teleop_examples/spacenav_to_twist.cpp

AndyZe, 2022-11-21:
So with a little bit of logic, you can achieve what you want with Python. And feel free to open a pull request if it works!

closed 2022-11-21 by AndyZe