moveit/moveit #3273: How to implement a python-based motion planning algorithm in moveit?

Evansika959, 2022-11-21:
Moveit has many motion planning library, but to my knowledge they're all implemented by cpp. We wonder if there is any python-based motion-planning library(like OMPL in py version) so that we could implement our python-based motion planning algorithm in moveit? 

Thank you in advance!!

rhaschke, 2022-11-21:
To my knowledge, there is no way to implement a scheduling plugin in Python (yet). Since the planning is usually very time-consuming, this is also not recommended.

Evansika959, 2022-11-22:
> To my knowledge, there is no way to implement a scheduling plugin in Python (yet). Since the planning is usually very time-consuming, this is also not recommended.

thanks! cpp is indeed more suitble for planning

gavanderhoorn, 2022-11-22:
While I agree with @rhaschke about the potential problems with performance, I've always been curious whether [embedding a Python interpreter](https://docs.python.org/3/extending/embedding.html) (perhaps using something like [reefactor/cppy3](https://github.com/reefactor/cppy3)) could not be used host an interpreter in a planning plugin.

Then it'd be a matter of exposing the right amount of state (from MoveIt to the Python3 interpreter) and writing the actual business logic (ie: planning algorithm).

rhaschke, 2022-11-22:
You are right, using pybind11 as in #2910 and exposing some more state, this could become possible.

gavanderhoorn, 2022-11-22:
Ah, `pybind11` can also embed. I forgot.

Evansika959, 2022-11-23:
Thank you for sharing!!

Evansika959, 2022-11-23:
I wonder if there're actual projects done by implementing python planning program in this manner? 
Or some simple examples of embedding py planning programs in moveit could also be a great help!
Thanks! 

> While I agree with @rhaschke about the potential problems with performance, I've always been curious whether [embedding a Python interpreter](https://docs.python.org/3/extending/embedding.html) (perhaps using something like [reefactor/cppy3](https://github.com/reefactor/cppy3), or even the well-known [boost.python](https://www.boost.org/doc/libs/1_80_0/libs/python/doc/html/index.html)) could not be used host an interpreter in a planning plugin.
> 
> Then it'd be a matter of exposing the right amount of state (from MoveIt to the Python3 interpreter) and writing the actual business logic (ie: planning algorithm).

rhaschke, 2022-11-23:
I'm not aware of any python-based planner for MoveIt.