moveit/moveit #3278: Request for MOVEit Servoing Implementation

Akumar201, 2022-11-22:
**Is your feature request related to a problem? Please describe.**
I want to use moveit servoing on [XArm](https://github.com/xArm-Developer/xarm_ros#6-mode-change), however there are no tutorials that can help me to do so. 

**Describe the solution you'd like**
Please provide me a tutorial or any resource link that can be useful to move my arm using MoveIt Servoing. 

**Describe alternatives you've considered**
As far as my Understanding from the tutorial [MoveIt Servoing](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html) I am right now following these steps:

1. Launching gazebo file of my robotic arm
2. Launching the moveit_rviz file of my robot. 
3. Switch to a compatible type of ros-control controller.
4. Then launching roslaunch moveit_servo spacenav_cpp.launch
However, I am getting few error that needs to be resolved before moving forward. Please let me know if I am in the right direction.

I am getting the following error 

`[ERROR] [1669150346.098234549]: Group 'manipulator' not found in model 'xarm6'
`
As far as I understand with this is in moveit servoing package there is a code that is looking for move group manipulator however, they are not able to find it as while creating moviet file XArm people might have name is differently.

There are two ways to correct it. 
1. Rebuild moveit pakage with same name for move group
2. Change the name in the Moveit servoing pakage -> can you please direct me to the file which I should edit, I don't want to mess with the package.

Another thing please correct me if I am wrong I need not to change anything in your [Movie servoing](https://github.com/ros-planning/moveit/tree/master/moveit_ros/moveit_servo) I need to use launch files for running my robot. Please let me know if there are changes that needs to be done in order to use it. Looking forward to hearing back

AndyZe, 2022-11-22:
This is the line you need to update for you Group:  https://github.com/ros-planning/moveit/blob/9cc7e8fb0d5b9ceb09d5ba68b524e7a4ab7ca02f/moveit_ros/moveit_servo/config/ur_simulated_config.yaml#L31

To avoid modifying the MoveIt package, create your own copy of [this launch file](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/launch/spacenav_cpp.launch), and your own copy of `ur_simulated_config.yaml`. Obviously you probably want to rename those files.

If you search through the existing MoveIt issues, you can probably find somebody else using an Xarm and if you're lucky, you could ask them for their config/launch files.

Akumar201, 2022-11-22:
Hi, can you please tell me the steps I am taking in order to work with XArm ( or any other arm) are in right direction. I will rewrite them here for your easy access. 

As far as my Understanding from the tutorial [MoveIt Servoing](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html) I am right now following these steps:

    Launching gazebo file of my robotic arm
    Launching the moveit_rviz file of my robot.
    Switch to a compatible type of ros-control controller.
    Then launching roslaunch moveit_servo spacenav_cpp.launch
    However, I am getting few error that needs to be resolved before moving forward. Please let me know if I am in the right direction.

AndyZe, 2022-11-22:
That sounds mostly correct but you  need to modify `spacenav_cpp.launch` to use a new yaml file for your robot.

Akumar201, 2022-11-22:
Hi, thank you for your prompt reply, I am new to ROS, I am sorry for asking basic questions. So, do I have to create a new yaml file from scratch or Moveit assistant does it? or I can edit the existing file accordingly. Once finished with the project I would like to contribute by creating tutorial for anyone who wants to use it with their robot.

AndyZe, 2022-11-22:
I would save a copy of this one then modify it:  https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/config/ur_simulated_config.yaml

Akumar201, 2023-02-08:
For future users , please review the https://github.com/ros-planning/moveit/issues/3340 , it works with little bit of tuning in config file. 

I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2023-02-08 by Akumar201