moveit/moveit #3283: Could not open the space navigator device. Did you remember to run spacenavd (as root)?

Akumar201, 2022-11-28:
## Background

I am following the (moveit servoing tutorial)[https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html], however I am getting the following error related to spacenavd

```
[ERROR] [1669672889.199512421]: Could not open the space navigator device. Did you remember to run spacenavd (as root)?
[spacenav_node_akumar_ThinkPad_P17_Gen_2i_29034_5978491666214438105-1] process has died [pid 29049, exit code 1, cmd /home/akumar/ws_moveit/devel/lib/spacenav_node/spacenav_node __name:=spacenav_node_akumar_ThinkPad_P17_Gen_2i_29034_5978491666214438105 __log:=/home/akumar/.ros/log/7bcc805e-6f61-11ed-8cc6-fb1b3c97baa8/spacenav_node_akumar_ThinkPad_P17_Gen_2i_29034_5978491666214438105-1.log].
log file: /home/abc/.ros/log/7bcc805e-6f61-11ed-8cc6-fb1b3c97baa8/spacenav_node_abc_ThinkPad_P17_Gen_2i_29034_5978491666214438105-1*.log

```
## Instructions
I did git clone [UR Repo](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver). 
Run the following command 

1. roslaunch ur5_bringup.launch
2. roslaunch moveit_planning_execution.launch sim:=true
3. roslaunch move_group.launch

However, after the roslaunch move_group.launch I am getting the above error.
I tried to fix the the code using the following site http://wiki.ros.org/spacenav_node/Troubleshooting

I am using Ubuntu 20.04 and ROS Noetic

TemugeB, 2023-03-16:
For those trying the ```moveit_servo``` demo and running into this issue, simply don't launch the ```spacenav_cpp.launch```

Instead, bring up the servo server with: 
```rosrun moveit_servo servo_server```

Then publish move commands directly to topic with command:
```
rostopic pub -r 100 -s /servo_server/delta_twist_cmds geometry_msgs/TwistStamped "header: auto
twist:
  linear:
    x: 0.0
    y: 0.01
    z: -0.01
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```

Strangely though when I stop publishing on this topic, the robot keeps moving in the simulation. So be careful trying this on a real robot.

AndyZe, 2023-03-16:
> Strangely though when I stop publishing on this topic, the robot keeps moving in the simulation. So be careful trying this on a real robot.

The `incoming_command_timeout` parameter should prevent that. If it's not working, can you create a separate issue for that?

https://github.com/ros-planning/moveit/blob/9ee989d67867f8a5cc91907cb2733dc88bf8be51/moveit_ros/moveit_servo/config/ur_simulated_config.yaml#L39

TemugeB, 2023-03-17:
I tried running the ```moveit_servo``` demo again and the issue is still present. I made a video of the issue on [youtube](https://youtu.be/im6PkPwvy1k).

Following Moveit Noetic documentation ([link](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html)), I'm using the ```1.2.7``` branch for ```universal_robot```. Because of https://github.com/ros-industrial/universal_robot/issues/627, I'm not using the latest package. Should I create a separate issue?

AndyZe, 2023-03-17:
Thanks for the video! Can you please create a new Issue for that, since it's not related to this Issue?

devvaibhav455, 2023-06-13:
> 

Hi,

When I run ` rosrun moveit_servo servo_server`, I get the error `Link 'ee_link' not found in model 'ur5_robot'`

dev@dev-Dell-G15-5511:~$  rosrun moveit_servo servo_server
[ INFO] [1686684547.223705099]: Loading robot model 'ur5_robot'...
[ WARN] [1686684547.252379919]: IK plugin for group 'manipulator' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1686684547.253902859]: IK Using joint shoulder_link -6.28319 6.28319
[ INFO] [1686684547.253920224]: IK Using joint upper_arm_link -6.28319 6.28319
[ INFO] [1686684547.253927325]: IK Using joint forearm_link -3.14159 3.14159
[ INFO] [1686684547.253934123]: IK Using joint wrist_1_link -6.28319 6.28319
[ INFO] [1686684547.253940384]: IK Using joint wrist_2_link -6.28319 6.28319
[ INFO] [1686684547.253947706]: IK Using joint wrist_3_link -6.28319 6.28319
[ INFO] [1686684547.253957119]: Looking in common namespaces for param name: manipulator/position_only_ik
[ INFO] [1686684547.254500274]: Looking in common namespaces for param name: manipulator/solve_type
[ INFO] [1686684547.255271722]: Using solve type Distance
[ INFO] [1686684547.273768636]: Starting planning scene monitor
[ INFO] [1686684547.274843045]: Listening to '/planning_scene'
[ INFO] [1686684547.274857225]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1686684547.275514657]: Listening to '/collision_object'
[ INFO] [1686684547.276201359]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1686684547.277677972]: Listening to '/attached_collision_object' for attached collision objects
[ERROR] [1686684547.585423193, 968.243000000]: Link 'ee_link' not found in model 'ur5_robot'
Segmentation fault (core dumped)

So, I changed the the line # 35 in `ur_simulated_config.yaml` to `ee_frame_name: wrist_3_link  # The name of the end effector link, used to return the EE pose`.

I get these messages now after running the servo server and loading the parameter from the files.

process[servo_server-1]: started with pid [58035]
[ INFO] [1686687381.818404687]: Loading robot model 'ur5_robot'...
[ WARN] [1686687381.842205231]: IK plugin for group 'manipulator' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1686687381.843535293]: IK Using joint shoulder_link -6.28319 6.28319
[ INFO] [1686687381.843549892]: IK Using joint upper_arm_link -6.28319 6.28319
[ INFO] [1686687381.843556750]: IK Using joint forearm_link -3.14159 3.14159
[ INFO] [1686687381.843564354]: IK Using joint wrist_1_link -6.28319 6.28319
[ INFO] [1686687381.843570187]: IK Using joint wrist_2_link -6.28319 6.28319
[ INFO] [1686687381.843576664]: IK Using joint wrist_3_link -6.28319 6.28319
[ INFO] [1686687381.843584430]: Looking in common namespaces for param name: manipulator/position_only_ik
[ INFO] [1686687381.844027213]: Looking in common namespaces for param name: manipulator/solve_type
[ INFO] [1686687381.844670911]: Using solve type Distance
[ INFO] [1686687381.862901940]: Starting planning scene monitor
[ INFO] [1686687381.863630120]: Listening to '/planning_scene'
[ INFO] [1686687381.863638770]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1686687381.864306486]: Listening to '/collision_object'
[ INFO] [1686687381.864974543]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1686687381.866415639]: Listening to '/attached_collision_object' for attached collision objects
[ WARN] [1686687382.174591649, 577.131000000]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking

Nothing happens in gazebo after publishing to `/servo_server/delta_twist_cmds` using the command. Any idea?

Thanks in advance!

rhaschke, 2023-06-13:
@devvaibhav455, your comment is unrelated to this issue. Please open a new one instead of hijacking existing threads!

michael-fenelon, 2024-04-14:
Just for others in the future. What was essential was the installation of the spacenav_node package: http://wiki.ros.org/spacenav_node
Run: 
$ sudo apt install spacenavd
$ sudo apt install ros-noetic-spacenav-node

Then run: roslaunch moveit_servo spacenav_cpp.launch