moveit/moveit #3284: Some question happened in my dual arm motion planning project

zzzbbs, 2022-11-29:
### Description
I treat left arm ,right arm,head_joint as a whole_body group,but when I only excute right_arm to the named home, it seems that the left arm and head joint which I don't want to actuate begin to move. 

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04

### Steps to reproduce
I set the left_arm, right_arm and head_joint as whole_body group.
After excuting right_arm to the target, I try to reset the right_arm using the whole_body group.
The following video is the problem of what I met.
Code:
`cur_joint = self.whole_body_group.get_current_joint_values()
        print(cur_joint) #LSP -> LWR , RSP -> RWR
        #joint_value = [0.0 for x in range(0, len(self.both_arm_group.get_joints()))]
        
        self.whole_body_group.set_named_target("whole_home")
        
        #self.both_arm_group.set_joint_value_target(joint_value)
        # print(self.both_arm_group.plan())
        plan_success,traj,planning_time,error_code = self.whole_body_group.plan()`

https://user-images.githubusercontent.com/81844650/204484756-3e2595e9-fcfb-49d6-8820-2aeca6f6a8eb.mp4

### Expected behaviour
While using the whole body group to reset whole body, the right arm which is away from the robot body should be actuated, and the head arm and the left arm which is now at the named home should not be actuated.

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2022-11-29:
If you want to restrict motion (and planning) to a specific group of joints, you need to define and use a JointModelGroup (JMG) for them. Planning with MoveIt/OMPL is sampling-based, i.e. random. It considers=samples all joints that are part of the selected JMG.

closed 2022-11-29 by rhaschke

zzzbbs, 2022-11-29:
> If you want to restrict motion (and planning) to a specific group of joints, you need to define and use a JointModelGroup (JMG) for them. Planning with MoveIt/OMPL is sampling-based, i.e. random. It considers=samples all joints that are part of the selected JMG.

Thank you very much for your reply. Do you mean that I should devide these joints into different move groups?

rhaschke, 2022-11-29:
Yes, define JMGs for each set of joints you want to plan for separately.

zzzbbs, 2022-11-29:
Thank you, I get it. 
What I want to achieve is that both arm and head can move together to their own  "home position" without affecting each other. So I composite them into "whole_body" group. If I devided into three move_group , it will be executed serially rather than in parallel. E.g, left_arm -> right_arm -> head. 
Could you offer me some suggestions about how to use only one move_group to make these motion planning or excuting in parallel?

rhaschke, 2022-11-29:
[MTC](https://github.com/ros-planning/moveit_task_constructor)'s `Merger` provides the desired functionality. 
MoveIt itself doesn't: https://github.com/ros-planning/moveit/issues/705

simonschmeisser, 2022-11-29:
See also #3243 for ongoing work to support parallel independent execution.

zzzbbs, 2022-11-29:
> See also #3243 for ongoing work to support parallel independent execution.

Thank you for your suggestion! I will have a look.

rhaschke, 2022-11-29:
Yes, the merger.py script illustrates the use of the `Merger`. You still need to use separate JMGs for left/right arm and head. However, trajectories for those separate groups are merged into a single trajectory eventually.