moveit/moveit #3286: No Motion UR Arm using MoveIt servoing. ( Tutorial Not working)

Akumar201, 2022-11-29:
## Background

I want to use [MoveIt Servoing](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html)
example. However in the final step even after publishing the msg to move the robot , it's not moving. 

## Instructions

I am doing the following
1. Git Clone https://github.com/ros-industrial/universal_robot
2. roslaunch ur_gazebo ur5_bringup.launch 
3. roslaunch ur5_moveit_config moveit_planning_execution.launch sim:=true
4. roslaunch ur5_moveit_config moveit_rviz.launch 

After this I tried to see available ROS controller using the command 
`
rosservice call /controller_manager/list_controllers
`
and got the following result 

```
controller: 
  - 
    name: "joint_state_controller"
    state: "running"
    type: "joint_state_controller/JointStateController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::JointStateInterface"
        resources: []
  - 
    name: "joint_group_eff_controller"
    state: "initialized"
    type: "effort_controllers/JointGroupEffortController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::EffortJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint
  - 
    name: "eff_joint_traj_controller"
    state: "running"
    type: "effort_controllers/JointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::EffortJointInterface"
        resources: 
          - elbow_joint
          - shoulder_lift_joint
          - shoulder_pan_joint
          - wrist_1_joint
          - wrist_2_joint
          - wrist_3_joint

```
Then switched the controller to 

```
rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_position_controller']
stop_controllers: ['arm_controller']
strictness: 0
start_asap: false
timeout: 0.0"
```

After this I tried to run 
`roslaunch moveit_servo spacenav_cpp.launch`

As I do not have SpaceNav 3D mouse I ran the following command, however there was no motion in the arm 

```
rostopic pub -r 100 -s /servo_server/delta_twist_cmds geometry_msgs/TwistStamped "header: auto
twist:
  linear:
    x: 0.0
    y: 0.01
    z: -0.01
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```

As you can see from the rqt graph the controller is not changed
[image]

I am not sure what is going wrong and why I am not able to change the controller using the given command 

```
rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_position_controller']
stop_controllers: ['arm_controller']
strictness: 0
start_asap: false
timeout: 0.0"
```

AndyZe, 2022-11-29:
It looks like there was a commit in the `universal_robot` repo that broke the demo:

https://github.com/ros-industrial/universal_robot/commits/melodic-devel/ur_gazebo/config/ur5_controllers.yaml

Sorry. The easiest fix probably would be to roll your commit of `universal_robot` back by 2 commits.

Akumar201, 2022-11-29:
Hi, I am newbee, please correct me if I am wrong,  so by rolling back by 2 commit means that you want me to use the older version of there software. I was thinking of using 1.3.0 as it was modified 19 days ago or should I use 1.2.7 which was comited on Nov 23 2019 . Link to Tags https://github.com/ros-industrial/universal_robot/tags .

Akumar201, 2022-11-30:
Hi, I checkout to 1.2.7 as the commands in the tutorial matches with that version. I was able to change the controller in to Joint_group_position_controller however when using the following command 

```
rostopic pub -r 100 -s /servo_server/delta_twist_cmds geometry_msgs/TwistStamped "header: auto
twist:
  linear:
    x: 0.0
    y: 0.01
    z: -0.01
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```
I can not see the arm moving 
I would like to share the rqt graph. Please help me to debug the issue 

[image]