moveit/moveit #3288: Very close to a singularity, emergency stop - MoveIt servoing

Akumar201, 2022-12-06:
## Background
I am trying to mimic the [MoveIt  Servoing Tutorial](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html)

## Instructions
I am following the tutorial command 

```
1. roslaunch ur_gazebo ur5.launch
2. roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch sim:=true
3. roslaunch ur5_moveit_config moveit_rviz.launch config:=true
4.rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_position_controller']
  stop_controllers: ['arm_controller']
  strictness: 0
  start_asap: false
  timeout: 0.0"
5. roslaunch moveit_servo spacenav_cpp.launch
6. rostopic pub -r 100 -s /servo_server/delta_twist_cmds geometry_msgs/TwistStamped "header: auto
twist:
  linear:
    x: 0.0
    y: 0.01
    z: -0.01
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```

Sometimes the robot move, however many times it gives out the following error, I also tried moving it using rviz to avoid singularity. 

```
[ WARN] [1670292370.686508103, 1035.807000000]: Very close to a singularity, emergency stop
[ WARN] [1670292400.723346959, 1065.807000000]: Very close to a singularity, emergency stop
[ WARN] [1670292430.760863066, 1095.807000000]: Very close to a singularity, emergency stop

```

rhaschke, 2022-12-06:
If you move the robot into a singularity, the emergency stop is the intended behavior. 
What exactly do you want us to do about this?

Akumar201, 2022-12-06:
I would like to know that the values I am providing by the following command is causing the robot to move to singularity position and is the values relative to base or it's incremental ? I tried move the robot at different position using rviz also before using the following command. Once the desired controller is switched for example pointGroupPositionControllers the arm does not take plan a path using rviz and shows path execution as failed. 

```

rostopic pub -r 100 -s /servo_server/delta_twist_cmds geometry_msgs/TwistStamped "header: auto
twist:
  linear:
    x: 0.0
    y: 0.01
    z: -0.01
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```