moveit/moveit #3289: How to make the robotic arm reach a fixed position at a specified time

zzzbbs, 2022-12-06:
Hi, 
I have a question while using moveit planning interface. I want the robotic arm to arrive at a certain position at a fixed time, but the plan provided by moving it seems to have no interface for specifying the arrival time. Could you give me some suggestions about this.Thanks~

closed 2022-12-09 by zzzbbs

zzzbbs, 2022-12-09:
I found a easy solution, which may not work well on real time motion planning using moveit. Firstly, I set the target position of what I want. Secondly, I used plan() function to get a RobotTrajectory. Because I don't need to work on real time, I used the position of other joints only. I extract the position from trajectory file, and interpolate the position series to the time of what I want. Finally, I transform the interpolated trajectory to the RobotTrajectory format, used moveit to actuate the motor to the target position in duration. The most disadvantage of this method is that it cannot work on the real time environment.

rhaschke, 2022-12-09:
Thanks. So, what actually is missing is a method to scale down an existing RobotTrajectory to a longer duration.