moveit/moveit #3290: Failed to build moveit_ros_perception due to missing VTK library

shuobh, 2022-12-06:
### Description

Failed to build moveit_ros_perception due to missing VTK library. This could be fixed by adding a line in the CMake file.

### Your environment
* ROS Distro: [Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source build?
* master branch

### Steps to reproduce
Followed the [tutorial](https://moveit.ros.org/install/source/) to build but failed to build moveit_ros_perception due to missing VTK library.
```
/usr/bin/ld: cannot find -lvtkRenderingOpenGL2
/usr/bin/ld: cannot find -lvtkCommonCore
/usr/bin/ld: cannot find -lvtksys
/usr/bin/ld: cannot find -lvtkCommonDataModel
/usr/bin/ld: cannot find -lvtkCommonMath
/usr/bin/ld: cannot find -lvtkCommonMisc
/usr/bin/ld: cannot find -lvtkCommonSystem
/usr/bin/ld: cannot find -lvtkCommonTransforms
/usr/bin/ld: cannot find -lvtkCommonExecutionModel
/usr/bin/ld: cannot find -lvtkIOImage
/usr/bin/ld: cannot find -lvtkDICOMParser
/usr/bin/ld: cannot find -lvtkmetaio
/usr/bin/ld: cannot find -lvtkImagingCore
/usr/bin/ld: cannot find -lvtkRenderingCore
/usr/bin/ld: cannot find -lvtkCommonColor
/usr/bin/ld: cannot find -lvtkCommonComputationalGeometry
/usr/bin/ld: cannot find -lvtkFiltersCore
/usr/bin/ld: cannot find -lvtkFiltersGeneral
/usr/bin/ld: cannot find -lvtkFiltersGeometry
/usr/bin/ld: cannot find -lvtkFiltersSources
/usr/bin/ld: cannot find -lvtkInteractionStyle
/usr/bin/ld: cannot find -lvtkFiltersExtraction
/usr/bin/ld: cannot find -lvtkFiltersStatistics
/usr/bin/ld: cannot find -lvtkImagingFourier
/usr/bin/ld: cannot find -lvtkalglib
/usr/bin/ld: cannot find -lvtkRenderingLOD
/usr/bin/ld: cannot find -lvtkFiltersModeling
/usr/bin/ld: cannot find -lvtkIOPLY
/usr/bin/ld: cannot find -lvtkIOCore
/usr/bin/ld: cannot find -lvtkFiltersTexture
/usr/bin/ld: cannot find -lvtkRenderingFreeType
/usr/bin/ld: cannot find -lvtkIOExport
/usr/bin/ld: cannot find -lvtkRenderingGL2PSOpenGL2
/usr/bin/ld: cannot find -lvtkIOGeometry
/usr/bin/ld: cannot find -lvtkIOLegacy
collect2: error: ld returned 1 exit status
```

### Solution

Add `find_package(VTK REQUIRED)` to moveit/moveit_ros/perception/CMakeLists.txt (line 47) fixed the issue.

shuobh, 2022-12-07:
> I don't think MoveIt depends on VTK. Where exactly did you find this dependency? Could you please provide more context?
I've included the entire error log from the build, seems like libmoveit_semantic_world.so.1.1.10 are looking for the VTK libraries?

rhaschke, 2022-12-07:
Nope, libmoveit_semantic_world.so doesn't need VTK:
https://github.com/ros-planning/moveit/blob/1f4435190f7498d746b8066fb72c956eebe8eeef/moveit_ros/perception/semantic_world/CMakeLists.txt#L3-L6

```sh
ldd /opt/ros/noetic/lib/libmoveit_semantic_world.so | grep vtk
```

Do you use a custom OpenCV with VTK support? If OpenCV pulls in VTK, you should file an issue with them, because they don't include the required find_package(VTK) in their cmake config.

shuobh, 2022-12-07:
> Nope, libmoveit_semantic_world.so doesn't need VTK:
> 
> https://github.com/ros-planning/moveit/blob/1f4435190f7498d746b8066fb72c956eebe8eeef/moveit_ros/perception/semantic_world/CMakeLists.txt#L3-L6
> 
> ```shell
> ldd /opt/ros/noetic/lib/libmoveit_semantic_world.so | grep vtk
> ```
> 
> Do you use a custom OpenCV with VTK support? If OpenCV pulls in VTK, you should file an issue with them, because they don't include the required find_package(VTK) in their cmake config.

I see, that might be the issue. I will check OpenCV. Thanks for the response.

closed 2022-12-07 by shuobh