moveit/moveit #3292: My mobile base in Moveit! cannot move.

TKroos18, 2022-12-08:
### Description

(I think maybe it is not a bug......)
When I have read this(https://moveit.picknik.ai/humble/doc/examples/mobile_base_arm/mobile_base_arm_tutorial.html#planning-for-differential-drive-mobile-base-and-an-arm), I started to think about whether it is possible for this plugin to be used in Moveit!(not Moveit2). So I backported this plugin to ROS1. However, I found that my plugin was working but my mobile base cannot move. 

I have done these things:
1、Modify the cpp file of plugin
2、Register my plugin as a KinematicsBase implementation
3、Create three planning group: mobile_base_arm, arm, base. arm and base are sub-groups of mobile_base_arm. The kinematics solver of arm is kdl. The kinematics solver of base is none. The kinematics solver of mobile_base_arm is my plugin.  
4、 Set up two joint properties.
5、Run demo.launch. I found that my plugin was working and my arm moved, but my mobile base did not move. 

so did I omit any important steps...?

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04

simonschmeisser, 2023-01-02:
Did you get any errors or warnings? Might be helpful to know exactly what you mean by "did not move"

Also if you can upload you changes and you moveit setup to github and link it here maybe someone can reproduce it

scchow, 2023-02-06:
Hi, I have also been looking into mobile manipulation and potentially backporting the differential drive aspects of MoveIt!2 to MoveIt!. 

From an initial glance, it appears the tutorial enables diff drive by setting `differential drive` and `motion_model` joint properties [here](https://moveit.picknik.ai/humble/doc/examples/mobile_base_arm/mobile_base_arm_tutorial.html#setting-up-the-kinematics-model-for-the-base). These properties are parsed by [MoveIt!2 in `planar_joint_model.cpp`](https://github.com/ros-planning/moveit2/blob/main/moveit_core/robot_model/src/planar_joint_model.cpp#L149). 

However, if you look at the [corresponding file in ROS 1 Moveit](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_model/src/planar_joint_model.cpp), it appears that this functionality has not been implemented.

It would be necessary to backport this functionality into MoveIt! (in addition to other compatibility issues that might crop up) if you wish to reproduce the tutorial in Moveit! 1. It appears that the developers of Stretch added this functionality to MoveIt2 in this [PR](https://github.com/ros-planning/moveit2/pull/390). You may wish to use this as a reference in your backport.

I will probably be doing something similar, but won't have the cycles for it until later this month/next month, so let me know if you do move forward with this in the meantime. Good luck!

TKroos18, 2023-02-08:
@simonschmeisser Thank you. I will log the errors when I try again next time.

@scchow I have thought that maybe Moveit! doesn't support differential drive and motion_model joint properties. After reading your comment, I fully believe it now. Thank you very much!
Now I have two ideas.(I don't know whether they can work) One of them is that I can design a kinematic plugin for differential planer joint to make my mobile base move. Another is that I can set three joint(two prismatic joints and a revolute joint) instead of a planer joint. 
I am also interested in your idea. Good luck!