moveit/moveit #3293: New Release for Noetic

vatanaksoytezer, 2022-12-08:
I want to cut a new release for Noetic tomorrow morning since there have been many updates and I've had a new release requested from a few different people.

Let me know if any of the maintainers have any objections/concerns. Tagging @v4hn and @rhaschke for visibility.

rhaschke, 2022-12-08:
I planned a new release already two weeks ago: https://github.com/ros-planning/moveit/issues/3275. Unfortunately, no maintainer reviewed the open PRs listed there... 
I'm fine to cut the new release. I have actually already prepared most of it... But I'm waiting for some reviews!

v4hn, 2022-12-09:
Sorry, as you undoubtedly noticed I'm busy with non-moveit projects and family emergencies since September already and thus hard to reach here. I have plans to change that situation again soon, but we will see how it goes.
If you need urgent reviews from my side please send me a personal mail with links and I'll try to allocate some time to address them.

rhaschke, 2022-12-09:
@v4hn, sure I noticed that you were busy with other stuff. You are not the only MoveIt maintainer...
If you have some time, I would like to sync about MTC via a phone call at some point.

AndyZe, 2022-12-20:
@rhaschke do you mind if @vatanaksoytezer takes care of this? A client is anxious to get a new release

rhaschke, 2022-12-20:
Yes, I do mind, because I already prepared most of it. I was waiting for the ruckig backport to get finished. Do you think it is ready now?  I can finalize the release tomorrow.

rhaschke, 2022-12-21:
This is blocked currently by failing `ruckig` tests: https://github.com/ros-planning/moveit/pull/2902#issuecomment-1361459380

vatanaksoytezer, 2022-12-22:
Thank you @AndyZe and @rhaschke for making this happen, closing this as completed!

closed 2022-12-22 by vatanaksoytezer