moveit/moveit #3299: Collision detection gives wrong result (and no info on closest points)

elena-ecn, 2022-12-20:
### Description

Hi. I am trying to compute the minimum distance between a robot and an object and find the nearest points (and eventually the normal vector connecting them). Essentially I need the information from the collision_detection::CollisionEnv.distanceRobot() method. I am adding a collision object (similar to the "Move Group C++ Interface" tutorial) and checking whether there is a collision in two different ways: using PlanningScene.checkCollision() and using CollisionEnvFCL.distanceRobot(). 

However, both methods **incorrectly** report that there is no collision even though there is one and the minimum distances between the robot and the object are also wrong.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Source
* If source, which branch? master

### Steps to reproduce

Consider the following example node:

```cpp
#include <ros/ros.h>

// MoveIt
#include <moveit/robot_model_loader/robot_model_loader.h>
#include <moveit/planning_scene/planning_scene.h>
#include <moveit_msgs/CollisionObject.h>
#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit/collision_detection_fcl/collision_env_fcl.h>

int main(int argc, char** argv)
{
  ros::init(argc, argv, "panda_coll_detection");
  ros::AsyncSpinner spinner(1);
  spinner.start();

  // ---------------------------------------------------------------------------
  // Define interfaces
  // ---------------------------------------------------------------------------

  // Define move_group_interface & planning_scene_interface
  static const std::string PLANNING_GROUP = "panda_arm";
  moveit::planning_interface::MoveGroupInterface move_group_interface(PLANNING_GROUP);
  moveit::planning_interface::PlanningSceneInterface planning_scene_interface; 

  // Define planning_scene
  robot_model_loader::RobotModelLoader robot_model_loader("robot_description");
  const moveit::core::RobotModelPtr& kinematic_model = robot_model_loader.getModel();
  planning_scene::PlanningScene planning_scene(kinematic_model);

  // ---------------------------------------------------------------------------
  // Add a collision object
  // ---------------------------------------------------------------------------

  // Define a collision object ROS message for the robot to avoid
  moveit_msgs::CollisionObject collision_object;
  collision_object.header.frame_id = move_group_interface.getPlanningFrame();
  collision_object.id = "box1";

  // Define a box to add to the world
  shape_msgs::SolidPrimitive primitive;
  primitive.type = primitive.BOX;
  primitive.dimensions.resize(3);
  primitive.dimensions[primitive.BOX_X] = 0.4;
  primitive.dimensions[primitive.BOX_Y] = 0.02;
  primitive.dimensions[primitive.BOX_Z] = 0.02;

  // Define a pose for the box (specified relative to frame_id)
  geometry_msgs::Pose box_pose;
  box_pose.orientation.w = 1.0;
  box_pose.position.x = 0.1;
  box_pose.position.y = 0.0;
  box_pose.position.z = 0.25; 

  // Add box to collision_object
  collision_object.primitives.push_back(primitive);
  collision_object.primitive_poses.push_back(box_pose);
  collision_object.operation = collision_object.ADD;

  // Define a vector of collision objects that could contain additional objects
  std::vector<moveit_msgs::CollisionObject> collision_objects;
  collision_objects.push_back(collision_object);

  // Add the collision object into the world  
  planning_scene_interface.applyCollisionObjects(collision_objects);

  // ---------------------------------------------------------------------------
  // Test 1: Collision Checking with planning_scene.checkCollision()
  // ---------------------------------------------------------------------------

  // Define collision request & result
  collision_detection::CollisionRequest collision_request;
  collision_detection::CollisionResult collision_result;
  collision_request.distance = true;
  collision_request.contacts = true;

  // Get robot state 
  moveit::core::RobotState copied_state1 = planning_scene.getCurrentState();

  // Check whether a specified state (robot_state) is in collision
  planning_scene.checkCollision(collision_request, collision_result, copied_state1);
  
  // Show result
  ROS_INFO_STREAM("Test 1: Current state is " << (collision_result.collision ? "in" : "not in") << " collision");
  ROS_INFO_STREAM("Min distance between two bodies=" << collision_result.distance << "\n");
  
  // If collision, show pairs of bodies in contact 
  collision_detection::CollisionResult::ContactMap::const_iterator it;
  for (it = collision_result.contacts.begin(); it != collision_result.contacts.end(); ++it)
  {
    ROS_INFO("Contact between: %s and %s", it->first.first.c_str(), it->first.second.c_str());
  }

  
  // ---------------------------------------------------------------------------
  // Test 2: Collision Checking with CollisionEnvFCL.distanceRobot()
  // ---------------------------------------------------------------------------
  
  // Define distance request & result
  auto distance_request = collision_detection::DistanceRequest();
  auto distance_result = collision_detection::DistanceResult();
  distance_request.enable_nearest_points =true;

  // Get robot state & model
  moveit::core::RobotState copied_state2 = planning_scene.getCurrentState();
  const moveit::core::RobotModelConstPtr kinematic_model_cnstPtr = robot_model_loader.getModel();
  
  // Defince collision_env
  collision_detection::CollisionEnvFCL collision_env(kinematic_model_cnstPtr, 0.0, 1.0);
  
  // Check whether a specified state (robot_state) is in collision
  collision_env.distanceRobot(distance_request, distance_result, copied_state2);

  // Show result
  ROS_INFO_STREAM("Test 2: Current state is " << (distance_result.collision ? "in" : "not in") << " collision");
  ROS_INFO_STREAM("Min distance between two bodies=" << distance_result.minimum_distance.distance << "\n");

  std::cout << "min dist data...\n" <<
        "distance = " << distance_result.minimum_distance.distance << "\n"
        "link name = " << distance_result.minimum_distance.link_names[0] << "\n"
        "nearest pt = " << distance_result.minimum_distance.nearest_points[0] << "\n"
        "link name = " << distance_result.minimum_distance.link_names[1] << "\n"
        "nearest pt = " << distance_result.minimum_distance.nearest_points[1] << "\n"
        << std::endl;
  

  ros::shutdown();
  return 0;
}
```

Run `roslaunch panda_moveit_config demo.launch` and then in another terminal run the above node.

### Behaviour

Initially, the object is placed in a non-colliding position (e.g., box_pose.position is 1.0, 0.0, 0.95). 

[image]

However, the minimum distance is wrong and no information about the nearest points is returned.

```
[ INFO] [1671499841.390624825]: Test 1: Current state is not in collision
[ INFO] [1671499841.390692223]: Min distance between two bodies=0.123477

[ INFO] [1671499841.390780315]: Test 2: Current state is not in collision
[ INFO] [1671499841.390811337]: Min distance between two bodies=1.79769e+308

min dist data...
distance = 1.79769e+308
link name = 
nearest pt = 0
0
0
link name = 
nearest pt = 0
0
0
```

Moreover, when the robot is in collision with the object, as can be seen in the following figure

[image]

the collision detection incorrectly reports that there is no collision. Additionally, the min distances are the same as before and again no information on the nearest points is returned.

```
[ INFO] [1671500298.648777367]: Test 1: Current state is not in collision
[ INFO] [1671500298.648834477]: Min distance between two bodies=0.123477

[ INFO] [1671500298.648897442]: Test 2: Current state is not in collision
[ INFO] [1671500298.648932176]: Min distance between two bodies=1.79769e+308

min dist data...
distance = 1.79769e+308
link name = 
nearest pt = 0
0
0
link name = 
nearest pt = 0
0
0
```

Any help on how to resolve this would be greatly appreciated.

AndyZe, 2022-12-20:
I don't think this is the issue, but it would be good form to make this change:

`ros::AsyncSpinner spinner(8);`

I usually set this equal to num_threads available for my computer.

AndyZe, 2022-12-20:
A quick guess is that this isn't being used properly. I noticed this isn't in the tutorial:

`moveit::planning_interface::PlanningSceneInterface planning_scene_interface;`

elena-ecn, 2022-12-20:
@AndyZe Thank you for your reply (and the spinner suggestion). 

Actually, this line (`moveit::planning_interface::PlanningSceneInterface planning_scene_interface;`) is from the "Move Group C++ Interface" tutorial. The planning_scene_interface is used to add the collision object to the world. However, the collision detection is done via planning_scene so I think maybe there is a disconnect there.

I added the following code to see if the object is known to both the planning_scene and planning_scene_interface.
```cpp
// Show known objects in planning_scene
planning_scene.printKnownObjects();
  
// Show known objects in planning_scene_interface
std::vector< std::string > obj_names = planning_scene_interface.getKnownObjectNames();
for (std::string i: obj_names)
  std::cout << i << "\n";
```

and this is the result I get:
```
-----------------------------------------
PlanningScene Known Objects:     // <- this is from planning_scene
  - Collision World Objects:
   - Attached Bodies:
-----------------------------------------
box1  // <- this is from planning_scene_interface
```

So, it seems that planning_scene, which is responsible for the collision checking, does not know about the collision object.

rhaschke, 2022-12-21:
The `PlanningScene` class is just a data object _holding_ a specific scene state. There is no automatic synchronization with the move_group node's current `PlanningScene`.
The `PlanningSceneInterface` class on the other allows modification of the move_group node's `PlanningScene` via appropriate ROS communication (topics or services). However, it doesn't hold any data.

To fetch the current `PlanningScene` state from the move_group node, you need to call the `get_planning_scene` service:
```c++
	planning_scene = std::make_shared<planning_scene::PlanningScene>(robot_model_);

	ros::NodeHandle h;
	ros::ServiceClient client = h.serviceClient<moveit_msgs::GetPlanningScene>("get_planning_scene");

	ros::Duration timeout(1.0);
	if (client.waitForExistence(timeout)) {
		moveit_msgs::GetPlanningScene::Request req;
		moveit_msgs::GetPlanningScene::Response res;

		if (client.call(req, res))
			planning_scene->setPlanningSceneMsg(res.scene); // apply result to actual PlanningScene
	}
```

elena-ecn, 2022-12-22:
Oh, I see. This makes sense now. Thank you so much! 

After incorporating your suggestion, Test 1 via the planning_scene.checkCollision() now correctly reports whether there is a collision or not. 

However, I still have the following issues:

1. In Test 1, the `collision_result.distance` is the same even when the object is placed in different positions. I would expect this to give the minimum distance between the robot and the object. Isn't this the case?
2. Test 2 still fails though. However, I realize that currently no information about the scene is passed to it and I assume that I am probably not defining collision_env properly. 
I attempted to also declare it as `collision_detection::CollisionEnvFCL collision_env(kinematic_model_cnstPtr, planning_scene->getWorld(), 0.0, 1.0);` or `collision_detection::CollisionEnvFCL collision_env = planning_scene->getCollisionEnv();` but I get errors due to type mismatch. What would be the right way to declare this so I can use distanceRobot()?

simonschmeisser, 2022-12-23:
A `PlanningScene` can have one or multiple collision detectors set. These maintain a CollisionEnvironment internaly which is updated via callbacks when the PlanningScene is modified. You create a stand alone empty CollisionEnvironment instead of getting the one from the PlanningScene which contains your object. You would use `PlanningScene::getCollisionEnv` to get the currently active one and could then cast to eg the FCL one if you really need to.  But the `PlanningScene::checkCollision` functions will probably expose all the features you need?

elena-ecn, 2022-12-24:
Thank you for your reply! I am gaining a better understanding of how this works.

Since I need to find the nearest points between the robot and the object and the normal vector connecting them, I need the information from `CollisionEnvFCL::distanceRobot` (`PlanningScene::checkCollision` does not provide these). Unfortunately though, `distanceRobot()` is not available through the Python API and since I am not familiar with C++, this is a little challenging. I could get the current collision environment by:
`const collision_detection::CollisionEnvConstPtr c_env = planning_scene->getCollisionEnv();`
Could you please let me know how to cast this to the FCL environment?

elena-ecn, 2023-01-09:
Could someone please help me with this? 

What is the proper way to initialize `collision_detection::CollisionEnvFCL` with the currently active environment so I can use the `CollisionEnvFCL::distanceRobot` method?

If I get the active collision environment via
`const collision_detection::CollisionEnvConstPtr c_env = planning_scene->getCollisionEnv();`, how can I cast this to an FCL environment?

matthias88, 2023-06-05:
Hi @elena-ecn, did you find a solution for casting the active collision environment to FCL environment?

simonschmeisser, 2023-06-05:
https://en.cppreference.com/w/cpp/language/dynamic_cast ?

```
const collision_detection::CollisionEnvFclConstPtr fclEnv = dynamic_cast<collision_detection::CollisionEnvFclConstPtr>(planning_scene->getCollisionEnv());
if (!fclEnv) ROS_ERROR("Not using fcl actually!");
....
```

elena-ecn, 2023-06-06:
Hi @matthias88, I couldn't find a way to cast it properly. Also, using the suggestion above, I get: `error: ‘CollisionEnvFclConstPtr’ in namespace ‘collision_detection’ does not name a type; did you mean ‘CollisionEnvConstPtr’?`

elena-ecn, 2023-06-06:
Hi @matthias88, I could not find a way to cast it properly. Also, using the suggestion above I get an error.

However, I did find a solution. I only very recently started working on this again, so I didn't get the chance to update here.

It turns out that the active collision environment was indeed the FCL environment. You can check with:
`planning_scene->getActiveCollisionDetectorName()`

However, in general, you can change  the active environment to be the FCL one with:
```cpp
#include <moveit/collision_detection_fcl/collision_detector_allocator_fcl.h>

planning_scene->setActiveCollisionDetector(collision_detection::CollisionDetectorAllocatorFCL::create(), true);
ROS_INFO_STREAM("Current active collision env is: " << planning_scene->getActiveCollisionDetectorName());
```

I am closing this issue. Thank you to everyone who helped!

closed 2023-06-06 by elena-ecn