moveit/moveit #3301: Python Bindings for trajectory processing module

uahic, 2022-12-28:
Some of Moveits interesting features are available for C++ only, in this case the trajectory processing module as well as the robot trajectory module. I love to prototype in Python and often use it in production (well, in research/public projects) too. In this case I want to smoothly control two arms to follow some waypoints and I tried implementing my own controller for this, ended up with playing around with ruckig and digging into the math of cubic/quintic splines and discovered that moveit is also integrating path smoothing via rucking and has (regarding to my coworker) quite good interpolation out of the box. So I sit down and implemented some python bindings. But on the _melodic-devel_ and noetic branch (currently my real setup is unfortunately still working with melodic). Porting it to the master branch should be easy though.

Before I open a merge request, I'd like to know if this can be done on the melodic branch or if it has to be the master branch first. Secondly, please give me feedback if the bindings expose stuff that doesnt make sense or if you see critical copy operations of vectors that you'd like to see optimized. I'm using pybind11 for the first time (had worked with Cython years ago though) and did not have a look how to avoid copy operations, handling pointers and references properly. 

Last question is if there are any test functions for pybindings or if you restrict yourself only to test the cpp code. 

My Fork:
https://github.com/uahic/moveit
 
Cheers!

simonschmeisser, 2022-12-29:
Welcome! Your experiments sound quite interesting. There are others more knowledgeable about the Python specific aspects but development in general happens either on the master branch or in moveit2. A PR against `melodic-devel` would need to be rebased before merging. Note that you can run `master` on melodic/Ubuntu 18.04 just fine when compiling from source.

rhaschke, 2023-01-03:
Simon is right. You should target the `master` branch and you can use/compile `master` on Melodic as well.
Your python bindings are very welcome. Looks like you doubled some work we have done in still not yet merged PRs #2910 and #2613. An attempt to merge and consolidate all these attempts would be very much welcome!