moveit/moveit #3304: Execution delay issue with move_group execute method

ConradLiuuu, 2023-01-10:
Hi Everyone, 

As title mentioned, I use **move_group execute** method to execute a planned trajectory. 
The trajectory is expected to take 2 seconds from position A to position B.
However, the real situation is it took about 2.8 even 3 seconds to finish execution.

Here are parts of the program
```cpp
moveit::planning_interface::MoveGroupInterface::Plan plan = ...; // assign a planned trajectory

auto t1 = ros::Time::now();
move_group->execute(plan);
auto t2 = ros::Time::now();

cout << "execution time: " << (t2 - t1).toSec() << endl;
``` 

Could you tell me how to solve this issue and what is the point makes this issue?

Thank you

rhaschke, 2023-01-10:
Your robot is not capable to accelerate or move as fast as MoveIt assumes. Adjust max velocity and acceleration parameters to tell MoveIt about your robot's capabilities!

ConradLiuuu, 2023-01-10:
Hi rhaschke, 

Thank you for your reply!

I edited **max_velocity** and **max_acceleration** in joint_limits.yaml. 
This file is located in a package which is created by MoveIt Setup Assistant
Current parameters are 100, previous parameters are 3.14

But it didn't work, I still encountered execution delay issue.

Did I edit a wrong file or wrong parameters?

Here is joint_limits.yaml detail
```
joint_limits:
  joint_1:
    has_velocity_limits: true
    max_velocity: 100
    has_acceleration_limits: true
    max_acceleration: 100
  joint_2:
    has_velocity_limits: true
    max_velocity: 100
    has_acceleration_limits: true
    max_acceleration: 100
  joint_3:
    has_velocity_limits: true
    max_velocity: 100
    has_acceleration_limits: true
    max_acceleration: 100
  joint_4:
    has_velocity_limits: true
    max_velocity: 100
    has_acceleration_limits: true
    max_acceleration: 100
  joint_5:
    has_velocity_limits: true
    max_velocity: 100
    has_acceleration_limits: true
    max_acceleration: 100
  joint_6:
    has_velocity_limits: true
    max_velocity: 100
    has_acceleration_limits: true
    max_acceleration: 100
```

rhaschke, 2023-01-10:
So you even increased the limits! You need to decrease them because your robot is slower than expected...

ConradLiuuu, 2023-01-11:
Hi rhaschke,

Thanks for your advisement!

I edited max_velocity and max_acceleration from 3.14 to 1.5 but it did not work.

Even I set **has_velocity_limits** and **has_acceleration_limits** to false, the situation is the same.

Maybe I misunderstood your advisement, could you describe the solution more detail?

Thank you

rhaschke, 2023-01-11:
`has_*_limits` should be true. Otherwise, the limit will be ignored completely.
Reducing the limits should have an effect. Try to reduce even more, say to 0.1 to validate that the new values are considered.
You can check that the values are actually used by calling python's `RobotInterface::get_joint_limits(joint_name)`.
Note that you cannot change those values dynamically. They have to be set on the parameter server before starting any node (e.g. move_group) using them.

ConradLiuuu, 2023-01-12:
Hi rhaschke,

I tried to set `max_velocity` and `max_acceleration` to 0.1, also set `has_velocity_limits` and `has_acceleration_limits` to true.
Before I ran `move_group` node, I also checked values of `has_*_limits` and `max_*` by rosparam command, they are correct values.

However robot motion is slower than I expected. (I expect the motion took 3 seconds to finish, but it took 3.4 seconds)

Here are parameters I set
```
joint_limits:
  joint_1:
    has_velocity_limits: true
    max_velocity: 0.1
    has_acceleration_limits: true
    max_acceleration: 0.1
  joint_2:
    has_velocity_limits: true
    max_velocity: 0.1
    has_acceleration_limits: true
    max_acceleration: 0.1
  joint_3:
    has_velocity_limits: true
    max_velocity: 0.1
    has_acceleration_limits: true
    max_acceleration: 0.1
  joint_4:
    has_velocity_limits: true
    max_velocity: 0.1
    has_acceleration_limits: true
    max_acceleration: 0.1
  joint_5:
    has_velocity_limits: true
    max_velocity: 0.1
    has_acceleration_limits: true
    max_acceleration: 0.1
  joint_6:
    has_velocity_limits: true
    max_velocity: 0.1
    has_acceleration_limits: true
    max_acceleration: 0.1
```

rhaschke, 2023-01-12:
Did the robot slow down at all when changing the parameters, i.e. do the parameters have an effect on the motion speed?
How do you know the expected length of 3s? Do you send a hard-coded trajectory?

MoveIt uses these ROS parameters to compute the timing of a computed path. 
Which time parameterization plugin do you use?

ConradLiuuu, 2023-01-13:
Hi rhaschke, 

The parameters, they did have an effect on motion when I use move_group `setJointValueTarget` and `move` method.

For example: 

I use `move` method to control robot moves from initial position to position A.

When I set the parameters to 0.1, it moves very slow! 

In contrast, I set the parameters to 100 robot moves faster.

This is the whole story that how I treat the trajectory.

1. Plan a trajectory with my own motion planning model, it output `moveit_msgs::RobotTrajectory` type trajectory and I assign it to `moveit::planning_interface::MoveGroup::Plan` type object.
2. Save the trajectory as a yaml file. (Look attachment for detail)
3. Load trajectory in another node to avoid re-plan same trajectory to save time.
4. Use MoveIt `execute` method in that node to control robot.

As the attachment shows, the last `time_from_start` value is 3.

I checked this value in the node after I load trajectory, its `last time_from_start` value is 3 too.

The following program is how I check the motion time

```cpp
cout << plan.trajectory_.joint_trajectory.points[plan.trajectory_.joint_trajectory.points.size()-1].time_from_start.toSec() << endl;
```

As I mentioned above, I plan a trajectory by myself then I use move_group `execute` method to control robot.
These are planned trajectories 
[trajectories.zip](https://github.com/ros-planning/moveit/files/10407231/trajectories.zip)

I do not use other parameter to affect planning time and computed path.

That’s why I said that I encountered execution delay issue when I used move_group `execute` method.

Hope this information can help to analysis problem

Thank you!

ConradLiuuu, 2023-01-13:
# reply for rhaschke 2023-01-12

Hi rhaschke, 

The parameters, they did have an effect on motion when I use move_group `setJointValueTarget` and `move` method.

For example: 

I use `move` method to control robot moves from initial position to position A.

When I set the parameters to 0.1, it moves very slow! 

In contrast, I set the parameters to 100 robot moves faster.

This is the whole story that how I treat the trajectory.

1. Plan a trajectory with my own motion planning model, it output `moveit_msgs::RobotTrajectory` type trajectory and I assign it to `moveit::planning_interface::MoveGroup::Plan` type object.
2. Save the trajectory as a yaml file. (Look attachment for detail)
3. Load trajectory in another node to avoid re-plan same trajectory to save time.
4. Use MoveIt `execute` method in that node to control robot.

As the attachment shows, the last `time_from_start` value is 3.

I checked this value in the node after I load trajectory, its `last time_from_start` value is 3 too.

The following program is how I check the motion time

```cpp
cout << plan.trajectory_.joint_trajectory.points[plan.trajectory_.joint_trajectory.points.size()-1].time_from_start.toSec() << endl;
```

As I mentioned above, I plan a trajectory by myself then I use move_group `execute` method to control robot.
These are planned trajectories
[trajectories.zip](https://github.com/ros-planning/moveit/files/10407238/trajectories.zip)

I do not use other parameter to affect planning time and computed path.

That’s why I said that I encountered execution delay issue when I used move_group `execute` method.

Hope this information can help to analysis problem

Thank you!

rhaschke, 2023-01-15:
Your experiments with the `move` method prove that MoveIt correctly reads and interprets the limit parameters.
I think the culprit is your _"own planning model"_:
> 1. Plan a trajectory with my own motion planning model, it output moveit_msgs::RobotTrajectory type trajectory and I assign it to moveit::planning_interface::MoveGroup::Plan type object.

Obviously, _your_ planning method does not consider these limit parameters. The computed trajectory is _too fast_ for your robot.

ConradLiuuu, 2023-01-16:
Hi rhaschke,

Thank you for pointing out possible problem.

According to your advisement, I did some experiments to find out the exact problem.
1. I set motion time to 5 second and I checked my robot has capability to execute this trajectory. But the real execution time is about 5.4 second.
2.  I decrease points amount in trajectory from 100 to 10, the result is the same, execution time also is about 5.4 second.
3. I set `GoalJointTolerance`, `GoalPositionTolerance` and `GoalOrientationTolerance` of MoveIt from 0.000001 to 0.001, execution time also is about 5.4 second.
4. I use `plan` method of MoveIt to make sure the trajectory contained limits parameters we talked above.
Motion time of the plan is about 0.76 second and execution time is about 1.24 second, they have 0.48 different. This value is close to the delay time I get.

Here is how I ues `plan` and `execute` method of MoveIt
```
moveit::planning_interface::MoveGroupInterface::Plan my_plan;
auto success = (move_group->plan(my_plan) == moveit::planning_interface::MoveItErrorCode::SUCCESS);
cout << "motion time: " << my_plan.trajectory_.joint_trajectory.points[my_plan.trajectory_.joint_trajectory.points.size()-1].time_from_start.toSec() << endl;
// motion time is about 0.76 second

auto t11 = ros::Time::now();
move_group->execute(my_plan); auto t22 = ros::Time::now();
cout << "execution time: " << (t22-t11).toSec() << endl;
// execution time is about 1.24 second
```

Overall, I encountered delay issue when I use `execute` method.
In this experiment, the delay time is about 0.4 second.

rhaschke, 2023-01-16:
In your code snippet, you not only measure execution time, but also the time to communicate the plan to the move_group node and to prepare execution of the plan. However, I estimate that time to be less than 0.1s.
So, your robot controller cannot nicely track a given trajectory. There are multiple tolerance parameters to adjust in that case. See [this tutorial](https://sir.upc.edu/projects/rostutorials/10-gazebo_control_tutorial/index.html) for an overview. 
I suggest increasing the `goal_time` tolerance parameter. See the [JointTrajectoryController's wiki](http://wiki.ros.org/joint_trajectory_controller#Parameters).

I'm going to close this issue now as this is not MoveIt-related. Please refer to https://answers.ros.org for further help.

closed 2023-01-16 by rhaschke

VideoSystemsTech, 2023-05-26:
I also noticed @ConradLiuuu 's problem. You can reproduce it with a simple script (here I'm using Python). I consistently get around 0.4s in this example.

```
import sys
import time
import rospy
import moveit_commander
from moveit_msgs.msg import RobotTrajectory

moveit_commander.roscpp_initialize(sys.argv)
rospy.init_node("move_group_python_interface_tutorial", anonymous=True)

group_name = "arm"
move_group = moveit_commander.MoveGroupCommander(group_name)

trajectory = RobotTrajectory()
trajectory.joint_trajectory.joint_names = []
trajectory.joint_trajectory.points = []

start = time.time()
move_group.execute(trajectory, wait=True)
end = time.time()
print(f"Took {(end-start)*1000} ms")
```

I can also confirm this happens when executing meaningful trajectories: the difference between the "ideal" execution time and the actual execution time ranges from ~0.1s to ~0.5s, regardless of its duration.

rhaschke, 2023-05-30:
Thanks for tracking down this problem.