moveit/moveit #3307: CMake Error：(message):   Could NOT find Boost

mrcjw, 2023-01-13:
### Description

Hello, I'm trying to use moveit on the machine with ros installed. "Could NOT find Boost" appears during compilation.

### Your environment
* ROS Distro: [Melodic]
* OS Version: Ubuntu 18.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? melodic-devel

### Steps to reproduce
I follow the tutorial“ https://moveit.ros.org/install/source/ "Install the source code. not" catkin config -- blacklist ", I executed" Optional: Install a Compiler Cache ". When I run catkin build, I get the following error
### Expected behaviour
It should be built without error.

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output
Starting  >>> moveit_ros_planning_interface                                    
_______________________________________________________________________________
Errors     << moveit_ros_planning_interface:cmake /home/mrc/moveit_ws/logs/moveit_ros_planning_interface/build.cmake.010.log
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Boost (missing: python27) (found version "1.65.1")
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /opt/ros/melodic/share/eigenpy/cmake/boost/FindBoost.cmake:2384 (find_package_handle_standard_args)
  /opt/ros/melodic/share/eigenpy/cmake/boost.cmake:144 (find_package)
  /opt/ros/melodic/share/eigenpy/cmake/eigenpyConfig.cmake:144 (SEARCH_FOR_BOOST_PYTHON)
  CMakeLists.txt:50 (find_package)

cd /home/mrc/moveit_ws/build/moveit_ros_planning_interface; catkin build --get-env moveit_ros_planning_interface | catkin env -si  /usr/bin/cmake /home/mrc/moveit_ws/src/moveit/moveit_ros/planning_interface --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mrc/moveit_ws/devel/.private/moveit_ros_planning_interface -DCMAKE_INSTALL_PREFIX=/home/mrc/moveit_ws/install -DCMAKE_BUILD_TYPE=Release -DBoost_DEBUG=ON; cd -
...............................................................................
Failed     << moveit_ros_planning_interface:cmake                            [ Exited with code 1 ]
Failed    <<< moveit_ros_planning_interface                                  [ 0.8 seconds ]
Abandoned <<< moveit_commander                                               [ Unrelated job failed ]
Abandoned <<< moveit_planners_chomp                                          [ Unrelated job failed ]
Abandoned <<< moveit_ros_visualization                                       [ Unrelated job failed ]
Abandoned <<< moveit_servo                                                   [ Unrelated job failed ]
Abandoned <<< moveit_setup_assistant                                         [ Unrelated job failed ]
Abandoned <<< moveit_tutorials                                               [ Unrelated job failed ]
Abandoned <<< pilz_industrial_motion_planner_testutils                       [ Unrelated job failed ]
Abandoned <<< pilz_industrial_motion_planner                                 [ Unrelated job failed ]
[build] Summary: 32 of 41 packages succeeded.                                  
[build]   Ignored:   6 packages were skipped or are blacklisted.               
[build]   Warnings:  22 packages succeeded with warnings.                      
[build]   Abandoned: 8 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 14.9 seconds total.

rhaschke, 2023-01-15:
Verify that you have the debian package `libboost-python-dev` installed. This should have been installed via: 
`rosdep install -y --from-paths src --ignore-src --rosdistro melodic`

run from your catkin workspace folder.

mrcjw, 2023-01-16:
mrc@ubuntu:~/moveit_ws$ rosdep install -y --from-paths src --ignore-src --rosdistro melodic
#All required rosdeps installed successfully
mrc@ubuntu:~/moveit_ws$ sudo apt-get install libboost-python-dev 
[sudo] password for mrc: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libboost-python-dev is already the newest version (1.65.1.0ubuntu1).
libboost-python-dev set to manually installed.
The following packages were automatically installed and are no longer required:
  fonts-liberation2 fonts-opensymbol gir1.2-gst-plugins-base-1.0
  gir1.2-gudev-1.0 gir1.2-udisks-2.0 grilo-plugins-0.3-base libcdr-0.1-1
  libclucene-contribs1v5 libclucene-core1v5 libcmis-0.5-5v5 libdazzle-1.0-0
  libe-book-0.1-1 libedataserverui-1.2-2 libeot0 libepubgen-0.1-1
  libetonyek-0.1-1 libevent-2.1-6 libexiv2-14 libfreerdp-client2-2
  libfreerdp2-2 libgc1c2 libgee-0.8-2 libgexiv2-2 libgom-1.0-0 libgpgmepp6
  libgpod-common libgpod4 liblangtag-common liblangtag1 liblirc-client0
  liblua5.3-0 libmediaart-2.0-0 libmng2 libmspub-0.1-1 libodfgen-0.1-1
  libqqwing2v5 libqt4-dbus libqt4-declarative libqt4-network libqt4-script
  libqt4-sql libqt4-sql-mysql libqt4-xml libqt4-xmlpatterns libqtcore4
  libqtdbus4 libqtgui4 librevenge-0.0-0 libsgutils2-2 libssh-4 libvncclient1
  libwinpr2-2 libxapian30 libxmlsec1-nss linux-hwe-5.4-headers-5.4.0-110
  linux-hwe-5.4-headers-5.4.0-113 linux-hwe-5.4-headers-5.4.0-117
  linux-hwe-5.4-headers-5.4.0-120 linux-hwe-5.4-headers-5.4.0-121
  linux-hwe-5.4-headers-5.4.0-122 linux-hwe-5.4-headers-5.4.0-124
  linux-hwe-5.4-headers-5.4.0-125 linux-hwe-5.4-headers-5.4.0-126
  linux-hwe-5.4-headers-5.4.0-128 linux-hwe-5.4-headers-5.4.0-131
  linux-hwe-5.4-headers-5.4.0-132 linux-hwe-5.4-headers-5.4.0-84 lp-solve
  media-player-info qdbus qt-at-spi qtcore4-l10n syslinux syslinux-common
  syslinux-legacy usb-creator-common
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

mrcjw, 2023-01-16:
“Could NOT find Boost (missing: python27) (found version "1.65.1")”，I don't know why the compiler still reports this error

mrcjw, 2023-01-16:
Is this the reason for the version of python 3.0 or higher in my computer

rhaschke, 2023-01-16:
ROS Melodic only runs with python2. Thus, if your python interpreter defaults to version 3, this might be the root cause.
You cannot (easily) combine virtualenv with ROS as well. So, return to the system default!