moveit/moveit #3310: cannot compile from source

lianghongzhuo, 2023-01-19:
### Description

Overview of your issue here.
Compile from source, ruckig is also from source master branch.
moveit_core compile success, moveit_ros_planning failed.

```
Errors     << moveit_ros_planning:make src/logs/moveit_ros_planning/build.make.010.log                                            
In file included from /src/moveit/moveit/moveit_ros/planning/planning_request_adapter_plugins/src/add_ruckig_traj_smoothing.cpp:37:
/src/moveit/moveit/moveit_core/trajectory_processing/include/moveit/trajectory_processing/ruckig_traj_smoothing.h:41:10: fatal error: ruckig/ruckig.hpp: No such file or directory
   41 | #include <ruckig/ruckig.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
```

### Your environment
* ROS Distro: noetic
* OS Version: e.g. Ubuntu 22.04
* Source master branch

rhaschke, 2023-01-20:
Install all (new) dependencies via `rosdep`!

lianghongzhuo, 2023-01-22:
The problem is that I want to install all ros packages from source. Can we just put ruckig into a ros workspace and compile it? Or there needs some modification?

rhaschke, 2023-01-23:
If you want to build [ruckig](https://github.com/pantor/ruckig.git) from source as well (why?), obviously you need to include the source package into your catkin workspace too. Then it will be build as usual.

lianghongzhuo, 2023-01-23:
because there is no binary package available in ubuntu22.04

v4hn, 2023-01-23:
We are debugging this issue right now and it's most likely a bug in the moveit cmake configs that might only trigger on Ubuntu 22.04. (https://github.com/v4hn/moveit/tree/pr-master-fix-ruckig-with-catkin for the unverified beta patch).

PR #3312 (2023-01-23): fix ruckig dependency with catkin
catkin was not made for cmake target dependencies.

We do the same with fcl right above, but when porting/merging ruckig nobody cared...

FTR: ruckig does not have any `INTERFACE_LINK_LIBRARIES` and adding them for good measures in case it changes sets the variable to `ruckig_INTERFACE_LINK_LIBRARIES-NOT-FOUND` (breaking the build).

Fixes #3310 .
closed 2023-01-23 by v4hn