moveit/moveit #3318: RViz Rendering Issues since last update

julianmueller, 2023-01-25:
Hi,
since the latest update the "MotionPlanning" and "PlanningScene" Plugins for Rviz behave strange. I stumbled upon the following issues:

- A visualisation bug for STL files that are loaded as collision objects to the scene (as mesh file or from a scene file), where the normals are completely messed with models that worked fine before the update. The models where checked with external tools for watertightness and correct normals. (see in the image how the lower shelf boards are infornt of the upper ones - which is not how the are in real life...) I managed to find out, that this only happens, when the scene has some alpha transparency.
[image]
- A different behavior for planning scene collision objects: when they are moved with the interactive marker (also when added or removed), their position is not updated until the changes are published. This might have a reason, but I find it misleading. A better way would be to have a color for the current active planning scene and another color for "ghost" collision objects that are currently moved with the Rviz overlay.
- Maybe rather a feature request: have a bounding box and coordinate origin shown for every selected collision object in the scene - this would be easier for debugging purposes!

Thats it so far, thanks for the updates on the cone and plane though!

rhaschke, 2023-01-26:
As this seems to be a regression, it would be very helpful to know from which old to which new version you updated.
Please provide your .stl file as well for testing.
Local planning scene changes in rviz need to be published explicitly. This was partially true in the past and is consistently handled now for all kinds of changes.

rhaschke, 2023-01-26:
> I managed to find out, that this only happens when the scene has some alpha transparency.

Did you verify that this bug didn't exist for alpha transparency in the old version as well?
As we didn't change mesh parsing or visualization code recently, I don't think it is a new problem.