moveit/moveit #3322: MoveIt Servo Publishes zeros for joint positions

autonomousTurtle, 2023-01-30:
Hello! 

I am working to set up the realtime servo package for my Kinova Gen3 robot arm. The Kinova Gen3 [does not have a joint_group_position_controller but does use individual joint_position_controllers](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_control/arms/gen3/7dof/config/joint_position_controllers.yaml). I am using the published commands from the servo_server and re-publishing them to each individual joint_position_controller which seems to be working alright. The issue is currently that the output of the servo package is all 0's regardless of the starting position of the robot or the published twist command. Does anyone have any suggestions or prior learnings why this is happening? Is it robot arm specific?

Setup: 7DOF Kinova Gen3 Arm in Gazebo, Ros 1 Noetic, ubuntu 20

`kinova_servo_config.yaml`

```###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: true # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.6  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.3 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.01
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

## Properties of outgoing commands
publish_period: 0.008  # 1/Nominal publish rate [seconds]
low_latency_mode: false  # Set this to true to publish as soon as an incoming Twist command is received (publish_period is ignored)

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController or JointGroupPositionController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: std_msgs/Float64MultiArray
#command_out_type: trajectory_msgs/JointTrajectory

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

## MoveIt properties
move_group_name:  arm  # Often 'manipulator' or 'arm'
planning_frame: base_link  # The MoveIt planning frame. Often 'base_link' or 'world'

## Other frames
ee_frame_name: tool_frame  # The name of the end effector link, used to return the EE pose
robot_link_command_frame:  world  # commands must be given in the frame of a robot link. Usually either the base or end effector

## Stopping behaviour
incoming_command_timeout:  0.1  # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 4

## Configure handling of singularities and joint limits
lower_singularity_threshold:  17  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

## Topic names
cartesian_command_in_topic: delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: delta_joint_cmds # Topic for incoming joint angle commands
joint_topic: joint_states
status_topic: status # Publish status to this topic
command_out_topic: /my_gen3/servo_command #a
#command_out_topic: /my_gen3/joint_group_position_controller/command # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 50 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available:
# "threshold_distance" begins slowing down when nearer than a specified distance. Good if you want to tune collision thresholds manually.
[... 146 lines cut ...]
Arm position regardless of what twist_command is sent to the servo node:
[image]

Published command:

`rostopic pub -r 100 /my_gen3/servo_server/delta_twist_cmds geometry_msgs/TwistStamped "header: auto 
twist:
  linear:
    x: 0.0
    y: 0.0
    z: -0.3
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
`

Resulting published servo position command on topic /my_gen3/servo_command: 
` layout: 
  dim: []
  data_offset: 0
data: [0.005554816272679201, -0.0005437459532294251, -0.002998160079626163, 0.0019001863823941674, -0.004841051418582366, 0.0009435981206462135, 12.78951388262439]
`

Thank you!

AndyZe, 2023-01-30:
I would check what the `/joint_states` topic says. `ros2 topic echo /joint_states`. Is it all-zeros? Because that is what Servo listens to. Please report back.

And it looks like you are running a simulation? Maybe you haven't set the initial joint positions in ros2_control. If you don't specify the initial joint positions, then all-zero positions are assumed

autonomousTurtle, 2023-01-30:
Thanks for the quick reply! 

The /servo_command topic echo is: 

```
layout: 
  dim: []
  data_offset: 0
data: [0.005554816272679201, -0.0005437459532294251, -0.002998160079626163, 0.0019001863823941674, -0.004841051418582366, 0.0009435981206462135, 12.78951388262439]
```

the /joint_states topic echo is : 
```
position: [-0.0019328244755918433, -0.0008636491528672963, 0.008954387712051481, 0.0013439824318384197, -0.004817023491765049, -0.0021026502289842597, 12.790975637770668]
velocity: [-12.037213179066596, -0.43831707883244436, 24.01821299397806, 2.5639526083482598, -16.369953971187726, -4.710592092747538, 2.821806987185937]
effort: [-39.0, 20.528012788090244, 39.0, 39.0, -9.0, -9.0, 0.13850501241989832]
---
```
Joint positions match the published joint positions from the servo_command topic. The initial joint positions were set [here](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_gazebo/launch/robot_launch.launch#:~:text=%3Cnode%20name%3D%22urdf_spawner%22%20pkg%3D%22gazebo_ros%22%20type%3D%22spawn_model%22%20respawn%3D%22false%22%20output%3D%22screen%22%20if%3D%22%24(arg%20is7dof))

In case it makes a difference, this is in ROS 1 and all of the files are launched under the namespace my_gen3. Does the joint_state topic in the .yaml need a /ns/ in front of it? 

Thanks!

autonomousTurtle, 2023-01-30:
Okay thanks, I'll dig in. Would you please be able to point me to a proper implementation of setting the initial joints? 

It appears that the servo_server node is subscribe to the /joint_states topic
[image]

Thanks!

autonomousTurtle, 2023-01-30:
Seems to be the same issue on the real robot (RVIZ shown but it is also connected to the real robot)- the only difference is this is using publish_joint_velocites instead of position

From `kinova_servo_config.yaml` 

```# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: false
publish_joint_velocities: true
publish_joint_accelerations: false
```

[image]

Thanks!