moveit/moveit #3324: What is a good way to constraint robot workspace?

genkv, 2023-01-31:
Hi, I'm not from robotics major so forgive me if "workspace" is the accurate term to use here.

Say I have a panda robot arm installed on a table (a platform). I don't want the trajectory planning to ever let any link to reach any space below the table, because it would have hit the table in the first place.

I can place a CollisionObject as the table, so the motion planning algorithm would try to avoid the potential collision. But I wonder if there's a more proper way to define the "workspace" of a robot? For example, a 3D bounding box that all links of the robot should not reach outside of it when planning trajectory. Thank you.

genkv, 2023-01-31:
As far as I understand, setWorkspace is not for this purpose. It's only meant to be used to bound mobile robots like turtlebot. Stationary robot arms have revolute joints which don't apply the workspace limits set by setWorkspace.

rhaschke, 2023-01-31:
Short answer: Using a CollisionObject to constrain the robot is the right way to go.

The workspace parameters you mentioned are used for sampling in Cartesian space (if you are using a Cartesian-space planner). Normally, sampling-based planners sample in the joint space.

genkv, 2023-02-01:
> Short answer: Using a CollisionObject to constrain the robot is the right way to go.
> 
> The workspace parameters you mentioned are used for sampling in Cartesian space (if you are using a Cartesian-space planner). Normally, sampling-based planners sample in the joint space.

Thanks for the reply. I'm using MoveIt Noetic. Just to confirm, does this mean that if I use the move group interface to plan a path in Cartesian space, combined with a plane constraint:

```
geometry_msgs::Pose target_pose = /... some 6D coordinate in space .../
move_group_interface.setPoseTarget(target_pose);
move_group_interface.setPathConstraints(plane_constraints);
move_group_interface.setWorkspace( /...workspace parameters.../ );
move_group_interface.plan(plan);
```

The move group will honor the setWorkspace on panda robot?

genkv, 2023-02-01:
> Short answer: Using a CollisionObject to constrain the robot is the right way to go.
> 
> The workspace parameters you mentioned are used for sampling in Cartesian space (if you are using a Cartesian-space planner). Normally, sampling-based planners sample in the joint space.

Thank you. I will close this issue.

closed 2023-02-01 by genkv