moveit/moveit #3325: set_goal_orientation_tolerance makes path planning fail

cschindlbeck, 2023-01-31:
### Description

Setting set_goal_orientation via moveit_commander makes path planning fail.
Without

```
self.group.set_goal_orientation_tolerance(np.deg2rad(30))
```

path planing works, setting this param makes it fail

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Tested on both, currently source
* If binary, which release version?
* If source, which branch? https://github.com/ros-planning/moveit/commit/f606813e918281478f8c468ed72a81d18f2bb1bd

### Steps to reproduce

Repo with workspace and docker instructions to reproduce the error can be found here:

https://github.com/cschindlbeck/five_dof_ws

### Expected behaviour

```
[ INFO] [1675164983.696811725]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1675164983.704115456]: Planner configuration 'complete[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1675164983.706058506]: complete/complete[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1675164983.828675201]: complete/complete[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1675164983.828761341]: Solution found in 0.123027 seconds
[ INFO] [1675164983.854366235]: SimpleSetup: Path simplification took 0.002769 seconds and changed from 4 to 2 states
[ INFO] [1675164983.895624281]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1675164983.895810248]: Planning attempt 1 of at most 1
[ INFO] [1675164983.898481174]: Planner configuration 'complete[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1675164983.898681640]: complete/complete[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1675164984.060654681]: complete/complete[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1675164984.060700767]: Solution found in 0.162092 seconds
[ INFO] [1675164984.081674277]: SimpleSetup: Path simplification took 0.000966 seconds and changed from 3 to 2 states
[ INFO] [1675164984.095841903]: Fake execution of trajectory
```

### Actual behaviour

```
[ INFO] [1675164780.656631350]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1675164780.664081891]: Planner configuration 'complete[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1675164780.665911520]: complete/complete[RRTConnect]: Starting planning with 1 states already in datastructure
[ERROR] [1675164785.665606790]: complete/complete[RRTConnect]: Unable to sample any valid states for goal tree
[ INFO] [1675164785.665681306]: complete/complete[RRTConnect]: Created 1 states (1 start + 0 goal)
[ INFO] [1675164785.665744262]: No solution found after 5.000203 seconds
[ WARN] [1675164785.665814559]: Timed out
[ INFO] [1675164785.683101088]: Unable to solve the planning problem
[ WARN] [1675164785.683374260]: Fail: ABORTED: TIMED_OUT
[ERROR] [1675164785.683622]: Path planning failed with error code val: -6 in 6.9472304124674e-310 seconds
```

hollydinkel, 2024-08-02:
ROS Distro: Noetic
OS: Ubuntu 20.04

I am also having the issue described above. [**This**](https://github.com/RMDLO/abb_dual_arm) is my moveit control and motion planning package. I have this issue when I implement the orientation constraint as follows:

```bash
  def execute(self, pose_goal):
      orientation_constraint = OrientationConstraint()
      orientation_constraint.header.frame_id = "base_link_m"
      orientation_constraint.link_name = "gripper_tcp_m"
      orientation_constraint.orientation = pose_goal.orientation
      orientation_constraint.absolute_x_axis_tolerance = 0.1
      orientation_constraint.absolute_y_axis_tolerance = 0.1
      orientation_constraint.absolute_z_axis_tolerance = 0.1
      orientation_constraint.weight = 1.0
      constraints = Constraints()
      constraints.orientation_constraints.append(orientation_constraint)
      self.group.set_path_constraints(constraints)

      waypoints = []
      waypoints.append(self.group.get_current_pose().pose)
      waypoints.append(pose_goal)
      (plan, fraction) = self.group.compute_cartesian_path(waypoints, 0.01, 0.0, avoid_collisions=True)

      joint_state = JointState()
      joint_state.header.stamp = rospy.Time.now()
      joint_state.name = joint_names
      joint_state.position = self.group.get_current_joint_values()

      display_trajectory = DisplayTrajectory()
      display_trajectory.model_id = group_name
      display_trajectory.trajectory.append(plan)
      display_trajectory.trajectory_start.joint_state = joint_state

      self.group.set_pose_target(pose_goal)
      success = self.group.go(wait=True)
      self.group.stop()
      self.group.clear_pose_targets()
      return success
```

Please find a video of the timeout behavior below.

https://github.com/user-attachments/assets/dc60cb5f-bd06-462c-97bc-34d627e4a5c4

rhaschke, 2024-08-12:
@cschindlbeck: Obviously, a 5-dof arm cannot reach all possible 6-dof poses in space. If you increase goal orientation tolerance (from the default 1e-3 to 0.52), the sampling range for orientation increases and the probability for IK to find a solution for the sampled pose decreases in your case: According to the error message, the goal sampler wasn't able to find any valid state.

@hollydinkel: Your issue is different. First, you use 6-dof arms. Second, you don't call `set_goal_orientation_tolerance()`. Instead, you are using an orientation constraint as a path constraint. I'm curious to see your console log (please paste text instead of using an image or video). The video doesn't play for me.

hollydinkel, 2024-08-12:
@rhaschke Thanks for your response! the console log just says [INFO] ABORTED: TIMED OUT (shown in the video). I also tested playing this video in Google Chrome and Firefox and on my iPhone web browser (iOS Safari) - maybe try a different browser or device? You have to click the arrow next to "timeout.mp4" for the video to display.

As an additional update, I increased the "absolute_x_axis_tolerance"/"absolute_y_axis_tolerance"/"absolute_z_axis_tolerance" on the constraint (from 0.1 to 0.75 - good enough for me) to avoid the timeout. Video of this result is shown below.

https://github.com/user-attachments/assets/02267e3d-843c-4645-aaa8-e3b53ad06481

However, for about 50% of the computed paths, I get this weird behavior where halfway through the path the robot stops mid-way and rotates 360 degrees. In the context of my application (manipulating rope), this ruins the demo.

https://github.com/user-attachments/assets/5109e383-5d21-4a73-95cc-f3fcfaf5eb4c

To address this, I added a joint constraint on the end effector joint to try to fix this issue:

```bash
joint_constraint = JointConstraint()
joint_constraint.joint_name = "joint_7_m"
joint_constraint.position = 0.0  # Desired joint position in radians
joint_constraint.tolerance_above = 0.2  # Upper tolerance
joint_constraint.tolerance_below = -0.2 # Lower tolerance
joint_constraint.weight = 1.0
constraints.joint_constraints.append(joint_constraint)
```
However, the path that the end effector follows then is also semi-circular in the x-y plane which is also undesirable (the x position of the start and goal transforms is the same). I want a straight line path between a start pose and goal pose without any gripper z-axis rotation - it seems like it should be simple enough for a robot with few workspace constraints/obstacles. Any ideas on why this happening or how to fix would be greatly appreciated! Thank you!

https://github.com/user-attachments/assets/bfd04c24-f0a6-4651-8dcd-ff23b902c7be

rhaschke, 2024-08-12:
@hollydinkel: Your issue might be addressed by recent PR #3615. Please try a build from source.

hollydinkel, 2024-08-12:
@rhaschke thanks for the quick reply! Now when I only implement the orientation constraint, i.e.,

```bash
orientation_constraint = OrientationConstraint()
orientation_constraint.header.frame_id = "base_link_m"
orientation_constraint.link_name = self.group.get_end_effector_link()
orientation_constraint.orientation = pose_goal.orientation
orientation_constraint.absolute_x_axis_tolerance = 0.6
orientation_constraint.absolute_y_axis_tolerance = 0.6
orientation_constraint.absolute_z_axis_tolerance = 0.6
orientation_constraint.weight = 10.0
constraints.orientation_constraints.append(orientation_constraint)
```

I get the behavior I saw when I implemented the orientation + joint constraint, where there is no rotation about the z axis midway through the trajectory, but the path from start to goal pose is semi-circular in the x-y plane (similar behavior to the last two videos shown above). I need a path that does not rotate the gripper and is a mostly-straight line in the x-y plane. The arm also frequently get stuck at the pose 'idx_reset_3' and is unable to return to the default position (all joints zero except link_6 which is 90 degrees).

```bash
name: ''
joint_constraints: []
position_constraints: []
orientation_constraints: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "base_link_m"
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.7071067811865475
      w: 0.7071067811865475
    link_name: "gripper_tcp_m"
    absolute_x_axis_tolerance: 0.6
    absolute_y_axis_tolerance: 0.6
    absolute_z_axis_tolerance: 0.6
    parameterization: 0
    weight: 10.0
visibility_constraints: []
[ INFO] [1723488851.817554452]: ABORTED: TIMED_OUT
Replanning...
Slide?  True
name: ''
joint_constraints: []
position_constraints: []
orientation_constraints: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "base_link_m"
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.7071067811865475
      w: 0.7071067811865475
    link_name: "gripper_tcp_m"
    absolute_x_axis_tolerance: 0.6
    absolute_y_axis_tolerance: 0.6
    absolute_z_axis_tolerance: 0.6
    parameterization: 0
    weight: 10.0
visibility_constraints: []
[ INFO] [1723488866.056159004]: ABORTED: TIMED_OUT
Executed?  False
```

rhaschke, 2024-08-13:
If you need straight-line Cartesian paths, you need to go with `compute_cartesian_path`. All sampling-based planners essentially interpolate in joint-space, creating those semi-circular paths.

hollydinkel, 2024-08-13:
@rhaschke I am using `compute_cartesian_path` - this is my execution function:

```bash
def execute(self):
      orientation_constraint = OrientationConstraint()
      orientation_constraint.header.frame_id = "base_link_m"
      orientation_constraint.link_name = self.group.get_end_effector_link()
      orientation_constraint.orientation = pose_goal.orientation
      orientation_constraint.absolute_x_axis_tolerance = 0.5
      orientation_constraint.absolute_y_axis_tolerance = 0.5
      orientation_constraint.absolute_z_axis_tolerance = 0.5
      orientation_constraint.weight = 10.0
      constraints.orientation_constraints.append(orientation_constraint)
      self.group.set_path_constraints(constraints)
              
      waypoints = []
      waypoints.append(self.group.get_current_pose().pose)
      waypoints.append(pose_goal)
      (plan, fraction) = self.group.compute_cartesian_path(waypoints, 0.01, 0.0, avoid_collisions=True)
      
      joint_state = JointState()
      joint_state.header.stamp = rospy.Time.now()
      joint_state.name = joint_names
      joint_state.position = self.group.get_current_joint_values()
      
      display_trajectory = DisplayTrajectory()
      display_trajectory.model_id = group_name
      display_trajectory.trajectory.append(plan)
      display_trajectory.trajectory_start.joint_state = joint_state
      
      self.group.set_pose_target(pose_goal)
      success = self.group.go(wait=True)
      self.group.stop()
      self.group.clear_pose_targets()
      if success == False:
          print("Replanning...")
          self.execute(pose_goal)
      return success
```

Any idea what is wrong with this? Thanks again for your help!

rhaschke, 2024-08-13:
@cschindlbeck: Increasing planning time can be a solution. In your case, I suggest switching to rejection-sampling in joint-space. To this end, use the parameter `enforce_joint_model_state_space` as explained [here](https://moveit.github.io/moveit_tutorials/doc/ompl_interface/ompl_interface_tutorial.html#enforce-planning-in-joint-space).

@hollydinkel: You are not using the plan generated by `compute_cartesian_path`, but planning again with OMPL:

```python
# generate a straight-line Cartesian path along the given waypoints stopping before a collision (unused)
(plan, fraction) = self.group.compute_cartesian_path(waypoints, 0.01, 0.0, avoid_collisions=True)
      
# set a Cartesian target for sampling-based planning
self.group.set_pose_target(pose_goal)
# plan and execute towards the given target (pose_goal) using the configured planning pipeline (OMPL)
self.group.go(wait=True)
# this should fail with a type error (execute is expecting a plan)
self.execute(pose_goal)
```
You should use something like:
```python
(plan, fraction) = self.group.compute_cartesian_path(waypoints, 0.01, 0.0, avoid_collisions=True)
self.execute(plan)
```

closed 2024-08-13 by rhaschke