moveit/moveit #3327: Legacy and alternative of computeCartesianPath

genkv, 2023-02-01:
Version: MoveIt Noetic
System: Ubuntu 20.04
Robot: UR5

I read this post from PickNik [guide-to-cartesian-planners-in-moveit](https://picknik.ai/cartesian%20planners/moveit/motion%20planning/2021/01/07/guide-to-cartesian-planners-in-moveit.html), which states the following:

> For historical reasons, the MoveIt MoveGroup interface exposes a [computeCartesianPath()](https://ros-planning.github.io/moveit_tutorials/doc/move_group_interface/move_group_interface_tutorial.html) API that uses the [default Cartesian Interpolator](https://github.com/ros-planning/moveit/blob/master/moveit_core/robot_state/include/moveit/robot_state/cartesian_interpolator.h) functionality in MoveIt.
> ...
> While as of this writing this is still the default behavior, **we recommend avoiding this Cartesian planner**.

My aim is to plan a trajectory from a series of waypoints for the robot to go through, one-by-one. The end effector must remain a fix orientation throughout the entire path, therefore this is a Cartesian path. After reading some tutorial and documentation, it comes natural to use this API  `computeCartesianPath(waypoints, ...)`. But since this article suggests that `computeCartesianPath` is implicitly deprecated, I'm looking for some alternatives.

This tutorial from MoveIt 2 [Using OMPL Constrained Planning](https://moveit.picknik.ai/main/doc/how_to_guides/using_ompl_constrained_planning/ompl_constrained_planning.html) is probably the most up-to-date solution. However, I have some questions:

1. Is this feature currently only available on MoveIt 2? 
I wrote similar code for my MoveIt 1 Noetic.
```
moveit_msgs::PositionConstraint box_constraint;
box_constraint.header.frame_id = move_group_interface.getPoseReferenceFrame();
box_constraint.link_name = move_group_interface.getEndEffectorLink();
shape_msgs::SolidPrimitive box;
box.type = shape_msgs::SolidPrimitive::BOX;
box.dimensions = { 1.0, 0.0005, 1.0 };
box_constraint.constraint_region.primitives.emplace_back(box);

geometry_msgs::Pose box_pose;
// ... some code to give box_pose a desired coordinate...

box_constraint.constraint_region.primitive_poses.emplace_back(box_pose);
box_constraint.weight = 1.0;

moveit_msgs::Constraints plane_constraints;
plane_constraints.position_constraints.emplace_back(box_constraint);
plane_constraints.name = "use_equality_constraints";

// Plan the trajectory
move_group_interface.setPoseTarget(target_pose);
move_group_interface.setPlanningTime(10.0);
move_group_interface.setPathConstraints(plane_constraints);
move_group_interface.plan(plan);
```

And this code works on my Noetic setup, which surprised me. Because I didn't do anything like the below code from the tutorial. How come it still works?
```
moveit_config.planning_pipelines["ompl"]["panda_arm"]["enforce_constrained_state_space"] = True
moveit_config.planning_pipelines["ompl"]["panda_arm"]["projection_evaluator"] = "joints(panda_joint1,panda_joint2)"
moveit_config.planning_pipelines["ompl"]["panda_arm_hand"]["enforce_constrained_state_space"] = True
moveit_config.planning_pipelines["ompl"]["panda_arm_hand"]["projection_evaluator"] = "joints(panda_joint1,panda_joint2)"
```
Kindly asking the author of the tutorial @JeroenDM 

2. The `computeCartesianPath` basically finds a straight line from current pose to target pose. This behavior is implied and self-explanatory by its name. Now if I switch to using the OMPL Constraint Planner, do I need to manually set a straight line constraint between the two points to achieve the same behavior?

3. Multiple way points
[image]
This table shows that OMPL Constrained Planning is not suitable for multiple way points. But Pilz Industrial Motion Planner and some other planners do. So is it better to use Pilz Industrial Motion Planner as an alternative to `computeCartesianPath`, since the OMPL Constraint Planning doesn't support multiple waypoints?

Thank you for reading my questions to this point. I know these questions may seem very basic, but as a newbie to Moveit I'd appreciate any help.

AndyZe, 2023-02-01:
Check out the "Sequence of multiple segments" section of the tutorial for the Pilz planner.

https://ros-planning.github.io/moveit_tutorials/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#sequence-of-multiple-segments

genkv, 2023-02-01:
> Check out the "Sequence of multiple segments" section of the tutorial for the Pilz planner.
> 
> https://ros-planning.github.io/moveit_tutorials/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#sequence-of-multiple-segments

Thanks for the reply. I should've made myself clearer that I also need to find Cartesian path with only one target pose. This goes back to my first 2 questions, especially the second question.

dcconner, 2023-05-02:
I am also interested in the planned path forward for this (hah!).  Is there going to be a planner agnostic version of this that will allow us to call something like computeCartesianPath generically, just not from the robot state interface?  

Will this be part of the standard MoveIt2 API, or only on planner-by-planner basis?

rhaschke, 2023-05-02:
First of all, `computeCartesianPath` is _not_ deprecated. Rather, it is still the recommended solution if you need to plan a straight-line Cartesian path to a target pose. However, note that straight-line motion generation is inherently incompatible with obstacle avoidance: `computeCartesianPath` does not plan around obstacles (because that would violate the straight-line motion). It just stops in front of the obstacle, returning a fraction of the achieved motion less than 100%.
Thus, if you want to plan toward a Cartesian pose _while avoiding obstacles_ (and not moving along a straight line)  `computeCartesianPath` is not your choice. Instead, use OMPL constrained-based planning, which is also implemented in MoveIt 1.
The Pilz planner is very similar to `computeCartesianPath`: it generates straight-line (or circular) Cartesian motions while checking but not avoiding obstacles. In contrast to `computeCartesianPath`, it allows for multiple Cartesian path segments and smooth interpolation between them.

closed 2023-05-02 by rhaschke