moveit/moveit #3329: AITstar planner stuck in infinite loop

alaflaquiere, 2023-02-02:
### My environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* From source
* master branch, commit 8eb1d1010b72a71bdf27aafcaff4a48a806047d6

### Description

I am trying to add the AITstar planner from OMPL-1.6.0 into Moveit, so that i can easily use it from the MotionPlanning module in RViz.
Following some resources I could find online, I did the following steps:
- installed ompl-1.6.0 using the installation script ([source](https://ompl.kavrakilab.org/installation.html))
- cloned MoveIt repo and dependencies, commit `8eb1d1010b72a71bdf27aafcaff4a48a806047d6` ([source](https://moveit.ros.org/install/source/))
- modified `planning_context_manager.cpp` to include AITstar: `#include <ompl/geometric/planners/informedtrees/AITstar.h>` and `registerPlannerAllocatorHelper<og::AITstar>("geometric::AITstar");`
- used the MoveIt assistant to create a config package for my robot, and updated `ompl_planning.yaml` to include AITstar:
  - `planner_configs:`
    - `...`
    - `AITstar:`
      - `type: geometric::AITstar`
      - `use_k_nearest: 0`
      - `rewire_factor: 1.0`
      - `samples_per_batch: 100`
      - `use_graph_pruning: 0`
      - `find_approximate_solutions: 1`
  - `robot:`
    - `planner_configs:`
      - `...`
      - `AITstar`
- catkin built everything

As a result, I can see AITstar with the proper parameters in the Context tab of the MotionPlanning module of RViz...
[image]

... and the planner sometimes get to work properly, like in the following example: 
[image]

But in a large number of cases, the planning process seems to simply get stuck, and nothing is ever returned, even after the planning time threshold has been exceeded; like in the following example: 
[image]

As far as I can tell, changing the AITstar parameters defined in `ompl_planning.yaml` have no effect on this behavior. Similarly, the problem also exists with the AITstar implementation from OMPL-1.5.0.

I haven't been able to figure out more about the nature of the problem for now. Would you any idea where I should look?

simonschmeisser, 2023-02-02:
There seems to be some problem with sampling. MoveIt provides some adapted samplers that are feed into OMPL to integrate the two. Maybe this needs special treatment for AIT*?

Please open a (draft) PR with your changes, then it's easier to discuss and test. Thanks a lot for your effort in enhancing planners!

alaflaquiere, 2023-02-02:
Thanks for the feedback. I created the following draft PR #3330.

To reproduce the behavior, you would need to create a new moveit config for  a robot using the setup assistant, and then try to plan with the AITstar planner from RViz. If the planned motion is complex enough, my estimaon is that the planner gets stuck ~50% of the time.

alaflaquiere, 2023-02-06:
Additional information: the original authors of AITstar also imagine it might be an issue with sampling.

Would you have any pointer to where I should start looking?

haryngod, 2023-02-15:
I had the same problem with a complex environment.
I guess this part is the root cause of the infinite loop.
I made PR on ompl repository. https://github.com/ompl/ompl/pull/990.

rowanborder, 2023-02-16:
I experienced the same issue when interfacing with AIT* via the plugin-based framework from KavrakiLab:- https://github.com/KavrakiLab/moveit_ompl_planning_interface 

In this case the hanging issue was caused by the caching of the StateValidityChecker (https://github.com/KavrakiLab/moveit_ompl_planning_interface/blob/52db1c475e845e5b55d4fceb9c0102e7120acf02/ompl_interface/include/moveit/ompl_interface/ompl_planning_context.h#L73). Disabling this caching resolved the issue.

alaflaquiere, 2023-02-20:
Thank you @haryngod !
I did a few tests, and it looks like the change from [ompl/ompl#990](https://github.com/ompl/ompl/pull/990) solved my issue.  :pray:

closed 2023-02-20 by alaflaquiere