moveit/moveit #3333: basic_servo_test.test RLException: roslaunch file contains multiple nodes named [/joint_state_publisher]

Akumar201, 2023-02-03:
Hi, I am trying to run [basic_servo_tests.test](https://github.com/ros-planning/moveit/tree/master/moveit_ros/moveit_servo/test) with real xarm and modified the basic_servo_tests.test file accordingly with the configuration file. 

roslaunch basic_servo_tests.test
```

<?xml version="1.0"?>
<launch>
  <!-- Load URDF, SRDF -->
  <include file="$(find xarm6_moveit_config)/launch/realMove_exec.launch" >
  </include>

  <!-- Initial joint positions -->
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <rosparam command="load" file="$(find moveit_servo)/test/config/initial_position.yaml" />
    <param name="publish_frequency" type="double" value="50.0"/>
  </node>

  <test pkg="moveit_servo" type="basic_servo_tests" test-name="basic_servo_tests" time-limit="60" args="">
    <rosparam command="load" file="$(find moveit_servo)/test/config/servo_settings_low_latency.yaml"/>
  </test>
</launch>
```
If I am running the following launch file, I am getting the following error

```
... logging to /home/akumar/.ros/log/c1e7ddae-a3f0-11ed-9bf6-2be3be7854f6/roslaunch-abc-ThinkPad-P17-Gen-2i-37953.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
RLException: roslaunch file contains multiple nodes named [/joint_state_publisher].
Please check all <node> 'name' attributes to make sure they are unique.
Also check that $(anon id) use different ids.
The traceback for the exception was written to the log file
```
Correct me I am wrong, I guess the problem is with the below lines 
```
  <!-- Initial joint positions -->
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <rosparam command="load" file="$(find moveit_servo)/test/config/initial_position.yaml" />
    <param name="publish_frequency" type="double" value="50.0"/>
  </node>

```
I also tried to find the joint_state_publisher in my directory but was not able to. Any help will be much appriciated

AndyZe, 2023-02-03:
Well, does this other launch file have a `joint_state_publisher`? You only need one of them.

`<include file="$(find xarm6_moveit_config)/launch/realMove_exec.launch" >`

Akumar201, 2023-02-03:
This is the launch file 

```
<launch>

  <arg name="robot_ip" default="192.168.1.xxx" />
  <arg name="report_type" default="dev" />	
  <!-- load the default move_group planner (not xarm_simple_planner) -->
  <arg name="show_rviz" default="true" />
  <!-- NO_GUI_CTRL means that Rviz configuration is just for display purpose, no (Moveit) control panel loaded -->
  <arg name="no_gui_plan" default="false" />
  <arg name="xarm_hw_ns" default="xarm" />
  <arg name="ext_ns" default="" />
  <arg name="velocity_control" default="false"/>
  <arg name="enforce_limits" default="true" />
  <arg name="baud_checkset" default="true" />
	<arg name="default_gripper_baud" default="2000000" />

  <arg name="add_other_geometry" default="false"/>
  <arg name="geometry_type" default="box"/>
  <arg name="geometry_mass" default="0.1"/>
  <arg name="geometry_height" default="0.1"/>
  <arg name="geometry_radius" default="0.1"/>
  <arg name="geometry_length" default="0.1"/>
  <arg name="geometry_width" default="0.1"/>
  <arg name="geometry_mesh_filename" default=""/>
  <arg name="geometry_mesh_origin_xyz" default="'0 0 0'"/>
  <arg name="geometry_mesh_origin_rpy" default="'0 0 0'"/>
  <arg name="geometry_mesh_tcp_xyz" default="'0 0 0'"/>
  <arg name="geometry_mesh_tcp_rpy" default="'0 0 0'"/>

  <!-- common: bring up node to enable real xArm control -->
  <include file="$(find xarm_bringup)/launch/xarm6_server.launch">
    <arg name="robot_ip" value="$(arg robot_ip)" />
    <arg name="report_type" value="$(arg report_type)" />
    <arg name="use_moveit" value="true" />
    <arg name="ns" value="$(arg xarm_hw_ns)" />
    <arg name="velocity_control" value="$(arg velocity_control)" />
    <arg name="enforce_limits" value="$(arg enforce_limits)" />
    <arg name="baud_checkset" value="$(arg baud_checkset)" />
    <arg name="default_gripper_baud" value="$(arg default_gripper_baud)" />
  </include>

  <!-- load the joint trajectory controller specified in yaml -->
  <rosparam file="$(find xarm_controller)/config/xarm6_traj_controller.yaml" command="load" ns="$(arg xarm_hw_ns)"/>
  <arg name="xarm_traj_controller" value="$(eval 'xarm6_traj_controller_velocity' if arg('velocity_control') else 'xarm6_traj_controller')" />
  <node
    name="traj_controller_spawner"
    pkg="controller_manager"
    type="spawner"
    respawn="false"
    output="screen"
    ns="$(arg xarm_hw_ns)"
    args="$(arg xarm_traj_controller)"/>

  <!-- <remap from="/follow_joint_trajectory" to="xarm/xarm6_traj_controller/follow_joint_trajectory"/> -->
  <remap from="/follow_joint_trajectory" to="$(arg xarm_hw_ns)/$(arg xarm_traj_controller)/follow_joint_trajectory"/>

  <!-- common part -->
  <remap from="/tf" to="$(arg ext_ns)/tf" />
  <remap from="/tf_static" to="$(arg ext_ns)/tf_static" />
  <remap from="/move_group/monitored_planning_scene" to="$(arg ext_ns)/move_group/monitored_planning_scene" />
  <remap from="$(arg ext_ns)/source_list" to="/source_list" />
  <include file="$(find xarm6_moveit_config)/launch/moveit_rviz_common.launch">
    <arg name="jnt_stat_source" value="[$(arg xarm_hw_ns)/joint_states]" />
    <arg name="show_rviz" value="$(arg show_rviz)" />
    <arg name="NO_GUI_CTRL" value="$(arg no_gui_plan)"/>

    <arg name="add_other_geometry" default="$(arg add_other_geometry)"/>
    <arg name="geometry_type" default="$(arg geometry_type)"/>
    <arg name="geometry_mass" default="$(arg geometry_mass)"/>
    <arg name="geometry_height" default="$(arg geometry_height)"/>
    <arg name="geometry_radius" default="$(arg geometry_radius)"/>
    <arg name="geometry_length" default="$(arg geometry_length)"/>
    <arg name="geometry_width" default="$(arg geometry_width)"/>
    <arg name="geometry_mesh_filename" default="$(arg geometry_mesh_filename)"/>
    <arg name="geometry_mesh_origin_xyz" default="$(arg geometry_mesh_origin_xyz)"/>
    <arg name="geometry_mesh_origin_rpy" default="$(arg geometry_mesh_origin_rpy)"/>
    <arg name="geometry_mesh_tcp_xyz" default="$(arg geometry_mesh_tcp_xyz)"/>
    <arg name="geometry_mesh_tcp_rpy" default="$(arg geometry_mesh_tcp_rpy)"/>
  </include>

</launch>
```

Akumar201, 2023-02-03:
@AndyZe Thank you for your reply. I commented out the code below 
```

  <!-- Initial joint positions
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <rosparam command="load" file="$(find moveit_servo)/test/config/initial_position.yaml" />
    <param name="publish_frequency" type="double" value="50.0"/>
  </node> -->

```
and did `roslaunch basic_servo_tests.test ` and got rqt_graph 
It shows the same name 

[image]

So, in case as far I understand the launch file this part of code below , helps to move the arm away from initial singularity position 

```

  <!-- Initial joint positions
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <rosparam command="load" file="$(find moveit_servo)/test/config/initial_position.yaml" />
    <param name="publish_frequency" type="double" value="50.0"/>
  </node> -->

```
So, I can either remove it and run my code or else edit the config file according to my arm. Please correct me if I am wrong. As my arm is still not moving

AndyZe, 2023-02-03:
What you say sounds accurate. So the launching is successful now?

Do you see any errors or warnings in the terminal?

Akumar201, 2023-02-03:
Yes, There is no error, however my arm is not moving. I tried editing the `servo_settings_low_latency.yaml` file accordingly, however my arm is still not moving. I would like to share the log , please suggest me what I am doing wrong. 

`servo_settings_low_latency.yaml`

```
###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: false # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.003  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.006 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.01
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

## Properties of outgoing commands
low_latency_mode: true  # Set this to true to tie the output rate to the input rate
publish_period: 0.01  # 1/Nominal publish rate [seconds]

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController or JointGroupPositionController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: trajectory_msgs/JointTrajectory

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

## MoveIt properties
move_group_name:  xarm6  # Often 'manipulator' or 'arm'
planning_frame: link_base  # The MoveIt planning frame. Often 'panda_link0' or 'world'

## Other frames
ee_frame_name: link_eef  # The name of the end effector link, used to return the EE pose
robot_link_command_frame:  link_base  # commands must be given in the frame of a robot link. Usually either the base or end effector

## Stopping behaviour
incoming_command_timeout:  1  # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 1

## Configure handling of singularities and joint limits
lower_singularity_threshold:  30  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 45 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

## Topic names
cartesian_command_in_topic: servo_server/delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: servo_server/delta_joint_cmds # Topic for incoming joint angle commands
joint_topic:  joint_states
status_topic: servo_server/status # Publish status to this topic
command_out_topic: servo_server/command # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 5 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available:
# "threshold_distance" begins slowing down when nearer than a specified distance. Good if you want to tune collision thresholds manually.
# "stop_distance" stops if a collision is nearer than the worst-case stopping distance and the distance is decreasing. Requires joint acceleration limits
collision_check_type: stop_distance
# Parameters for "threshold_distance"-type collision checking
self_collision_proximity_threshold: 0.01 # Start decelerating when a collision is this far [m]
scene_collision_proximity_threshold: 0.03 # Start decelerating when a collision is this far [m]
[... 418 lines cut ...]
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1675452342.015687484]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1675452342.015697834]: MoveGroup context initialization complete

You can start planning now!

[INFO] [1675452343.517405]: Started controllers: xarm6_traj_controller
[ INFO] [1675452345.170260829]: Loading robot model 'xarm6'...
[ INFO] [1675452345.170297609]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1675452345.213463294]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_akumar_ThinkPad_P17_Gen_2i_41997_2660357324045220049/xarm6/kinematics_solver_attempts' from your configuration.
[ INFO] [1675452345.349136370]: Starting planning scene monitor
[ INFO] [1675452345.351066862]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1675452345.361230154]: Constructing new MoveGroup connection for group 'xarm6' in namespace ''
[ INFO] [1675452346.496773421]: Ready to take commands for planning group xarm6.
`
```

AndyZe, 2023-02-03:
I think this is wrong, it should be the topic of your ros_control controller:  `command_out_topic: servo_server/command # Publish outgoing commands here`

Akumar201, 2023-02-03:
Hi, in my rostopic list I can see the following controllers

```
/xarm/xarm6_traj_controller/command
/xarm/xarm6_traj_controller/follow_joint_trajectory/cancel
/xarm/xarm6_traj_controller/follow_joint_trajectory/feedback
/xarm/xarm6_traj_controller/follow_joint_trajectory/goal
/xarm/xarm6_traj_controller/follow_joint_trajectory/result
/xarm/xarm6_traj_controller/follow_joint_trajectory/status
/xarm/xarm6_traj_controller/state

```
and this is my rqt_graph 

[image]

Please correct me if I am wrong 
Should it be 

command_out_topic: xarm/uf_ros_controller

AndyZe, 2023-02-03:
Probably `/xarm/xarm6_traj_controller/command`

Akumar201, 2023-02-03:
@AndyZe Thank You for your prompt reply. It is still not moving, also in the rqt graph I can't see any node related to basic_servo_tests. I am sorry for troubling you 

```
###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: false # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.003  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.006 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.01
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

## Properties of outgoing commands
low_latency_mode: true  # Set this to true to tie the output rate to the input rate
publish_period: 0.01  # 1/Nominal publish rate [seconds]

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController or JointGroupPositionController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: trajectory_msgs/JointTrajectory

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

## MoveIt properties
move_group_name:  xarm6  # Often 'manipulator' or 'arm'
planning_frame: link_base  # The MoveIt planning frame. Often 'panda_link0' or 'world'

## Other frames
ee_frame_name: link_eef  # The name of the end effector link, used to return the EE pose
robot_link_command_frame:  link_base  # commands must be given in the frame of a robot link. Usually either the base or end effector

## Stopping behaviour
incoming_command_timeout:  1  # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 1

## Configure handling of singularities and joint limits
lower_singularity_threshold:  30  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 45 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

## Topic names
cartesian_command_in_topic: servo_server/delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: servo_server/delta_joint_cmds # Topic for incoming joint angle commands
joint_topic:  joint_states
status_topic: servo_server/status # Publish status to this topic
command_out_topic: /xarm/xarm6_traj_controller/command # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 5 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available:
# "threshold_distance" begins slowing down when nearer than a specified distance. Good if you want to tune collision thresholds manually.
# "stop_distance" stops if a collision is nearer than the worst-case stopping distance and the distance is decreasing. Requires joint acceleration limits
collision_check_type: stop_distance
# Parameters for "threshold_distance"-type collision checking
self_collision_proximity_threshold: 0.01 # Start decelerating when a collision is this far [m]
scene_collision_proximity_threshold: 0.03 # Start decelerating when a collision is this far [m]
# Parameters for "stop_distance"-type collision checking
collision_distance_safety_factor: 1000 # Must be >= 1. A large safety factor is recommended to account for latency
min_allowable_collision_distance: 0.01 # Stop if a collision is closer than this [m]
```

AndyZe, 2023-02-03:
You should rename that file from `basic_servo.test` to `basic_servo.launch` since you're not using it to run tests. Also you should delete this part (the testing part):

```
  <test pkg="moveit_servo" type="basic_servo_tests" test-name="basic_servo_tests" time-limit="60" args="">
    <rosparam command="load" file="$(find moveit_servo)/test/config/servo_settings_low_latency.yaml"/>
  </test>
```

Why did you decide to copy this file, anyway? This one would have been much better to copy:

https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/launch/spacenav_cpp.launch

Obviously delete the spacenav "stuff"

Akumar201, 2023-02-03:
Hi, I decided to go with the [basic_servo_tests.cpp](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/test/basic_servo_tests.cpp) as the code is fairly easy and have few test cases that I can edit later according to my experiments. [spacenav_cpp.launch](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/launch/spacenav_cpp.launch) was not running and giving some error due to sapacenav mouse , so thought it would be a good idea to make a code base example running which can be easily manipulate according to testing scenario. I would like to contribute my code to the community once it will start working. I did the recommended changes like `basic_servo_tests.launch` and launch file as 

```
<?xml version="1.0"?>
<launch>
  <!-- Load URDF, SRDF -->
  <include file="$(find xarm6_moveit_config)/launch/realMove_exec.launch" >
  </include>

  <!-- Initial joint positions
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    <rosparam command="load" file="$(find moveit_servo)/test/config/initial_position.yaml" />
    <param name="publish_frequency" type="double" value="50.0"/>
  </node> -->

  <!-- <test pkg="moveit_servo" type="basic_servo_tests" test-name="basic_servo_tests" time-limit="60" args=""> -->
  <rosparam command="load" file="$(find moveit_servo)/test/config/servo_settings_low_latency.yaml"/>
  <!-- </test> -->
</launch>

However, it's still not moving. 
```

Akumar201, 2023-02-06:
@AndyZe This issue has been resolved, I was missing the command` catkin_make run_tests`. Thank you for the suggestion , I would like to recommend to follow @AndyZe suggestion of using  
[spacenav_cpp.launch](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/launch/spacenav_cpp.launch)
or 
[pose_tracking_example.launch](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/launch/pose_tracking_example.launch)

This issue has been resolved and if the issue persist it can be reopen.

closed 2023-02-06 by Akumar201