moveit/moveit #3338: camera_rgb_optical_frame - Where is it defined?

genkv, 2023-02-07:
I noticed that in the [Perception Pipeline Tutorial](https://ros-planning.github.io/moveit_tutorials/doc/perception_pipeline/perception_pipeline_tutorial.html), we must define and publish a frame called **camera_rgb_optical_frame**. This obviously is the frame referenced by the octomap cubics.

It seems like a hardcoded value because I couldn't find any reference to it in the source code. Does anyone know where is this topic name defined and used?

rhaschke, 2023-02-07:
This frame is usually published by a RGBD camera driver.

genkv, 2023-02-07:
> This frame is usually published by a RGBD camera driver.

When I accidentally replaced this name by something else (for example, camera_rgb_optical_frame_1), the octomap stops showing up in Rviz. So I thought it must been consumed somewhere in the source code. I just couldn't find it. I'm trying to understand how the whole octomap updater works in order to adjust to my need. thank you.