moveit/moveit #3340: real arm not moving using servoing

Akumar201, 2023-02-07:
### Description
Hi, I am trying to move real xarm with the example [pose_tracking_example.cpp ](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/src/cpp_interface_example/pose_tracking_example.cpp). I am trying yo use ROS2 moveit servoing as an example and replicate it in ROS1.

I am not getting any error, I tried comparing the rqt_graph also and it looked fine to me. Refrence to https://github.com/ros-planning/moveit/issues/3335 , 
I am doubtfull as when I read the `pose_tracking_example.cpp` it is only publishing the data to target pose, however when I was using moveit tutorial to move my real arm I used planning as well as `group.go(joint_goal, wait=True)` to move the arm, is this file missing that that's why my real arm is not moving or servo.h takes care of planning and  .go . Please correct me if I am wrong 

 function

```
  for (size_t i = 0; i < 500; ++i)
  {
    // Modify the pose target a little bit each cycle
    // This is a dynamic pose target
    target_pose.pose.position.z += 0.0004;
    target_pose.header.stamp = ros::Time::now();
    target_pose_pub.publish(target_pose);

    loop_rate.sleep();
  }

  // Make sure the tracker is stopped and clean up
  tracker.stopMotion();
  move_to_pose_thread.join();

  return EXIT_SUCCESS;
```

### Your environment
* ROS Distro:Noetic
* OS Version: e.g. Ubuntu 20.04 LTS

### Steps to reproduce
1. ` roslaunch xarm6_server.launch` [launch file ](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_bringup/launch/xarm6_server.launch)
2. `roslaunch realMove_exec.launch `[launch_file](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm6_moveit_config/launch/realMove_exec.launch)
3. `roslaunch pose_tracking_example.launch` [launch_file](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/launch/pose_tracking_example.launch)

### Expected behaviour
Robot should move

### Actual behaviour
It's not moving 

### Doubt

AndyZe, 2023-02-07:
I'm sorry I don't have time to dig through your config files in detail.

> is this file missing that that's why my real arm is not moving or servo.h takes care of planning and .go

^This definitely isn't the problem. Servo is separate from the MoveGroup which it sounds like that example uses.

Akumar201, 2023-02-07:
Please correct me if I am wrong, the potential problem could be in configuration file of my robot. The following steps I am taking is right

1. ` roslaunch xarm6_server.launch` [launch file ](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_bringup/launch/xarm6_server.launch)
2. `roslaunch realMove_exec.launch `[launch_file](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm6_moveit_config/launch/realMove_exec.launch)
3. `roslaunch pose_tracking_example.launch` [launch_file](https://github.com/ros-planning/moveit/blob/master/moveit_ros/moveit_servo/launch/pose_tracking_example.launch)

Is there a possibility that as I am still using default controller joint_state_controller/JointStateController` could be the issue, I have . Please suggest me few things I took the confi file from working example in ros2 by https://github.com/xArm-Developer/xarm_ros2/blob/master/xarm_moveit_servo/config/xarm_moveit_servo_config.yaml

AndyZe, 2023-02-07:
The joint_state_controller/JointStateController just publishes the joint positions on `/joint_states`. If you can do `rostopic echo /joint_states`, then it's fine.

You do need another type of controller running besides the JointStateController.

Akumar201, 2023-02-07:
Thank you for your response. Yes, I can do a `rostopic echo /joint_states` . 
```
header: 
  seq: 1177
  stamp: 
    secs: 1675798428
    nsecs: 349937438
  frame_id: ''
name: 
  - joint1
  - joint2
  - joint3
  - joint4
  - joint5
  - joint6
position: [1.1947870254516602, 0.4018107056617737, -2.8023197650909424, -2.1242971420288086, 0.33111536502838135, -1.64092218875885]
velocity: [-2.3156080832919656e-15, -3.4240641980431974e-05, -1.903239489692968e-16, 3.172065860271246e-16, 2.4107699055753813e-15, -2.537652688216997e-15]
effort: [-2.5880089737917444e-11, -16.22670555114746, -5.969910621643066, 0.4981802999973297, 0.8521709442138672, -0.0063917189836502075]
---
```
My current active controller is I can also activate
` "velocity_controllers/JointTrajectoryController"
`
```
rosservice call /xarm/controller_manager/list_controllers "{}"`
```
controller: 
  - 
    name: "xarm6_traj_controller"
    state: "running"
    type: "position_controllers/JointTrajectoryController"
    claimed_resources: 
      - 
        hardware_interface: "hardware_interface::PositionJointInterface"
        resources: 
          - joint1
          - joint2
          - joint3
          - joint4
          - joint5
          - joint6
abc@abc-dell
```

It would be really helpfull if you can explain me a bit more about 
`You do need another type of controller running besides the JointStateController.`
Please correct me if I am wrong. 
So, I have to run two controller together in order to use moveit servoing for my xarm one more beside JointStateController. The controller can be JointGroupVelocityController or a JointGroupPositionController as suggested by moveit servoing tutorial. 
Also I saw the recent fix on xarm that states that [only one controller can be started for the same robotic arm ](https://github.com/xArm-Developer/xarm_ros/commit/35945287e074360f2ffdf4276408855c18ceaaf6). Thank You in advance for your help.

closed 2023-02-08 by Akumar201