moveit/moveit #3342: ailed to   clear octomap  by using  rosservice call /clear_octomap

dbdxnuliba, 2023-02-09:
### Description
**Failed to   clear octomap  by using  **rosservice call /clear_octomap****
Overview of your issue here.

### Your environment
* ROS Distro**:[Noetic]**
* OS Version: e.g.**20.04**
* Source or Binary build? **source**
* If source, which branch? **1.1.10**

After I load a octomap map by **rosservice call /move_group/load_map**
I successfully see the octomap in rviz .
But when I  try to clear the map by **rosservice call /clear_octomap**, the octomap in the rviz doesn't disappear
like the gif shown:
[image]

and the terminal shown info :
[image]

dbdxnuliba, 2023-02-28:
@rhaschke ,any ideas ,please

rhaschke, 2023-02-28:
I don't have time to look into that right now. In any case, it would be helpful if you could provide data (rosbag or yaml) to populate the octomap as described.

closed 2023-03-24 by rhaschke