moveit/moveit #3351: MoveIt Setup Assistant does not show anything for Self-Collision Checking

helen9975, 2023-02-24:
### Description

Self-collision checking feature shows blank output on interface when using moveit setup assistant to import new robot.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04

### Steps to reproduce
Following instructions from: http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html

On the Optimize Self-Collision Checking Step, nothing is shown after clicking on generate collision matrix (sampling density @ 10000 and min collisions @ 95%).

Everything else (adding virtual joints, planning groups, etc.) works fine.

The terminal output shows this after ckicking on "Generate Collision Matrix"

[image]

### Expected behaviour
Collision matrix should be generated.

### Actual behaviour
Nothing is shown on interface, only terminal output as shown above.

### Backtrace or Console output

See above

MikeBailleul, 2023-07-23:
I had the same issue, I found out it was because my URDF was only specifying the visual geometry.

```
<visual>
    <geometry>
        <mesh filename="package://mypackage/meshes/link.stl" />
    </geometry>
 </visual>
```
 By adding collision geometry, the UI was working like a charm.
  I used the same file for collision but feel free to use a more optimised one.
  ```
<collision>
    <geometry>
        <mesh filename="package://mypackage/meshes/link.stl" />
    </geometry>
 </collision>
```

rhaschke, 2023-07-26:
Sure, one needs to have collision models defined for collision to work :wink:
@helen9975, can you please confirm that your robot model has collision models? 
If it does, please provide your URDF + collision files for a more detailed analysis.

simonschmeisser, 2023-07-26:
There was a change in interpretation of a "missing" `<collision>` part between ROS Melodic and ROS Noetic. Formerly a missing collision definition meant that it was identical to the `visual` part. In ROS Noetic this was changed to having a purely visual link without collision.

There is a warning in your logs by the way, it's a good idea to read them :wink: