moveit/moveit #3357: Error while using moveit servo and ROS noetic with a real UR5e robot for real-time controlling

stevensu1838, 2023-03-01:
### Description

Hi,

I've been trying to do real-time jogging of a real UR5 robot with moveit servo and ROS noetic. The [Universal_Robots_ROS_Driver works well ](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) and I followed closely the [moveit servo document ](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html#). However, it throw the following errors when I run roslaunch moveit_servo spacenav_cpp.launch. Could you please help? 

NODES
  /
    servo_server (moveit_servo/servo_server)
    spacenav_node (spacenav_node/spacenav_node)
    spacenav_to_twist (moveit_servo/spacenav_to_twist)

ROS_MASTER_URI=http://localhost:11311

process[spacenav_node-1]: started with pid [213933]
process[spacenav_to_twist-2]: started with pid [213934]
process[servo_server-3]: started with pid [213935]
[ INFO] [1677647289.124081778]: Loading robot model 'ur5e_robot'...
[ WARN] [1677647289.167250718]: IK plugin for group 'manipulator' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1677647289.169026789]: IK Using joint shoulder_link -6.28319 6.28319
[ INFO] [1677647289.169044908]: IK Using joint upper_arm_link -6.28319 6.28319
[ INFO] [1677647289.169055029]: IK Using joint forearm_link -3.14159 3.14159
[ INFO] [1677647289.169064422]: IK Using joint wrist_1_link -6.28319 6.28319
[ INFO] [1677647289.169073395]: IK Using joint wrist_2_link -6.28319 6.28319
[ INFO] [1677647289.169082452]: IK Using joint wrist_3_link -6.28319 6.28319
[ INFO] [1677647289.169094018]: Looking in common namespaces for param name: manipulator/position_only_ik
[ INFO] [1677647289.169848896]: Looking in common namespaces for param name: manipulator/solve_type
[ INFO] [1677647289.171030919]: Using solve type Distance
[ INFO] [1677647289.199394860]: Starting planning scene monitor
[ INFO] [1677647289.200761364]: Listening to '/planning_scene'
[ INFO] [1677647289.200786020]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1677647289.202203818]: Listening to '/collision_object'
[ INFO] [1677647289.203511239]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1677647289.207034022]: Listening to '/attached_collision_object' for attached collision objects
[ERROR] [1677647289.520907886]: Link 'ee_link' not found in model 'ur5e_robot'
[servo_server-3] process has died [pid 213935, exit code -11, cmd /home/ysu66/catkin_ws/devel/lib/moveit_servo/servo_server __name:=servo_server __log:=/home/ysu66/.ros/log/9b224c74-b7ee-11ed-90a7-8388d6a30160/servo_server-3.log].
log file: /home/ysu66/.ros/log/9b224c74-b7ee-11ed-90a7-8388d6a30160/servo_server-3*.log
[image]

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
Terminal 1: roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.177
Terminal 2: roslaunch ur5e_moveit_config moveit_planning_execution.launch
Terminal 3: roslaunch ur5e_moveit_config moveit_rviz.launch rviz_config:=$(rospack find ur5e_moveit_config)/launch/moveit.rviz
Terminal 4: roslaunch moveit_servo spacenav_cpp.launch

Another question is I know that the moveit servo document is for simulated robot and I am working the real UR5e robot. Apart from changing use_gazebo: from true to false as show in the pic below, is there anything else I am supposed to change for the real UR5e robot? Thank you all so much
[image]

AndyZe, 2023-03-01:
You've gotta track down the issue here:

`[ERROR] [1677647289.520907886]: Link 'ee_link' not found in model 'ur5e_robot'`

Maybe you can turn on the tf display in RViz to see what the actual link names are, or look at your URDF.

> is there anything else I am supposed to change for the real UR5e robot?

Just make sure the controller topic name and `publish_period` is correct and you should be good to go. 0.008sec --> 125 Hz is your current publish frequency

stevensu1838, 2023-03-01:
@AndyZe @Akumar201 Hi, you are right. I turned on the TF display in RViz to see what the actual link names are and it turned out I had link named flange instead of ee_link.
[image]

And then I looked at the related URDF file in universal robot package, which is exactly the same as [this xacro file ](https://github.com/ros-industrial/universal_robot/blob/melodic-devel/ur_description/urdf/inc/ur_macro.xacro). I change all the flange to ee_link, as shown below:
from
[image]
to
[image]
I turned on the TF display again in RViz to see what the actual link names are and I have the ee_link now. 
[image]

May I please ask if I am solving the problem the right way?

Then I run ~$ roslaunch moveit_servo spacenav_cpp.launch and the following error appear. world link in missing.
[image]

I check the line shared by @Akumar201 and the world link does exist but it is not seen in the[ URDF in the ur_description package provided by Universal Robot](https://github.com/ros-industrial/universal_robot/blob/melodic-devel/ur_description/urdf/inc/ur_macro.xacro). 

Could you please show me what to do next? I'd love to add the world link to the URDF file myself. But not sure where to add it. Or do I need to use a different ur_description package to be compatible with moveit_servo?

stevensu1838, 2023-03-02:
@AndyZe @Akumar201 Hi, thanks a lot for your help. Changing the config file solves the problems. But now the robot doesn't move and it throws the following warnings after I run ~$ roslaunch moveit_servo spacenav_cpp.launch and the following error appear. world link is missing.

[image]

I double checked my config file with [Andy's advice here
](https://github.com/ros-planning/moveit/issues/3153#issuecomment-1143589245) and I confirm that I am using threshold_distance in the config file

Please check out my  ur_simulated_config.yaml as shown below:
###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: false # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
command_in_type: "speed_units" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.6  # Max linear velocity. Meters per publish_period. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.3 # Max angular velocity. Rads per publish_period. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Rads or Meters per publish period. Only used for joint commands on joint_command_in_topic.
  joint: 0.01
low_pass_filter_coeff: 2.  # Larger --> trust the filtered data more, trust the measurements less.

## Properties of outgoing commands
publish_period: 0.008  # 1/Nominal publish rate [seconds]
low_latency_mode: false  # Set this to true to publish as soon as an incoming Twist command is received (publish_period is ignored)

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray (for ros_control JointGroupVelocityController or JointGroupPositionController)
# or trajectory_msgs/JointTrajectory (for Universal Robots and other non-ros_control robots)
command_out_type: std_msgs/Float64MultiArray

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

## MoveIt properties
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
planning_frame: base_link  # The MoveIt planning frame. Often 'base_link' or 'world'

## Other frames
ee_frame_name: flange  # The name of the end effector link, used to return the EE pose
robot_link_command_frame: base_link # commands must be given in the frame of a robot link. Usually either the base or end effector

## Stopping behaviour
incoming_command_timeout:  0.1  # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 4

## Configure handling of singularities and joint limits
lower_singularity_threshold:  17  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.

## Topic names
cartesian_command_in_topic: delta_twist_cmds  # Topic for incoming Cartesian twist commands
joint_command_in_topic: delta_joint_cmds # Topic for incoming joint angle commands
joint_topic: joint_states
status_topic: status # Publish status to this topic
command_out_topic: /joint_group_position_controller/command # Publish outgoing commands here

## Collision checking for the entire robot body
check_collisions: true # Check collisions?
collision_check_rate: 50 # [Hz] Collision-checking can easily bog down a CPU if done too often.
# Two collision check algorithms are available:
# "threshold_distance" begins slowing down when nearer than a specified distance. Good if you want to tune collision thresholds manually.
# "stop_distance" stops if a collision is nearer than the worst-case stopping distance and the distance is decreasing. Requires joint acceleration limits
collision_check_type: threshold_distance
# Parameters for "threshold_distance"-type collision checking
self_collision_proximity_threshold: 0.01 # Start decelerating when a self-collision is this far [m]
scene_collision_proximity_threshold: 0.05 # Start decelerating when a scene collision is this far [m]
# Parameters for "stop_distance"-type collision checking
collision_distance_safety_factor: 1000 # Must be >= 1. A large safety factor is recommended to account for latency
min_allowable_collision_distance: 0.01 # Stop if a collision is closer than this [m]

And the rqt graph
[image]

stevensu1838, 2023-03-02:
Hi, please find the complete thing below:

**Terminal 1:**
`roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.177
... logging to /home/ysu66/.ros/log/5eda24e6-b94a-11ed-90a7-8388d6a30160/roslaunch-mech1936-264928.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/ysu66/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://mech1936:42617/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 500
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
 * /joint_based_cartesian_traj_controller/type: position_controll...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /pose_based_cartesian_traj_controller/base: base
 * /pose_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /pose_based_cartesian_traj_controller/tip: tool0_controller
 * /pose_based_cartesian_traj_controller/type: pose_controllers/...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/goal_time: 0.6
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /scaled_pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /scaled_pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
[... 612 lines cut ...]
[ INFO] [1677796535.243419868]: Listening to '/attached_collision_object' for attached collision objects
[ WARN] [1677796535.457549694]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796565.465334308]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796595.465494078]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796625.473414062]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796655.473470434]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796685.481518123]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796715.489404282]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796745.489404384]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796775.489442768]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677796805.497420297]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking

`
And I highlight the errors again with screen shots as shown below:
T1:
[image]
[image]
T2:
[image]
T3:
[image]
T4:
[image]

To be clear, I can move the real robot with moveit with no problem. The robot just does't respond while controling with the 3D SpaceMouse. Could you experts help? Thank you so much for everything.

AndyZe, 2023-03-02:
The control rate of a UR5e robot is 500 Hz, not 125 Hz. Might be part of the problem.

stevensu1838, 2023-03-03:
@AndyZe Hi Andy, thanks a lot for pointing this out. I've made the change in config file as you instructed:
[image]
And it robot still dosen't move with the 3D SpaceMouse as it should. 
The errors are as follow:
$ roslaunch moveit_servo spacenav_cpp.launch
... logging to /home/ysu66/.ros/log/2525d8d2-b94d-11ed-90a7-8388d6a30160/roslaunch-mech1936-275687.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/ysu66/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://mech1936:35881/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /servo_server/cartesian_command_in_topic: delta_twist_cmds
 * /servo_server/check_collisions: True
 * /servo_server/collision_check_rate: 50
 * /servo_server/collision_check_type: threshold_distance
 * /servo_server/collision_distance_safety_factor: 1000
 * /servo_server/command_in_type: unitless
 * /servo_server/command_out_topic: /joint_group_posi...
 * /servo_server/command_out_type: std_msgs/Float64M...
 * /servo_server/ee_frame_name: flange
 * /servo_server/hard_stop_singularity_threshold: 30
 * /servo_server/incoming_command_timeout: 0.1
 * /servo_server/joint_command_in_topic: delta_joint_cmds
 * /servo_server/joint_limit_margin: 0.1
 * /servo_server/joint_topic: joint_states
 * /servo_server/low_latency_mode: False
 * /servo_server/low_pass_filter_coeff: 2.0
 * /servo_server/lower_singularity_threshold: 17
 * /servo_server/min_allowable_collision_distance: 0.01
 * /servo_server/move_group_name: manipulator
 * /servo_server/num_outgoing_halt_msgs_to_publish: 4
 * /servo_server/planning_frame: base_link
 * /servo_server/publish_joint_accelerations: False
 * /servo_server/publish_joint_positions: True
 * /servo_server/publish_joint_velocities: False
 * /servo_server/publish_period: 0.0002
 * /servo_server/robot_link_command_frame: base_link
 * /servo_server/scale/joint: 0.01
 * /servo_server/scale/linear: 0.6
 * /servo_server/scale/rotational: 0.3
 * /servo_server/scene_collision_proximity_threshold: 0.05
 * /servo_server/self_collision_proximity_threshold: 0.01
 * /servo_server/status_topic: status
 * /servo_server/use_gazebo: False

NODES
  /
    servo_server (moveit_servo/servo_server)
    spacenav_node (spacenav_node/spacenav_node)
    spacenav_to_twist (moveit_servo/spacenav_to_twist)

ROS_MASTER_URI=http://localhost:11311

process[spacenav_node-1]: started with pid [275701]
process[spacenav_to_twist-2]: started with pid [275702]
process[servo_server-3]: started with pid [275703]
[ INFO] [1677825477.349827924]: Loading robot model 'ur5e_robot'...
[ WARN] [1677825477.392564590]: IK plugin for group 'manipulator' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1677825477.394902310]: IK Using joint shoulder_link -6.28319 6.28319
[ INFO] [1677825477.394924927]: IK Using joint upper_arm_link -6.28319 6.28319
[ INFO] [1677825477.394938961]: IK Using joint forearm_link -3.14159 3.14159
[ INFO] [1677825477.394951404]: IK Using joint wrist_1_link -6.28319 6.28319
[... 25 lines cut ...]
[ WARN] [1677825687.650368496]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677825702.895734335]: run_duration: 0.000325828 (0.0002)
[ WARN] [1677825717.650562572]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677825733.555729501]: run_duration: 0.000321145 (0.0002)
[ WARN] [1677825747.650578405]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[image]

All the command lines I use:
roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.177
roslaunch ur5e_moveit_config moveit_planning_execution.launch
roslaunch ur5e_moveit_config moveit_rviz.launch rviz_config:=$(rospack find ur5e_moveit_config)/launch/moveit.rviz
rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_position_controller']
stop_controllers: ['arm_controller']
strictness: 0
start_asap: false
timeout: 0.0"
roslaunch moveit_servo spacenav_cpp.launch

rqt_graph:
[image]

[image]
[image]

Thank you so much. Anyway. Andy

stevensu1838, 2023-03-04:
@Akumar201 Hi, here you go:
`~/catkin_ws$ roslaunch moveit_servo spacenav_cpp.launch
... logging to /home/ysu66/.ros/log/a9dcf8c8-ba40-11ed-a2a7-53434ed7b613/roslaunch-mech1936-15781.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/ysu66/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://mech1936:45505/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /servo_server/cartesian_command_in_topic: delta_twist_cmds
 * /servo_server/check_collisions: True
 * /servo_server/collision_check_rate: 50
 * /servo_server/collision_check_type: threshold_distance
 * /servo_server/collision_distance_safety_factor: 1000
 * /servo_server/command_in_type: unitless
 * /servo_server/command_out_topic: /joint_group_posi...
 * /servo_server/command_out_type: std_msgs/Float64M...
 * /servo_server/ee_frame_name: flange
 * /servo_server/hard_stop_singularity_threshold: 30
 * /servo_server/incoming_command_timeout: 0.1
 * /servo_server/joint_command_in_topic: delta_joint_cmds
 * /servo_server/joint_limit_margin: 0.1
 * /servo_server/joint_topic: joint_states
 * /servo_server/low_latency_mode: False
 * /servo_server/low_pass_filter_coeff: 2.0
 * /servo_server/lower_singularity_threshold: 17
 * /servo_server/min_allowable_collision_distance: 0.01
 * /servo_server/move_group_name: manipulator
 * /servo_server/num_outgoing_halt_msgs_to_publish: 4
 * /servo_server/planning_frame: base_link
 * /servo_server/publish_joint_accelerations: False
 * /servo_server/publish_joint_positions: True
 * /servo_server/publish_joint_velocities: False
 * /servo_server/publish_period: 0.0002
 * /servo_server/robot_link_command_frame: base_link
 * /servo_server/scale/joint: 0.01
 * /servo_server/scale/linear: 0.6
 * /servo_server/scale/rotational: 0.3
 * /servo_server/scene_collision_proximity_threshold: 0.05
 * /servo_server/self_collision_proximity_threshold: 0.01
 * /servo_server/status_topic: status
 * /servo_server/use_gazebo: False

NODES
  /
    servo_server (moveit_servo/servo_server)
    spacenav_node (spacenav_node/spacenav_node)
    spacenav_to_twist (moveit_servo/spacenav_to_twist)

ROS_MASTER_URI=http://localhost:11311

process[spacenav_node-1]: started with pid [15798]
process[spacenav_to_twist-2]: started with pid [15799]
process[servo_server-3]: started with pid [15800]
[ INFO] [1677904054.252088380]: Loading robot model 'ur5e_robot'...
[ WARN] [1677904054.297484192]: IK plugin for group 'manipulator' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1677904054.300286095]: IK Using joint shoulder_link -6.28319 6.28319
[ INFO] [1677904054.300313251]: IK Using joint upper_arm_link -6.28319 6.28319
[ INFO] [1677904054.300335081]: IK Using joint forearm_link -3.14159 3.14159
[ INFO] [1677904054.300358323]: IK Using joint wrist_1_link -6.28319 6.28319
[ INFO] [1677904054.300382763]: IK Using joint wrist_2_link -6.28319 6.28319
[ INFO] [1677904054.300406757]: IK Using joint wrist_3_link -6.28319 6.28319
[ INFO] [1677904054.300435112]: Looking in common namespaces for param name: manipulator/position_only_ik
[ INFO] [1677904054.301374547]: Looking in common namespaces for param name: manipulator/solve_type
[ INFO] [1677904054.302672002]: Using solve type Distance
[ INFO] [1677904054.334405073]: Starting planning scene monitor
[ INFO] [1677904054.336160209]: Listening to '/planning_scene'
[ INFO] [1677904054.336182452]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1677904054.337872209]: Listening to '/collision_object'
[ INFO] [1677904054.339396767]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1677904054.342989431]: Listening to '/attached_collision_object' for attached collision objects
[ WARN] [1677904054.555635587]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677904055.518666200]: run_duration: 0.000560836 (0.0002)
[ WARN] [1677904084.555740741]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
[ WARN] [1677904085.714023851]: run_duration: 0.000209027 (0.0002)

`

I've tried everything but the robot still doesn't move with 3D mouse. Could you please take a look at my config file in my last comment? Is there anything wrong?

The 3D mouse works well and when I do rostopic echo /servo_server/delta_twist_cmds and touch the 3D mouse with my hand, the terminal shows the commands updating. 
---
header: 
  seq: 11595
  stamp: 
    secs: 1677904297
    nsecs:  85525449
  frame_id: ''
twist: 
  linear: 
    x: 0.01171875
    y: 0.060546875
    z: -0.228515625
  angular: 
    x: -0.68359375
    y: -0.0703125
    z: 0.31640625
---
header: 
  seq: 11596
  stamp: 
    secs: 1677904297
    nsecs:  93071129
  frame_id: ''
twist: 
  linear: 
    x: 0.01171875
    y: 0.09375
    z: -0.18359375
  angular: 
    x: -0.68359375
    y: -0.0703125
    z: 0.31640625
---
header: 
  seq: 11597
  stamp: 
    secs: 1677904297
    nsecs: 100695736
  frame_id: ''
twist: 
  linear: 
    x: 0.01171875
    y: 0.09375
    z: -0.18359375
  angular: 
    x: -0.68359375
    y: -0.078125
    z: 0.25
---
^Cheader: 
  seq: 11598
  stamp: 
    secs: 1677904297
    nsecs: 109162912
  frame_id: ''
twist: 
  linear: 
    x: 0.041015625
    y: 0.1484375
    z: 0.0546875
  angular: 
    x: -0.68359375
    y: -0.078125
    z: 0.25
[image]

And again I used [this ROS driver and followed the documents ](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver)to control the real UR5e robot and the robot does move with MoveIt.

Thank you so much for your quick response and great help. Cheers.

stevensu1838, 2023-03-05:
@AndyZe @Akumar201 Hi, I've been trying hard to solve the problem. 
First, I corrected one mistake in the config file, I change from publish_period: 0.0002 and 0.002. But it didn't help.
Second, I tried to change from ee_frame_name: flange to ee_frame_name: tool0. But it didn't work.
Third, I tried to change from robot_link_command_frame: base_link to flange or tool0. Neither worked. Btw, just curious, how do I know if my robot_link_command_frame should be base_link, flange or tool0. 

Thanks a lot. I will try my best to get it running with you support

stevensu1838, 2023-03-09:
@Akumar201 @AndyZe 

Hi, I am still trying to get moveit servo work with ROS noetic and UR5e. I simply want to provide the following info to see if it is useful for you to give me some advice.

When I control the arm with moveit, the arm moves and the cmd_servo_q and cmd_servo_q_last values updates.
However, when I use the 3D Space Mouse to jog the arm, the robot doesn't do anything and no values update on the teaching pendant of the UR5e.  

[image]

@Akumar201 Hi, could you please share how you set up ur5e with moveit servo and get it working? I am thinking of switching to ROS melodic. Cheers

AndyZe, 2023-03-15:
Solved in https://github.com/ros-planning/moveit/issues/3372

closed 2023-03-15 by AndyZe