moveit/moveit #3372: Can't use moveit servo and ROS Melodic for real time control of a real UR5e robot. Help, please.

stevensu1838, 2023-03-13:
### Description

Hi,

I've been trying to control a real UR5e robot with moveit servo and ROS. I started with ROS noetic and it didn't work. And then I switched to ROS Melodic. The issue I report here is **all about using ROS Melodic and moveit servo for controlling a real UR5e robot**. The [Universal_Robots_ROS_Driver works well ](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver)and I followed closely the [moveit servo document ](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html#). Moveit it does move the UR5e arm as expected. However, the robot doesn't react to the 3D SpaceMouse after I run roslaunch moveit_servo spacenav_cpp.launch. Could you please help?

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
1. UR5e driver

$ source /opt/ros/melodic/setup.bash
$ mkdir -p catkin_ws/src && cd catkin_ws
$ git clone https://github.com/UniversalRobots/Universal_Robots_ROS_Driver.git src/Universal_Robots_ROS_Driver
$ git clone -b melodic-devel https://github.com/ros-industrial/universal_robot.git src/universal_robot
$ sudo apt update -qq
$ rosdep update
$ rosdep install --from-paths src --ignore-src -y
$ catkin_make
$ source devel/setup.bash

2. Moveit and Moveit servo installed from source

rosdep update
sudo apt update
sudo apt dist-upgrade 
sudo apt install python-wstool python-catkin-tools python-rosdep 
cd catkin_ws
wstool init src
wstool merge -t src https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall
wstool update -t src
rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}
catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release 

3. Install spacenav_node

$ sudo apt install spacenavd
$ sudo apt install ros-melodic-spacenav-node

4. Run it

Terminal 1: roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.177
Terminal 2: roslaunch ur5e_moveit_config moveit_planning_execution.launch
Terminal 3: roslaunch ur5e_moveit_config moveit_rviz.launch rviz_config:=$(rospack find ur5e_moveit_config)/launch/moveit.rviz
Terminal 4: roslaunch moveit_servo spacenav_cpp.launch

After running those command lines, I make sure I click play button and choose remote control mode by pressing the teach pendant icon on the physical UR5e teach pendant. 

### Expected behaviour
I expected the TCP of UR5e would move while touching the cap of the 3D Space Mouse. 

### Actual behaviour
When I try to move the TCP of UR5e with the interactive markers in Moveit Rviz, it works perfectly. However, the UR5e arm doesn't move with the 3D SpaceMouse

### Backtrace or Console output
1. Please check out the frames in Moveit Rivz
[image]
2. The Config file
[image]
[image]

3. Moveit servo terminal
user@mech1646:~$ roslaunch moveit_servo spacenav_cpp.launch
... logging to /home/user/.ros/log/4ff329d0-c14e-11ed-8dc4-480fcf41d72a/roslaunch-mech1646-21083.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:44537/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /servo_server/cartesian_command_in_topic: delta_twist_cmds
 * /servo_server/check_collisions: True
 * /servo_server/collision_check_rate: 5
 * /servo_server/collision_check_type: threshold_distance
 * /servo_server/collision_distance_safety_factor: 1000
 * /servo_server/command_in_type: unitless
 * /servo_server/command_out_topic: /joint_group_posi...
 * /servo_server/command_out_type: std_msgs/Float64M...
 * /servo_server/ee_frame_name: flange
 * /servo_server/hard_stop_singularity_threshold: 30
 * /servo_server/incoming_command_timeout: 0.1
 * /servo_server/joint_command_in_topic: delta_joint_cmds
 * /servo_server/joint_limit_margin: 0.1
 * /servo_server/joint_topic: joint_states
 * /servo_server/low_latency_mode: False
 * /servo_server/low_pass_filter_coeff: 2.0
 * /servo_server/lower_singularity_threshold: 17
 * /servo_server/min_allowable_collision_distance: 0.01
 * /servo_server/move_group_name: manipulator
 * /servo_server/num_outgoing_halt_msgs_to_publish: 4
 * /servo_server/planning_frame: base_link
 * /servo_server/publish_joint_accelerations: False
 * /servo_server/publish_joint_positions: True
 * /servo_server/publish_joint_velocities: False
 * /servo_server/publish_period: 0.002
 * /servo_server/robot_link_command_frame: base_link
 * /servo_server/scale/joint: 0.01
 * /servo_server/scale/linear: 0.003
 * /servo_server/scale/rotational: 0.006
 * /servo_server/scene_collision_proximity_threshold: 0.05
 * /servo_server/self_collision_proximity_threshold: 0.01
 * /servo_server/status_topic: status
 * /servo_server/use_gazebo: False

NODES
  /
    servo_server (moveit_servo/servo_server)
    spacenav_node (spacenav_node/spacenav_node)
    spacenav_to_twist (moveit_servo/spacenav_to_twist)

ROS_MASTER_URI=http://localhost:11311

process[spacenav_node-1]: started with pid [21098]
process[spacenav_to_twist-2]: started with pid [21099]
process[servo_server-3]: started with pid [21105]
[ INFO] [1678678203.092871324]: Loading robot model 'ur5e_robot'...
[ WARN] [1678678203.144808395]: IK plugin for group 'manipulator' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1678678203.147478428]: IK Using joint shoulder_link -6.28319 6.28319
[ INFO] [1678678203.147516698]: IK Using joint upper_arm_link -6.28319 6.28319
[ INFO] [1678678203.147540431]: IK Using joint forearm_link -3.14159 3.14159
[ INFO] [1678678203.147562739]: IK Using joint wrist_1_link -6.28319 6.28319
[ INFO] [1678678203.147584521]: IK Using joint wrist_2_link -6.28319 6.28319
[ INFO] [1678678203.147597211]: IK Using joint wrist_3_link -6.28319 6.28319
[ INFO] [1678678203.147611403]: Looking in common namespaces for param name: manipulator/position_only_ik
[ INFO] [1678678203.148670727]: Looking in common namespaces for param name: manipulator/solve_type
[ INFO] [1678678203.150059770]: Using solve type Distance
[ INFO] [1678678203.184920352]: Starting planning scene monitor
[ INFO] [1678678203.186316523]: Listening to '/planning_scene'
[ INFO] [1678678203.186353975]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1678678203.187982639]: Listening to '/collision_object'
[ INFO] [1678678203.189774048]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] [1678678203.192745296]: Listening to '/attached_collision_object' for attached collision objects
[ WARN] [1678678203.364689373]: Collision check rate is low, increase it in yaml file if CPU allows
[ WARN] [1678678203.407585052]: An acceleration limit is not defined for this joint; minimum stop distance should not be used for collision checking
4. rqt_graph
[image]
5. rostopic echo 
rostopic echo /servo_server/delta_twist_cmds (the following values change as expected while 3D Mouse is touched)
header: 
  seq: 9873
  stamp: 
    secs: 1678678697
    nsecs: 731636113
  frame_id: ''
twist: 
  linear: 
    x: -0.009765625
    y: -0.037109375
    z: -0.025390625
  angular: 
    x: 0.01953125
    y: 0.025390625
    z: -0.041015625

rostopic echo /joint_states (the following values don't change over running time)
header: 
  seq: 104045
  stamp: 
    secs: 1678678598
    nsecs: 788078719
  frame_id: ''
name: [elbow_joint, shoulder_lift_joint, shoulder_pan_joint, wrist_1_joint, wrist_2_joint,
  wrist_3_joint]
position: [-2.5634050369262695, -0.914035515194275, -0.7236884276019495, 4.911894592433729, 1.4957938194274902, -2.5117059389697474]
velocity: [-0.0, 0.0, 0.0, 0.0, -0.0, -0.0]
effort: [1.8231045007705688, -0.3551892042160034, 0.1035819947719574, 0.27333542704582214, 0.05359789356589317, 0.07845792174339294]
---

Could you please help? I've been trying my best but have no idea what I did wrong or what I missed. Especially, How do I know what value to set for the following two parameters in the config file. I'd love to pay for your advice. Cheers. 

command_out_type: std_msgs/Float64MultiArray
#command_out_type: trajectory_msgs/JointTrajectory
command_out_topic: /joint_group_position_controller/command # Publish outgoing commands here

stevensu1838, 2023-03-15:
@AndyZe Hi,
I assumed that the problem was from my **command_out_topic/type settings** in the config file. So I did the following to ensure it was all right.

user@mech1646:~$ rostopic list | grep command
/joint_group_vel_controller/command
/pos_joint_traj_controller/command
/scaled_pos_joint_traj_controller/command
/ur_hardware_interface/script_command

@mech1646:~$ rostopic type /joint_group_vel_controller/command
std_msgs/Float64MultiArray

Correct me if I am wrong. I guess the topic /joint_group_vel_controller/command should be the **command_out_topic** I need to control my UR5e robot and std_msgs/Float64MultiArray should be the **command_out_type**. And I changed the them accordingly in the config file as shown below:

[image]

And after making the changes above, I run the command lines again

rqt_graph
 
[image]

It looks like the servo_server  is sending messages through topic /joint_group_vel_controller/command. 

However, when I do rostopic echo /joint_group_vel_controller/command, the values stay the same and the robot arm doesn't move as the 3D Space Mouse is touched. 

---
layout: 
  dim: []
  data_offset: 0
data: [-1.040838872505293, -0.9112890596473224, -2.1359463793435065, 4.5117494991431855, 1.4570941855037445, -2.827559475837126]
---
layout: 
  dim: []
  data_offset: 0
data: [-1.040835282009427, -0.9112965271447433, -2.1359504081757175, 4.511748115588173, 1.4570970478555392, -2.8275574101030436]
---
layout: 
  dim: []
  data_offset: 0
data: [-1.0408444166405757, -0.9113029899652944, -2.1359517511185966, 4.511737322932115, 1.4570980019428834, -2.8275543373414163]
---
layout: 
  dim: []
  data_offset: 0
data: [-1.0408488920301093, -0.9113039918824619, -2.135952198765332, 4.511734996948135, 1.4570931939720897, -2.8275469552578607]
---
layout: 
  dim: []
  data_offset: 0
data: [-1.0408463050679266, -0.9113001773776039, -2.1359473386546335, 4.51173222203977, 1.457096485011821, -2.8275519131308355]
---
layout: 
  dim: []
  data_offset: 0
data: [-1.0408426355514453, -0.9112959296224972, -2.135939517197727, 4.511738039505095, 1.4570961973698329, -2.8275609843184335]
---
layout: 
  dim: []
  data_offset: 0
data: [-1.0408426839358194, -0.9113019642873719, -2.1359357179507126, 4.51174363441386, 1.4570909754887573, -2.8275576502103563]
---
layout: 
  dim: []
  data_offset: 0
data: [-1.0408414284998728, -0.9113028036175624, -2.1359356436281463, 4.511743115195172, 1.4570955132169154, -2.8275513731069766]
---
layout: 
  dim: []
  data_offset: 0
data: [-1.0408410100218415, -0.9113030833963724, -2.135936810947201, 4.511739286376484, 1.4570907871710004, -2.827555638569872]
---

I feel like I am very close to solve the problem. Any thoughts, please

stevensu1838, 2023-03-15:
@AndyZe, Hi Andy, problem solved. I've learnt so much again this time. Many thanks. Please close all my issues. Cheers.

closed 2023-03-15 by AndyZe

dChaisumdet, 2023-05-01:
@stevensu1838 What did you change

> @AndyZe Hi, I assumed that the problem was from my **command_out_topic/type settings** in the config file. So I did the following to ensure it was all right.
>
> user@mech1646:~$ rostopic list | grep [...]

What did you change after this for it to work? I'm currently at the same stage and have no idea how to fix it!

MShields1986, 2023-07-12:
@stevensu1838 what was the final solution?

balthasarj, 2024-05-20:
@stevensu1838 I am also interested what the final solution was.

VJan2022, 2024-07-12:
@stevensu1838 I am also interested what the final solution was.

Jovan2186, 2024-11-12:
@stevensu1838 could you tell me how did you solve it