moveit/moveit #3373: Planner Produces Plans with Large Detours

febert, 2023-03-15:
Hi,

I'm trying to use xarm_planner to plan around a couple of obstacles.
However the planner finds paths that are highly suboptimal and take very large detours, as shown in the picture below (it does sometimes find short paths though):

[image]

Are there any planner settings or alternative planners that I can try? It should be using RRTConnect by default, is that correct?

Repeatedly sampling a batch of plans (e.g. 100) and picking the shortest would yield satisfactory plans, but it would be too slow for my application. Currently with the movegroup interface it is quite slow to run in a loop since group.plan(my_xarm_plan) performs visualization for every single plan.

I'm using the MoveIt integration of Xarm in this repo:
https://github.com/xArm-Developer/xarm_ros/tree/master/xarm_planner/src

### My environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04

Thanks a lot!

AndyZe, 2023-03-16:
I'd suggest to read up on parallel planning. There's [a tutorial](https://moveit.picknik.ai/main/doc/how_to_guides/parallel_planning/parallel_planning_tutorial.html?highlight=parallel%20planning) and [a blog post](https://picknik.ai/moveit%202/parallel%20planning/motion%20planning/2023/02/15/parallel-planning-with-MoveIt-2.html).

AndyZe, 2023-03-16:
Yes, RRTConnect is the default.

rhaschke, 2023-03-24:
As you use _sampling_-based planning, the resulting trajectory is _random_.

closed 2023-03-24 by rhaschke