moveit/moveit #3378: in scene objects tab of MotionPlanning panel in rviz  when I want to add a box and move it to a specfied position  in rviz panel ,it can't work well ,like the gif shown

dbdxnuliba, 2023-03-22:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic]
* OS Version: 20.04
* Source or Binary build? source
* If binary, which release version?
* If source, which branch? 1.1.11

### Steps to reproduce
in scene objects tab of MotionPlanning panel in rviz 
when I want to add a box and move it to a specfied position  in rviz panel ,it can't work well ,like the gif shown
[image]

rhaschke, 2023-03-23:
Indeed, publishing PS updates, I can confirm this behavior. Fixed in 0819713fb67e61ee6e4f3949a01f7922ae42aade

closed 2023-03-23 by rhaschke

dbdxnuliba, 2023-03-24:
Thank you for your prompt response and efficient handling,it works well now

dbdxnuliba, 2023-03-24:
but could you please have a look at the issue :https://github.com/ros-planning/moveit/issues/3380