moveit/moveit #3380: /clear_octomap can't clear octomap in moveit

dbdxnuliba, 2023-03-23:
### Description

Overview of your issue here.

I have found a bug in MoveIt. When the point cloud topic stops publishing, clicking "clear octomap" in MotionPlanning Panel of rviz does not clear the map. However, if the point cloud topic continues to publish, the map will continue to update based on the point cloud and cannot be cleared completely. This means that once Map Building is turned on in MoveIt, the map cannot be cleared unless MoveIt is restarted.
[image]

### Your environment
* ROS Distro:Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build?source
* If binary, which release version?
* If source, which branch?1.1.11
*

rhaschke, 2023-03-24:
Looks like the scene update is not correctly received by rviz. In the move_group's PS, the octomap is correctly cleared. This can be easily verified by disabling and re-enabling the display in rviz, which re-creates everything. I'm looking into the issue in more depth...

closed 2023-03-24 by rhaschke

dbdxnuliba, 2023-03-27:
thanks for your prompt answer,did you mean the issue has been fixed and will pub in the next version ,

rhaschke, 2023-03-27:
Yes, the issue will be fixed with the next release.

dbdxnuliba, 2023-04-17:
ok ,thanks a lot