moveit/moveit #3383: For a robot with a composite configuration of agv + arm,at the application layer,  when we use moveit interface, is  the variable of  target_pose  relative root of agv or the  arm base_link

dbdxnuliba, 2023-03-24:
### Description

Overview of your issue here.

### Your environment
* ROS Distro:Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? source
* If binary, which release version?
* If source, which branch?1.1.11

### Steps to reproduce
Question:
For a robot with a composite configuration of agv + arm,
I use urdf file to define the root coordinate system as  world, 
and the base_link of the arm is a sub-branch of the world coordinate system.

Then at the application layer,  when setting the target pose at the end of the manipulator in the moveit interface, 

**geometry_msgs ::PoseStamped target;
group.setPoseTarget(target);**

**Should the value of  variable  target be set to the pose of the end relative to the base of the robotic arm, or the pose of the end of the robotic arm relative to the  root coordinate system world?**

I guess we should set the variable target to be the pose of the end of the robotic arm relative to the  root coordinate system world,
because  in the the source code of moveit /moveit-1.1.11/moveit_core/robot_state/src/robot_state.cpp
we can see the code that 
[image]

Is my understanding correct, please give your answer and  analysis .

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

rhaschke, 2023-03-24:
This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here only.
In general, all operations performed by MoveIt are relative to the [planning frame](https://docs.ros.org/en/noetic/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#aaf47d321214b228138aab2d0eeeb4272). If you set a Cartesian target pose, you should be able to specify an arbitrary reference frame via `header.frame_id`.

closed 2023-03-24 by rhaschke

AndyZe, 2023-03-24:
Also noting that there is a `getPlanningFrame()` function so you can check that:

http://docs.ros.org/en/jade/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html

dbdxnuliba, 2023-03-27:
> This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here only.
> In general, all operations performed by MoveIt are relative to the [planning [...]
yes I have tried it ,but header.frame_id was set to world or base_link  does not affect the result

rhaschke, 2023-03-27:
Changing the target's `frame_id` definitely makes a difference for me.
- Do you use `group.set_pose_target()`?
- Do the used frames actually differ?

dbdxnuliba, 2023-04-20:
> Changing the target's `frame_id` definitely makes a difference for me.
> 
> * Do you use `group.set_pose_target()`?
> * Do the used frames actually differ?

yes it is.
I use group.set_pose_target();
I use target.header.frame_id = "world" or I use target.header.frame_id = "left_arm_base_link" ,the plan result is not affected

and