moveit/moveit #3401: Interactive marker not working for multiple robots

buswayne, 2023-04-14:
Hello there!

I'm simulating two robotic arms that share the same urdf. In gazebo and in Rviz, everything is coherent with the two namespaces that I gave the robots.

The behavior obtained with the MoveIt plugins for Rviz is as expected if the goal state is set modifying the joints (i.e, the trajectory is planned and executed correctly by the respective robot I want to control).

The issue is with the interactive marker that switches between one robot and the other, it cannot be moved and it "flickers". In short, apparently it is shared between the two robots (even though they have different namespaces!). My guess is that the problem is due to the fact that for both the robots the planning group is called "arm", and the moveit plugin code related to the interactive marker is not considering the namespace.

Do you have any suggestion to fix the issue?

Thanks in advance!

[image]

rhaschke, 2023-04-14:
I guess the problem stems from ambiguous TF names or do you have separate TF namespaces for the two robots as well?
(Note that this is different from the ROS parameter namespaces.)
To further study the issue, a full moveit and rviz config would be helpful.

buswayne, 2023-04-15:
Hi Robert,

Thanks for the reply! Yep, the TF have separate namespaces, too! Here below the details of the configs. You can see from the graph that the only topics without namespace are the one pulished by rviz that containt "robot_interaction_interactive_marker_topic", that's why I believe this is the issue
 
GRAPH:
[image]

TF TREE:
[image]

Thanks a lot for your support!

R

rhaschke, 2023-04-15:
Please test #3403.

buswayne, 2023-04-15:
It works like charm!! Thanks a lot Robert

[image]

PR #3403 (2023-04-24): MPD: Use different namespaces for RobotInteraction
Fixes #3401
All MotionPlanningDisplays use the same fixed namespace for `RobotInteraction`:
https://github.com/ros-planning/moveit/blob/6c5203338bf305e1a9303839ded3afebee423c78/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_display.cpp#L1146

This causes ambiguities when resolving interactive markers for the same robot model.
My proposal is to prepend the (typically empty) parent namespace of the `robot_description` parameter.
In this fashion, most existing code accessing the interactive marker topics too, will contin [... cut]
closed 2023-04-15 by buswayne