moveit/moveit #3402: MoveIt Setup Assistant crashes when you are

mlautman, 2023-04-14:
### Description

Bug detected in debian and source install of MoveIt where if you open the setup assistant, go to the perception page, then

### Your environment
* ROS Distro: [Noetic]
* OS Version: mint20 which is built on top of ubuntu20.
* Source or Binary build:  Both
* If binary, which release version: ros-noetic-moveit-setup-assistant version 1.1.11-1focal.20230216.032317
* If source, which branch: noetic-devel

### Steps to reproduce
[Video](https://user-images.githubusercontent.com/3253549/232148804-dded4c0c-ada0-4e32-b35c-67cf24e3ded3.mp4) https://user-images.githubusercontent.com/3253549/232148804-dded4c0c-ada0-4e32-b35c-67cf24e3ded3.mp4

### Actual behaviour
Crashes because `sensors_plugin_config_parameter_list_` is empty and line 1892 of perception_widget.cpp tries to access the vector at index 0 (`sensors_plugin_config_parameter_list_[0][name] = new_parameter;`)

### Backtrace or Console output
```
   #0  0x00007ffff7ec0e4c in std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, moveit_setup_assistant::GenericParameter>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, moveit_setup_assistant::GenericParameter> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, moveit_setup_assistant::GenericParameter> > >::_M_begin()
    (this=0x0) at /usr/include/c++/9/bits/stl_tree.h:750
#1  0x00007ffff7ec0f00 in std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, moveit_setup_assistant::GenericParameter>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, moveit_setup_assistant::GenericParameter> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, moveit_setup_assistant::GenericParameter> > >::lower_bound(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (this=0x0, __k="sensor_plugin") at /usr/include/c++/9/bits/stl_tree.h:1287
#2  0x00007ffff7ec09c5 in std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, moveit_setup_assistant::GenericParameter, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, moveit_setup_assistant::GenericParameter> > >::lower_bound(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (this=0x0, __x="sensor_plugin") at /usr/include/c++/9/bits/stl_map.h:1258
#3  0x00007ffff6999f5d in std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, moveit_setup_assistant::GenericParameter, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > --Type <RET> for more, q to quit, c to continue without paging--
const, moveit_setup_assistant::GenericParameter> > >::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (this=0x0, __k="sensor_plugin") at /usr/include/c++/9/bits/stl_map.h:495
#4  0x00007ffff6991682 in moveit_setup_assistant::MoveItConfigData::addGenericParameterToSensorPluginConfig(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (this=0x555555830c70, name="sensor_plugin", value="occupancy_map_monitor/PointCloudOctomapUpdater")
    at /home/contractor/ws_electric_eel/src/moveit/moveit_setup_assistant/src/tools/moveit_config_data.cpp:1892
#5  0x00007ffff7ebcd39 in moveit_setup_assistant::PerceptionWidget::focusLost() (this=0x5555572f9f70)
    at /home/contractor/ws_electric_eel/src/moveit/moveit_setup_assistant/src/widgets/perception_widget.cpp:211
#6  0x00007ffff7f14c88 in moveit_setup_assistant::SetupAssistantWidget::moveToScreen(int) (this=0x7fffffffccd0, index=10)
    at /home/contractor/ws_electric_eel/src/moveit/moveit_setup_assistant/src/widgets/setup_assistant_widget.cpp:209
#7  0x00007ffff7f14bfa in moveit_setup_assistant::SetupAssistantWidget::navigationClicked(QModelIndex const&) (this=0x7fffffffccd0, index=...)
    at /home/contractor/ws_electric_eel/src/moveit/moveit_setup_assistant/src/widgets/setup_assistant_widget.cpp:195
#8  0x00007ffff7e368c9 in moveit_setup_assistant::SetupAssistantWidget::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)
    (_o=0x7fffffffccd0, _c=QMetaObject::InvokeMetaMethod, _id=0, _a=0x7fffffffb9c0)
    at /home/contractor/ws_electric_eel/build/moveit_setup_assistant/moveit_setup_assistant_widgets_autogen/RZ4CM6YYJ6/moc_setup_assistant_widget.cpp:108
#9  0x00007ffff70b11d0 in QMetaObject::activate(QObject*, int, int, void**) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
--Type <RET> for more, q to quit, c to continue without paging--c
#10 0x00007ffff7958269 in QAbstractItemView::clicked(QModelIndex const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff795c510 in QAbstractItemView::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff79a2153 in QListView::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007ffff77342b6 in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff77e1d52 in QFrame::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff7966482 in QAbstractItemView::viewportEvent(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff708551b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff76f1a55 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff76fb343 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff708580a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff76fa457 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff775035d in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#22 0x00007ffff77531ec in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff76f1a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff76fb0f0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff708580a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff4f317d3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#27 0x00007ffff4f3310b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#28 0x00007ffff4f0d35b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#29 0x00007fffe91b532e in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#30 0x00007ffff460017d in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff4600400 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff46004a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff70dd435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
[... cut ...]
#36 0x0000555555575221 in main(int, char**) (argc=1, argv=0x7fffffffcef8) at /home/contractor/ws_electric_eel/src/moveit/moveit_setup_assistant/src/setup_assistant_main.cpp:107
(gdb) frame 4
#4  0x00007ffff6991682 in moveit_setup_assistant::MoveItConfigData::addGenericParameterToSensorPluginConfig (this=0x555555830c70, name="sensor_plugin", 
    value="occupancy_map_monitor/PointCloudOctomapUpdater") at /home/contractor/ws_electric_eel/src/moveit/moveit_setup_assistant/src/tools/moveit_config_data.cpp:1892
1892	  sensors_plugin_config_parameter_list_[0][name] = new_parameter;
```

rhaschke, 2023-04-14:
Duplicate of #3398, which was fixed in #3399.

closed 2023-04-14 by rhaschke

v4hn, 2023-04-16:
Independent of the reported issue, it's great to see you active again @mlautman .  You have been missed. 🥳