moveit/moveit #3404: Once the AITstar algorithm has been running, and there is no planning result for ten minutes,just like fall into an infinite loop or thread deadlock,and no error is reported, and the information is only printed in the first few seconds

dbdxnuliba, 2023-04-18:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build?Source
* If binary, which release version?
* If source, which branch?https://github.com/alaflaquiere/moveit-fork
* ompl1.6.0

when I  use AITstar which has been integrated with moveit in the version https://github.com/alaflaquiere/moveit-fork
[image]

Test a demo p2p in joint space for an arm as shown above fig.

Once the AITstar algorithm has been running, and there is no planning result for ten minutes,just like fall into an infinite loop or thread deadlock,and no error is reported, and the information is only printed in the first few seconds, and no information is output for the rest of the time,
and the following is the information it print
```
[ INFO] [1681797590.748300404, 657.259000000]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1681797590.748370254, 657.259000000]: Using planning pipeline 'ompl'
[ INFO] [1681797590.750759325, 657.262000000]: Planner configuration 'left_arm[AITstar]' will use planner 'geometric::AITstar'. Additional configuration parameters will be set when the planner is constructed.
[ WARN] [1681797590.751458594, 657.262000000]: left_arm/left_arm[AITstar]: No optimization objective has been specified. Defaulting to path length.
[ WARN] [1681797590.763233343, 657.274000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.763338939, 657.274000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.763390255, 657.274000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.763434913, 657.274000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ INFO] [1681797590.763470608, 657.274000000]: left_arm/left_arm[AITstar]: Solving the given planning problem. The current best solution cost is inf
[ INFO] [1681797590.763535262, 657.274000000]: left_arm/left_arm[AITstar]: Solving the given planning problem. The current best solution cost is inf
[ INFO] [1681797590.764055473, 657.275000000]: left_arm/left_arm[AITstar]: Solving the given planning problem. The current best solution cost is inf
[ INFO] [1681797590.764143138, 657.275000000]: left_arm/left_arm[AITstar]: Solving the given planning problem. The current best solution cost is inf
[ WARN] [1681797590.783376051, 657.285000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.785786154, 657.290000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.786995111, 657.290000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.788901633, 657.294000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ INFO] [1681797590.791153904, 657.294000000]: left_arm/left_arm[AITstar] (15 iterations): Found a new exact solution of cost 6.9114. Sampled a total of 182 states, 100 of which were valid samples (54.9 %). Processed 5 edges, 2 of which were collision checked (40.0 %). The forward search tree has 3 vertices, 1 of which are start states. The reverse search tree has 22 vertices, 3 of which are goal states.
[ INFO] [1681797590.793788454, 657.300000000]: left_arm/left_arm[AITstar] (24 iterations): Found a new exact solution of cost 11.8237. Sampled a total of 201 states, 100 of which were valid samples (49.8 %). Processed 8 edges, 2 of which were collision checked (25.0 %). The forward search tree has 3 vertices, 1 of which are start states. The reverse search tree has 35 vertices, 3 of which are goal states.
[ INFO] [1681797590.794931456, 657.301000000]: left_arm/left_arm[AITstar] (32 iterations): Found a new exact solution of cost 11.2852. Sampled a total of 182 states, 100 of which were valid samples (54.9 %). Processed 11 edges, 3 of which were collision checked (27.3 %). The forward search tree has 4 vertices, 1 of which are start states. The reverse search tree has 32 vertices, 3 of which are goal states.
[ INFO] [1681797590.795870824, 657.302000000]: left_arm/left_arm[AITstar] (44 iterations): Found a new exact solution of cost 12.9353. Sampled a total of 198 states, 100 of which were valid samples (50.5 %). Processed 16 edges, 4 of which were collision checked (25.0 %). The forward search tree has 5 vertices, 1 of which are start states. The reverse search tree has 35 vertices, 3 of which are goal states.
[ WARN] [1681797590.814489631, 657.318000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.816286095, 657.323000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.816592519, 657.323000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ INFO] [1681797590.821065709, 657.327000000]: left_arm/left_arm[AITstar] (68 iterations): Found a new exact solution of cost 11.6919. Sampled a total of 394 states, 200 of which were valid samples (50.8 %). Processed 23 edges, 6 of which were collision checked (26.1 %). The forward search tree has 5 vertices, 1 of which are start states. The reverse search tree has 17 vertices, 4 of which are goal states.
[ INFO] [1681797590.821670911, 657.327000000]: left_arm/left_arm[AITstar] (49 iterations): Found a new exact solution of cost 10.6903. Sampled a total of 385 states, 200 of which were valid samples (51.9 %). Processed 15 edges, 5 of which were collision checked (33.3 %). The forward search tree has 6 vertices, 1 of which are start states. The reverse search tree has 17 vertices, 4 of which are goal states.
[ WARN] [1681797590.839600707, 657.345000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.841360625, 657.346000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.841499635, 657.346000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ INFO] [1681797590.847279938, 657.350000000]: left_arm/left_arm[AITstar] (88 iterations): Found a new exact solution of cost 10.6461. Sampled a total of 535 states, 300 of which were valid samples (56.1 %). Processed 26 edges, 6 of which were collision checked (23.1 %). The forward search tree has 7 vertices, 1 of which are start states. The reverse search tree has 18 vertices, 5 of which are goal states.
[ WARN] [1681797590.860879716, 657.366000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.867036036, 657.369000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ INFO] [1681797590.871696177, 657.371000000]: left_arm/left_arm[AITstar] (113 iterations): Found a new exact solution of cost 8.1677. Sampled a total of 766 states, 400 of which were valid samples (52.2 %). Processed 34 edges, 9 of which were collision checked (26.5 %). The forward search tree has 8 vertices, 1 of which are start states. The reverse search tree has 21 vertices, 6 of which are goal states.
[ WARN] [1681797590.872067716, 657.371000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ WARN] [1681797590.899726600, 657.404000000]: PathLengthDirectInfSampler: Treating the StateSpace of type "STATE_SPACE_UNKNOWN" as type "STATE_SPACE_REAL_VECTOR".
[ INFO] [1681797590.904040058, 657.408000000]: left_arm/left_arm[AITstar] (209 iterations): Found a new exact solution of cost 8.5157. Sampled a total of 989 states, 500 of which were valid samples (50.6 %). Processed 65 edges, 12 of which were collision checked (18.5 %). The forward search tree has 10 verti [... cut ...] rt states. The reverse search tree has 53 vertices, 10 of which are goal states.
[ INFO] [1681797591.115258254, 657.602000000]: left_arm/left_arm[AITstar] (474 iterations): Found a new exact solution of cost 6.8636. Sampled a total of 2264 states, 1200 of which were valid samples (53.0 %). Processed 102 edges, 16 of which were collision checked (15.7 %). The forward search tree has 10 vertices, 1 of which are start states. The reverse search tree has 32 vertices, 10 of which are goal states.
```

yurirocha15, 2023-05-04:
AIT* is an optimizing planner, which means it will continue to try to improve the current solution until the termination condition is met. In your case, the termination condition is the `timedPlannerTerminationCondition` which means it will run until the time is up. You can try other termination conditions such as `exactSolnPlannerTerminationCondition` or `CostConvergenceTerminationCondition`.

dbdxnuliba, 2023-05-05:
ok ,I see,thanks

closed 2023-05-05 by dbdxnuliba