moveit/moveit #3408: is it a bug in moveit_servo pkg about flip its direction

dbdxnuliba, 2023-04-23:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?1.11.0

in servo_calcs.cpp of moveit_servo pkg,

the raw code is:
```
  if (ini_condition >= new_condition)
  {
    vector_toward_singularity *= -1;
  }
```
should it above code should  be modified to 
  if (ini_condition <= new_condition)
  {
    vector_toward_singularity *= -1;
  }

rhaschke, 2023-05-02:
Unfortunately, the comment is wrong here, but the code is correct: The condition number grows towards a singularity.
https://github.com/ros-planning/moveit/blob/51cff827c92709f3d0b2b6142e42c678dfbbd88a/moveit_ros/moveit_servo/src/servo_calcs.cpp#L723-L729

closed 2023-05-03 by rhaschke

dbdxnuliba, 2023-05-04:
ok  I see .thanks for your prompt reply

PR #3416 (2023-05-03): Fix misleading comment
Fixes #3408. The condition number of the Jacobian _grows_ towards a singularity.