moveit/moveit #3410: Add chomp/OptimizerAdapter causing error

jsbyysheng, 2023-04-23:
### Description

When I followed the [tutorial](https://ros-planning.github.io/moveit_tutorials/doc/planning_adapters/planning_adapters_tutorial.html) to add `chomp/OptimizerAdapter`, I get this error info.
```
[ERROR] [1682247764.891359252]: Exception while loading planning adapter plugin 'chomp/OptimizerAdapter': According to the loaded plugin descriptions the class chomp/OptimizerAdapter with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddRuckigTrajectorySmoothing default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds default_planner_request_adapters/LimitMaxCartesianLinkSpeed default_planner_request_adapters/ResolveConstraintFrames
```

### Your environment
* ROS Distro: [Noetic]
* OS Version: Ubuntu 20.04
* Latest Binary build

rhaschke, 2023-04-23:
Have you installed the package `ros-noetic-moveit-planners-chomp`?

jsbyysheng, 2023-04-23:
Yeap. Using chomp alone works well.

rhaschke, 2023-04-23:
Sorry. I meant the package `moveit_chomp_optimizer_adapter`, i.e. `ros-noetic-moveit-chomp-optimizer-adapter`.

jsbyysheng, 2023-04-23:
It works! Thank you very much.

closed 2023-04-23 by jsbyysheng

rhaschke, 2023-04-23:
I added the installation instruction to the tutorial: https://github.com/ros-planning/moveit_tutorials/commit/d0f827824736ceea6a60d68a20c650ffbe43944f