moveit/moveit #3411: When the object point cloud moves away, the octomap of the original occupied part should be updated to the non-occupied part, but it is very slow to update the non-occupancy part.

dbdxnuliba, 2023-04-24:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: |Noetic]
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? source
* If binary, which release version?
* If source, which branch?1.1.11

When the object point cloud appears, the map of the original non-occupied part is updated to the map of the occupied part very quickly, but when the object point cloud moves away, the map of the original occupied part should be updated to the non-occupied part, but we can see It is very slow to update the non-occupancy part. Is this a bug?
and the point_cloud_topic frequency is 6hz which is published by gazebo_ros ,
and here is the configure file realted 
<launch>
  <rosparam command="load" file="$(find robot_moveit_config)/config/sensors_d435_pointcloud_sim.yaml" />

  <param name="octomap_frame" type="string" value="base_footprint" />
  <param name="octomap_resolution" type="double" value="0.05" />
  <param name="max_range" type="double" value="1.5" />
</launch>

sensors:
  - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    point_cloud_topic: /head_camera_d435/down_sampled_cloud
    max_range: 1.5
    point_subsample: 1
    padding_offset: 0.1
    padding_scale: 1.0
    max_update_rate: 5.0
    filtered_cloud_topic: filtered_cloud

**Question :
1)Why is there such a big difference in speed between add pointcloud to octomap and remove it from the octomap
2)how can we speed up the process about when the object point cloud moves away, the map of the original occupied part update  to the non-occupied part**

rhaschke, 2023-04-28:
To facilitate debugging, it would be helpful if you would provide your full config and steps to reproduce the issue.
Given the configured `max_update_rate` of 5 Hz, I would expect faster reaction too.
To debug this issue, I suggest starting at the receive-message callback of the octomap updater:
https://github.com/ros-planning/moveit/blob/51cff827c92709f3d0b2b6142e42c678dfbbd88a/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp#L181

dbdxnuliba, 2023-04-28:
yes i debug from  here indeed ,and  by debug ,I found a
After the people in the gazebo environment are removed, the poincloud callback was called  be called immediately,
and here  is the  code about moveit to call octomap api to update free cells part
https://github.com/ros-planning/moveit/blob/51cff827c92709f3d0b2b6142e42c678dfbbd88a/moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp#L351)
It is about to update the free cell to the tree, and from the moveit and octomap source code  log I add, it does call the prune function in octomap code  to prune the tree, https://github.com/OctoMap/octomap/blob/92a9f14cafa5f567c47b58cbe91137c9a6ba1b0c/octomap/include/octomap/OcTreeBaseImpl.hxx#LL267C1-L268C1
but after pruning, the corresponding position of the tree in rviz is not immediately updated to the free state, but about waited for 2 seconds after that, it will be updated to free status in rviz 

https://github.com/ros-planning/moveit/issues/3411#:~:text=To%20debug%20this%20issue%2C%20I%20suggest%20starting%20at%20the%20receive%2Dmessage%20callback%20of%20the%20octomap%20updater%3A

rhaschke, 2023-04-28:
I don't have the time to prepare a configuration to reproduce your setup. Please provide a full moveit_config (ideally using the Panda robot used in the tutorials) and the corresponding launch file to start.

When debugging, can you please check whether the octomap was actually updated? Maybe, there is some time-constant before actually freeing a cell? I'm not very familiar with the octomap API, so I cannot help easily.

dbdxnuliba, 2023-04-28:
ok ,I find a panda gazebo realsense sim environment in ,https://github.com/SnehalDikhale/panda_simulation
and it has the same problem above 
here is the octomap update gif
[image]

the configure file about sensors_kinect_pointcloud.yaml is here:
sensors:
  - sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater
    #point_cloud_topic: /camera/depth_registered/points
    point_cloud_topic: /r200/camera/depth_registered/points
    max_range: 5.0
    point_subsample: 1
    padding_offset: 0.1
    padding_scale: 1.0
    max_update_rate: 5.0
    filtered_cloud_topic: filtered_cloud

the sensor_manager.launch.xml is here
<launch>

  

  
  <rosparam command="load" file="$(find panda_moveit_config)/config/sensors_kinect_pointcloud.yaml" />

  
  
  <param name="octomap_frame" type="string" value="camera_rgb_optical_frame" />
  <param name="octomap_resolution" type="double" value="0.05" />
  <param name="max_range" type="double" value="5.0" />

  
  <arg name="moveit_sensor_manager" default="panda" />
  <include file="$(find panda_moveit_config)/launch/$(arg moveit_sensor_manager)_moveit_sensor_manager.launch.xml" />

</launch>

amalnanavati, 2023-11-03:
I'm seeing this behavior as well in `moveit2` in ROS2 Humble with our Kinova JACO2 robot arm using the `DepthImageOctomapUpdater`. Has anyone found a fix yet?

siyuw-rf, 2024-07-05:
still no one fix? encountered the same problem in ROS2 humble