moveit/moveit #3414: srdf_model.getJointProperties(joint_name) not found in ros-noetic-srdfdom (0.6.3-1focal.20230215.215006)

lorenzosciuto, 2023-04-28:
### Description

srdf_model.getJointProperties(joint_name) not found in ros-noetic-srdfdom (0.6.3-1focal.20230215.215006) installed version.
Installing srdfdom from git sources fixes the problem.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? catkin_make from source
* If binary, which release version? *
* If source, which branch? master

### Steps to reproduce
catkin make the source

### Expected behaviour
Compile the project

### Actual behaviour
Broken with the following error on getJointProperties

### Backtrace or Console output

```
[ 13%] Built target moveit_version
/catkin_ws/src/moveit/moveit_core/robot_model/src/robot_model.cpp: In member function ‘moveit::core::JointModel* moveit::core::RobotModel::constructJointModel(const urdf::Joint*, const urdf::Link*, const srdf::Model&)’:
/catkin_ws/src/moveit/moveit_core/robot_model/src/robot_model.cpp:999:71: error: ‘const class srdf::Model’ has no member named ‘getJointProperties’
  999 |     for (const auto& [property_name, property_value_str] : srdf_model.getJointProperties(joint_name))
      |                                                                       ^~~~~~~~~~~~~~~~~~
make[2]: *** [moveit/moveit_core/robot_model/CMakeFiles/moveit_robot_model.dir/build.make:202: moveit/moveit_core/robot_model/CMakeFiles/moveit_robot_model.dir/src/robot_model.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7177: moveit/moveit_core/robot_model/CMakeFiles/moveit_robot_model.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

rhaschke, 2023-04-28:
When you build MoveIt from source, you also need to use latest srdfdom sources. Both packages were not released for a while.

closed 2023-04-28 by rhaschke

lorenzosciuto, 2023-04-28:
We were building MoveIt from source without installing from source also srdfdom util few days ago, but this commit made 2 weeks ago broke the compilation:

https://github.com/ros-planning/moveit/pull/3359

`for (const auto& [property_name, property_value_str] : srdf_model.getJointProperties(joint_name))`

Line 999 of `moveit_core/robot_model/src/robot_model.cpp`.

Are we sure there in no way to fix it so that it is not required to build other packages from sources?

rhaschke, 2023-04-28:
MoveIt is tightly connected to srdfdom (and a couple of other packages maintained by the MoveIt team). There is no way around building srdfdom too, when you want to build MoveIt master.