moveit/moveit #3428: Setting the planner algorithm from script

akmandor, 2023-05-11:
### Description

Instead of using the Rviz's MotionPlanning interface, I tried to set the motion planning algorithm to generate trajectories from a script. However, I couldn't manage that by using the MotionPlanRequest object as suggested in [Motion Planning Pipeline](https://ros-planning.github.io/moveit_tutorials/doc/motion_planning_pipeline/motion_planning_pipeline_tutorial.html). I would really appreciate if you could guide me on this issue.

It would be also nice, if one of these MoveIt Tutorials includes an example to set the planner.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Both
* Binary release version: MoveIt 1.0 (latest)
* Source branch: origin/master

### Steps to reproduce

1. Launch Gazebo simulation with the robot's model and controllers
`roslaunch panda_moveit_config demo_gazebo.launch`

2. Add `<param name="planning_plugin" value="ompl_interface/OMPLPlanner" />` in the _move_group_ node (Please check the second item in **Expected - Actual Behaviours and Backtrace or Console Outputs** section below for more details) in the `move.group.launch` and launch it. I tested to plan trajectories with different available algorithms using the RViz MotionPlanning interface and confirmed that it works as expected.
`roslaunch panda_moveit_config move_group.launch`

3. I followed the [Motion Planning Pipeline](https://ros-planning.github.io/moveit_tutorials/doc/motion_planning_pipeline/motion_planning_pipeline_tutorial.html) tutorial to write a simple script (coding contents given below) that generates a trajectory with a desired planning algorithm (e.g. RRTstar).

Here is the full content of the script:
```
#include <math.h>

#include <ros/ros.h>
#include <tf2/LinearMath/Quaternion.h>
#include <tf2_ros/transform_listener.h>
#include <tf2_ros/buffer.h>
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
#include <visualization_msgs/Marker.h>

// MoveIt
/*
#include <moveit/robot_model_loader/robot_model_loader.h>
#include <moveit/robot_state/conversions.h>
#include <moveit/planning_pipeline/planning_pipeline.h>
#include <moveit/planning_interface/planning_interface.h>
#include <moveit/planning_scene_monitor/planning_scene_monitor.h>
#include <moveit/kinematic_constraints/utils.h>
#include <moveit_msgs/DisplayTrajectory.h>
#include <moveit_msgs/PlanningScene.h>
#include <moveit_visual_tools/moveit_visual_tools.h>
*/

#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_pipeline/planning_pipeline.h>
#include <moveit/planning_interface/planning_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit/kinematic_constraints/utils.h>
#include <moveit_msgs/DisplayRobotState.h>
#include <moveit_msgs/DisplayTrajectory.h>
#include <moveit_msgs/AttachedCollisionObject.h>
#include <moveit_msgs/CollisionObject.h>
#include <moveit_visual_tools/moveit_visual_tools.h>

int main(int argc, char **argv)
{
  std::cout << "[test_moveit_planner::main] START" << std::endl;

  ros::init(argc, argv, "test_moveit_planner");
  ros::NodeHandle node_handle("~");
  
  //ros::AsyncSpinner spinner(4);
  //spinner.start();

  // Robot Model
  robot_model_loader::RobotModelLoaderPtr robotModelLoaderPtr(new robot_model_loader::RobotModelLoader("robot_description"));
[... 159 lines cut ...]
[ INFO] [1683824761.325217112, 22.320000000]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[ INFO] [1683824761.330323499, 22.325000000]: Listening to '/collision_object'
[ INFO] [1683824761.343230365, 22.337000000]: Listening to '/planning_scene_world' for planning scene world geometry
[ WARN] [1683824761.344275113, 22.337000000]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[ WARN] [1683824761.344433803, 22.337000000]: Target frame "panda_link0" specified but no TF instance (buffer) specified. No transforms will be applied to received data.
[ INFO] [1683824761.345312191, 22.339000000]: No 3D sensor plugin(s) defined for octomap updates
[ INFO] [1683824761.389782864, 22.383000000]: Listening to '/attached_collision_object' for attached collision objects
[ INFO] [1683824761.398094537, 22.392000000]: Multiple planning plugins available. You should specify the '~planning_plugin' parameter. Using 'chomp_interface/CHOMPPlanner' for now.
[ INFO] [1683824761.442795205, 22.436000000]: Using planning interface 'CHOMP'
[MoveitInterface::moveitPlanTrajectory] BEFORE generatePlan
[ERROR] [1683824761.744919786, 22.733000000]: Found empty JointState message
[ERROR] [1683824761.745039738, 22.733000000]: Only joint-space goals are supported
[MoveitInterface::moveitPlanTrajectory] AFTER generatePlan
[ERROR] [1683824761.745089534, 22.733000000]: [MoveitInterface::moveitPlan] Could not compute plan successfully
[ INFO] [1683824761.788271622, 22.776000000]: Stopping world geometry monitor
[ INFO] [1683824761.788669618, 22.776000000]: Stopping planning scene monitor
[test_moveit_planner-1] process has finished cleanly
log file: /home/akmandor/.ros/log/09afffda-f01e-11ed-8b1d-010f83376c04/test_moveit_planner-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Since the default _pipeline_ parameter in `move_group.launch` is set to _ompl_. I was expecting the default planner is selected among ompl algorithms. More suprisingly, even I delete the `ompl_planning.yaml` file in _panda_moveit_config/config_ folder, the planner still uses RRTConnect as the deafult planner and is able to plan the trajectory. However, in that case, user cannot plan using the Rviz's interface as expected.

rhaschke, 2023-05-12:
You can easily inspect the values of the `MotionPlanRequest` via: `rostopic echo /move_group/goal`. rviz uses the following values:
```
pipeline_id: "ompl"
planner_id: "RRTstar"
```
But it works the same way with an empty `pipeline_id`. **Thus, I cannot reproduce your issue.**

I'm not sure why (and where) you manually set the `planning_plugin` parameter (step 2).
The latest release of `panda_moveit_config` comes with a correctly pre-configured multi-pipeline configuration for ompl, chomp, and Pilz:
```
> rosparam list| grep planning_plugin
/move_group/planning_pipelines/chomp/planning_plugin
/move_group/planning_pipelines/ompl/planning_plugin
/move_group/planning_pipelines/pilz_industrial_motion_planner/planning_plugin
```
The OMPL planners are configured as follows:
```
> rosparam get /move_group/planning_pipelines/ompl/panda_arm
longest_valid_segment_fraction: 0.005
planner_configs:
- AnytimePathShortening
...
- RRT
- RRTConnect
- RRTstar
...
projection_evaluator: joints(panda_joint1,panda_joint2)
```
Maybe, you screwed your configuration?

akmandor, 2023-05-13:
First of all, **I haven't changed any configuration at all**, that's why Rviz is working as expected.

My issue is when you call the MotionPlanRequest **from script** by specifying parameters, such as group_name, planner_id, etc., the planner always generates using the **RRTConnect** given the following output:
```
[ WARN] [1683938336.589491664, 100.410000000]: Cannot find planning configuration for group 'panda_arm' using planner 'RRTstar'. Will use defaults instead.
[ERROR] [1683938336.590590295, 100.410000000]: Found empty JointState message
[ WARN] [1683938336.590740742, 100.410000000]: It looks like the planning volume was not specified.
[ INFO] [1683938336.593764558, 100.412000000]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
```

Here is what I exactly did to test the issue:
- I created a new workspace and include necessary packages: [franka_ros](https://github.com/frankaemika/franka_ros) and [panda_moveit_config](panda_moveit_config).
- I both tried binary and source (installing in that workspace) versions of MoveIt.
- I just write the script, which simply creates the plan using MotionPlanRequest, and run it.
- I verified this issue, when following above steps, with couple of people as well. They reproduced the same result as presented above!

In order to replicate the issue after installing necessary packages:
1. `roslaunch panda_moveit_config demo_gazebo.launch`
2. `rosrun <your_pkg_name_where_script_is_located> test_moveit_planner`

Here is the script, **test_moveit_planner**:
```
#include <math.h>
#include <ros/ros.h>
#include <tf2/LinearMath/Quaternion.h>
#include <tf2_ros/transform_listener.h>
#include <tf2_ros/buffer.h>
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
#include <visualization_msgs/Marker.h>
#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_pipeline/planning_pipeline.h>
#include <moveit/planning_interface/planning_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit/kinematic_constraints/utils.h>
#include <moveit_msgs/DisplayRobotState.h>
#include <moveit_msgs/DisplayTrajectory.h>
#include <moveit_msgs/AttachedCollisionObject.h>
#include <moveit_msgs/CollisionObject.h>
#include <moveit_visual_tools/moveit_visual_tools.h>

int main(int argc, char **argv)
{
  std::cout << "[test_moveit_planner::main] START" << std::endl;

  ros::init(argc, argv, "test_moveit_planner");
  ros::NodeHandle node_handle("");
  
  // Set the planner
  std::string planner_plugin_name = "ompl_interface/OMPLPlanner";
  node_handle.setParam("planning_plugin", planner_plugin_name);

  // Robot Model
  robot_model_loader::RobotModelLoaderPtr robotModelLoaderPtr(new robot_model_loader::RobotModelLoader("robot_description"));

  // Robot Model
  moveit::core::RobotModelPtr robotModelPtr = robotModelLoaderPtr->getModel();

  // Robot State
  moveit::core::RobotStatePtr robotStatePtr(new moveit::core::RobotState(robotModelPtr));

  // Planning Scene
  planning_scene_monitor::PlanningSceneMonitorPtr psmPtr(new planning_scene_monitor::PlanningSceneMonitor(robotModelLoaderPtr));

  // Listen for planning scene messages on topic /XXX and apply them to the internal planning scene accordingly
  psmPtr->startSceneMonitor();

  // Listens to changes of world geometry, collision objects, and (optionally) octomaps
  psmPtr->startWorldGeometryMonitor();
  
  // Listen to joint state updates as well as changes in attached collision objects and update the internal planning scene accordingly
[... 59 lines cut ...]
  display_trajectory.trajectory.clear();
  display_trajectory.trajectory.push_back(response.trajectory);

  // Publish Trajectory
  ros::Publisher display_pub = node_handle.advertise<moveit_msgs::DisplayTrajectory>("/test_moveit_planner/display_planned_path", 10, true);
  const moveit::core::JointModelGroup* jointModelGroupPtr = robotModelPtr->getJointModelGroup(group_name);

  display_pub.publish(display_trajectory);
  visual_tools.publishTrajectoryLine(display_trajectory.trajectory.back(), jointModelGroupPtr);

  /*
  while(ros::ok())
  {
    ros::spinOnce();
  }
  */

  std::cout << "[test_moveit_planner::main] END" << std::endl;

  ros::waitForShutdown();
  return 0;
}
```

Note: I slightly updated the code I provided when I opened the issue, so that you don't need to set the _planning_plugin_ as I explained in the second stage in **Steps to reproduce** section. **But you must set the _planning_plugin_**. If you don't, the generated plan in the script is using CHOMP as I explained previously!

rhaschke, 2023-05-13:
The need to set the `planning_plugin` parameters is suspicious. It turns out that you use the wrong namespace to search for the planning pipeline parameters. Use the following line to instantiate the pipeline:

<code>planning_pipeline::PlanningPipelinePtr planningPipelinePtr(new planning_pipeline::PlanningPipeline(
      robotModelPtr, ros::NodeHandle(<b>"/move_group/planning_pipelines/ompl"</b>), 
      "planning_plugin", "request_adapters"));
</code>

Instead of instantiating all required components (PSM, pipeline, etc) manually, I suggest using [MoveItCpp](https://ros-planning.github.io/moveit_tutorials/doc/moveit_cpp/moveitcpp_tutorial.html). This should also get rid of the warnings shown in your console.

Please file such usage questions at https://answers.ros.org in future. Here, only real issues and bug reports should be posted.

closed 2023-05-13 by rhaschke