moveit/moveit #3430: Libraries broken after update.

bohanfeng, 2023-05-17:
### Description
Libraries broken after update.
After updating the package today, the libraries appear to be broken:

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* sudo apt-get install ros-noetic-moveit

### Steps to reproduce
catkin build -DPYTHON_EXECUTABLE=/usr/bin/python3

### Actual behaviour
/usr/bin/ld: warning: libmoveit_robot_model.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_transforms.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_state.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_kinematic_constraints.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_move_group_interface.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_ros_occupancy_map_monitor.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_robot_trajectory.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_collision_detection.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_planning_scene.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_profiler.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_utils.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_planning_scene_monitor.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libmoveit_kinematics_base.so.1.1.11, needed by /opt/ros/noetic/lib/libmoveit_kinematics_plugin_loader.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libsrdfdom.so.0.6.3, needed by /opt/ros/noetic/lib/libmoveit_rdf_loader.so, not found (try using -rpath or -rpath-link)

The existed files are 
libmoveit_utils.so.1.1.12 etc.

bohanfeng, 2023-05-17:
I can temperally fix this probmle by using ln -s xxxx.1.1.12 xxxxx.1.1.11 .

rhaschke, 2023-05-17:
Better revert to the 1.1.11 versions.

It looks like that building the binary packages for MoveIt wasn't yet completed, but some new packages were already shipped. Particularly, the binary build for [`moveit_ros_move_group` failed](https://build.ros.org/view/Nbin_uF64/job/Nbin_uF64__moveit_ros_move_group__ubuntu_focal_amd64__binary) (due to unavailability of an ubuntu package server), leaving many downstream packages dangling. 
@clalancette or @sloretz could you please trigger that job manually again?

I was expecting that individual binary packages of a repo (here MoveIt) would only be shipped to the testing repo if all packages of that repo have been built successfully. That doesn't seem to be the case (or is broken).

jspricke, 2023-05-17:
> What is the convenient way to revert to the 1.1.11 versions?

```
sudo sed -i 's#ros/#ros-testing/#' /etc/apt/sources.list.d/ros-latest.list
sudo apt install "?and(~i,~nros-noetic-*)/focal"
```

> I was expecting that individual binary packages of a repo (here MoveIt) would only be shipped to the testing repo if all packages of that repo have been built successfully. That doesn't seem to be the case (or is broken).

I don't think the ROS build farm knows about a git repo containing packages. All it gets is the bloom release of individual ROS packages.

gavanderhoorn, 2023-05-17:
Everything build is instantly pushed to `ros-testing`. There's no gating. On purpose. That's the point of the testing repositories.

v4hn, 2023-05-17:
> Everything built is instantly pushed to ros-testing

There is `building` too and there are regular syncs from there to `testing` to group things together instead of pushing each build directly provoking inconsistent states throughout the day. I'm also not surprised to see that there is no such feature, but it would be useful in such events. :-)
I *am* surprised though that a single broken tcp connection can still break consistency in `testing` after all these years.

gavanderhoorn, 2023-05-17:
I should probably have added some words about my comment being based on my understanding from a couple years ago. Could very well be @cottsay and @nuclearsandwich have improved things since that time.

clalancette, 2023-05-17:
> I was expecting that individual binary packages of a repo (here MoveIt) would only be shipped to the testing repo if all packages of that repo have been built successfully. That doesn't seem to be the case (or is broken).

No, that is not how it works.  Each package is given its own job on the buildfarm, and doesn't really know anything about the repository it originally came from.

That said, this is all self-healing.  If a package fails to build for any reason whatsoever, it will be tried the next night (and every night, actually) until it succeeds.  So this should resolve itself on its own.

rhaschke, 2023-05-18:
[All MoveIt packages were correctly build now.](http://repositories.ros.org/status_page/ros_noetic_default.html?q=%5Emoveit_) Closing.

closed 2023-05-18 by rhaschke