moveit/moveit #3434: MoveIt servo doesnt move the robot

maxdgaa, 2023-05-18:
### Description
Hi!
I am in the process of getting moveit servo to work with my simulated Fanuc robotic arm. I've copied the servo_cpp_interface_demo.launch.py and accompanying file. My default joint positions are correctly being set to 0 and a mock_components/GenericSystem controller is being used. 

I know the following things:
moveit_servo is receiving the joint commands.
moveit_servo is outputting a trajectory_msgs/msg/JointTrajectory. 

The problem is: it just does not move. The trajectory_msgs/msg/JointTrajectory that is being published is all-zeros for the joints so that is not surprising. However, I'm also not getting any errors. When i echo the status topic, i get 0 as a response. MoveIt servo thinks everything is fine, but nothing is happening.

I will post all files that are in use:
Config:
`###############################################
# Modify all parameters related to servoing here
###############################################
use_gazebo: false # Whether the robot is started in a Gazebo simulation environment

## Properties of incoming commands
command_in_type: "unitless" # "unitless"> in the range [-1:1], as if from joystick. "speed_units"> cmds are in m/s and rad/s
scale:
  # Scale parameters are only used if command_in_type=="unitless"
  linear:  0.4  # Max linear velocity. Unit is [m/s]. Only used for Cartesian commands.
  rotational:  0.8 # Max angular velocity. Unit is [rad/s]. Only used for Cartesian commands.
  # Max joint angular/linear velocity. Only used for joint commands on joint_command_in_topic.
  joint: 0.5

# Optionally override Servo's internal velocity scaling when near singularity or collision (0.0 = use internal velocity scaling)
# override_velocity_scaling_factor = 0.0 # valid range [0.0:1.0]

## Properties of outgoing commands
publish_period: 0.034  # 1/Nominal publish rate [seconds]
low_latency_mode: false  # Set this to true to publish as soon as an incoming Twist command is received (publish_period is ignored)

# What type of topic does your robot driver expect?
# Currently supported are std_msgs/Float64MultiArray or trajectory_msgs/JointTrajectory
command_out_type: trajectory_msgs/JointTrajectory

# What to publish? Can save some bandwidth as most robots only require positions or velocities
publish_joint_positions: true
publish_joint_velocities: false
publish_joint_accelerations: false

## Plugins for smoothing outgoing commands
smoothing_filter_plugin_name: "online_signal_smoothing::ButterworthFilterPlugin"

# If is_primary_planning_scene_monitor is set to true, the Servo server's PlanningScene advertises the /get_planning_scene service,
# which other nodes can use as a source for information about the planning environment.
# NOTE: If a different node in your system is responsible for the "primary" planning scene instance (e.g. the MoveGroup node),
# then is_primary_planning_scene_monitor needs to be set to false.
is_primary_planning_scene_monitor: true

## MoveIt properties
move_group_name:  manipulator  # Often 'manipulator' or 'arm'
planning_frame: base_link  # The MoveIt planning frame. Often 'base_link' or 'world'

## Other frames
ee_frame_name: tool0  # The name of the end effector link, used to return the EE pose
robot_link_command_frame:  base_link  # commands must be given in the frame of a robot link. Usually either the base or end effector

## Stopping behaviour
incoming_command_timeout:  0.10000 # Stop servoing if X seconds elapse without a new command
# If 0, republish commands forever even if the robot is stationary. Otherwise, specify num. to publish.
# Important because ROS may drop some messages and we need the robot to halt reliably.
num_outgoing_halt_msgs_to_publish: 4

## Configure handling of singularities and joint limits
lower_singularity_threshold:  17.0  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 30.0 # Stop when the condition number hits this
joint_limit_margin: 0.1 # added as a buffer to joint limits [radians]. If moving quickly, make this larger.
leaving_singularity_threshold_multiplier: 2.0 # Multiply the hard stop limit by this when leaving singularity (see https://github.com/ros-planning/moveit2/pull/620)
[... 308 lines cut ...]

    return LaunchDescription(
        [rviz_node,
         static_tf,
         servo_node,
         control_node,
         joint_state_broadcaster_spawner,
         robot_state_publisher,
         joint_trajectory_controller_spawner]
    )

`
### Your environment
* ROS Distro: Humble 
* Ubuntu 22.04
* 

### Expected behaviour
The robotic arm should be moving. 

### Actual behaviour
It stays still. A JointTrajectory is being outputted, but its all-zeros.
### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

maxdgaa, 2023-05-18:
Hi, 
Thanks for the response. I see I posted it in the Moveit 1 repo by accident, sorry about that. 
1. I am starting servo with the 
```    servo->start();``` 
call. Its not running as a seperate node so the service call does not work. 
2. Yes, 
```ros2 topic info -v /fanuc_controller/joint_trajectory```
outputs:
```
Type: trajectory_msgs/msg/JointTrajectory

Publisher count: 1

Node name: servo_demo_node
Node namespace: /
Topic type: trajectory_msgs/msg/JointTrajectory
Endpoint type: PUBLISHER
GID: 01.10.14.44.5d.f3.5a.8b.70.ca.49.96.00.00.3a.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: fanuc_controller
Node namespace: /
Topic type: trajectory_msgs/msg/JointTrajectory
Endpoint type: SUBSCRIPTION
GID: 01.10.3b.ae.5f.c6.8a.81.44.27.86.e3.00.00.4c.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```
3. Yes
4. No, it does not move. 
5. Yes

It seems like the robot is unable to be moved. These are my files: 

fanuc.ros2_control.xacro:
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:macro name="fanuc_ros2_control" params="name initial_positions_file">
        <xacro:property name="initial_positions" value="${xacro.load_yaml(initial_positions_file)['initial_positions']}"/>
        <ros2_control name="${name}" type="system">
            <hardware>
<!--                Be very careful not to leave a space in the plugin, the pluginlib wont be able to find it.-->
<!--                <plugin>fanuc_m10ia_low_level_driver/FanucArmHardwareInterface</plugin>-->
                <plugin>mock_components/GenericSystem</plugin>
                <param name="example_param_hw_start_duration_sec">0</param>
                <param name="example_param_hw_stop_duration_sec">3.0</param>
                <param name="example_param_hw_slowdown">100</param>
            </hardware>
            <joint name="joint_1">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['joint_1']}</param>
                </state_interface>
            </joint>
            <joint name="joint_2">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['joint_2']}</param>
                </state_interface>
            </joint>
            <joint name="joint_3">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['joint_3']}</param>
                </state_interface>
            </joint>
            <joint name="joint_4">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['joint_4']}</param>
                </state_interface>
            </joint>
            <joint name="joint_5">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['joint_5']}</param>
                </state_interface>
            </joint>
            <joint name="joint_6">
                <command_interface name="position"/>
                <state_interface name="position">
                  <param name="initial_value">${initial_positions['joint_6']}</param>
                </state_interface>
            </joint>
        </ros2_control>
    </xacro:macro>
</robot>
```

ros2_controllers.yaml:
```

AndyZe, 2023-05-18:
Ok, this sounds like a ros2_control issue then. You could try `ros2 control list_controllers` to see if the JTC (joint trajectory controller) is running

More commands here:  https://control.ros.org/master/doc/ros2_control/ros2controlcli/doc/userdoc.html

maxdgaa, 2023-05-18:
Hi, 
If I run my launch file in which MoveGroup is also launched, I am able to command the arm using the RVIZ interface as well as with rqt_joint_trajectory_controller. 

Then, the second I run moveit servo, the controllers seem to stop working but at all times the JTC seems to be running fine:
``` 
fanuc_controller    [joint_trajectory_controller/JointTrajectoryController] active    
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active   
```

AndyZe, 2023-05-18:
Hmm, i don't understand this...

maxdgaa, 2023-05-18:
The weird thing is that with the Panda robot the servo demo works great which leads me to believe it is a configuration issue but I tested it: if I configure servo to work with invalid links or joint names it throws errors as expected, and it doesnt do that so the config seems fine. Furthermore, rqt_graph shows exactly what you'd expect.

maxdgaa, 2023-05-19:
Update: even with the following params:
```lower_singularity_threshold:  99.9  # Start decelerating when the condition number hits this (close to singularity)
hard_stop_singularity_threshold: 100.0 # Stop when the condition number hits this
```
I still get the error message
` Very close to a singularity, emergency stop`

maxdgaa, 2023-05-19:
Hi guys, I managed to fix it. 
For anyone that might encounter this unfortunate issue as well: it has to do with RVIZ.

I was using an RVIZ file that I also use while testing, so the MotionPlanning plugin was being loaded. I have a very strong feeling that once this plugin was loaded, it "claimed" all planning and arm control and just held the robot in place. What caused the 
> Very close to a singularity, emergency stop
error message is still unknown to me. Thanks @AndyZe for the help!

closed 2023-05-19 by maxdgaa

rhaschke, 2023-05-20:
I don't think that the MotionPlanning plugin is interfering with servo control.

maxdgaa, 2023-05-20:
Hi @rhaschke, 
After some more research I think you're right. I'm fairly certain the planning scenes were mismatched and that was causing the issue.

mohakmalviya, 2023-10-30:
hello I am also facing the same issue can you please help me in this

suyog800, 2024-03-14:
Hello @rhaschke , 
I don't think this issue is caused by MotionPlanning plugin of Moveit, as I am not spawning this plugin, but still facing the above issues. when I change controller to forward position controller (/forward_position_controller/commands) (data type:std_msgs/Float64MultiArray), the robot moves in gazebo, ie servo node  works fine with this controller. But the catch over here is, I can either publish position or velocity. But my actual arm controller needs both the values, therefore I need to use joint_trajectory_controller (data type : trajectory_msgs/JointTrajectory ). Any guess, why I am facing this Issue.
note - I am using right controllers and controllers are spawned properly

raquelbd-work, 2025-02-04:
Hi @suyog800, I am having the same issue. Were you able to figure it out?