moveit/moveit #3436: Time derivative of a Jacobian

ivatavuk, 2023-05-21:
Hello, 
I was wondering if there is a way of computing the Jacobian time derivative integrated in MoveIt? 
To clarify what I'm talking about, if a task-space velocity is:

$$\begin{equation}
  \boldsymbol{v} = \boldsymbol{J}\dot{\boldsymbol{q}}
\end{equation}$$

Acceleration is then:

$$\begin{equation}
  \boldsymbol{a} = \dot{\boldsymbol{J}}\dot{\boldsymbol{q}} + \boldsymbol{J}\ddot{\boldsymbol{q}}
\end{equation}$$

where $\dot{\boldsymbol{J}}$ is the Jacobian time derivative.

I would like to discuss the possibility of adding such functionality under RobotState, where Jacobian computation is located. I see this functionality is available in kdl, [here](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/chainjnttojacdotsolver.hpp).

I think this would be a useful feature to have, and would be glad to work on it and provide a pull request, but would first want to get some feedback from the MoveIt community:
  - Is there already such a feature integrated in MoveIt, and I'm missing it?
  - Is [robot_state](https://github.com/ros-planning/moveit/blob/noetic-devel/moveit_core/robot_state/src/robot_state.cpp) the appropriate place for it, in a form of a function similar to getJacobian? Would this feature be within the intended scope of MoveIt's RobotState?
  
 Thanks in advance!

simonschmeisser, 2023-05-21:
Welcome, what would be typical use cases for the time derivative of the jacobian?

How would you calculate it? Would you need to store it because it needs a trajectory to be calculated? `RobotState` is used in many places and should ideally not grow further in memory consumption

rhaschke, 2023-05-21:
RobotState is the right place to add this feature and your proposal to add a function `getJacobianDerivative` is perfect.
@simonschmeisser, we don't need to store this - we don't store the Jacobian either.
@ivatavuk, MoveIt includes an inverse [dynamics solver](https://github.com/ros-planning/moveit/blob/fa6331ca55f97a4615d4166c14a5e47d9bbfc7d9/moveit_core/dynamics_solver/include/moveit/dynamics_solver/dynamics_solver.h), but this doesn't expose acceleration-specific stuff.

ivatavuk, 2023-05-22:
Thank you for the feedback!

@simonschmeisser
This shouldn't require anything additional to be stored in the `RobotState`, the time derivative of a Jacobian doesn't depend on previous states or a trajectory. 
It's a function of the kinematic model, joint positions and joint velocities.
This would be a  low-level feature and could have various use cases. The most typical one would be acceleration level task-space control, to incorporate acceleration level constraints. It also has some other interesting use cases, I think it's used to construct a manipulability Jacobian, a Jacobian that maps joint velocities to the rate of change in the manipulability index, which can be used for redundancy resolution.

@rhaschke 
Thank you for the link! Looks like the dynamics solver acts as a sort of a interface for `KDL::ChainIdSolver_RNE`. It should be possible to implement a `getJacobianDerivative` function in such a way that it doesn't depend on kdl, I was thinking about using kdl's implementation for inspiration/testing. 

I'll start working on this as soon as possible, and update you on the progress!

kevinporras123, 2023-08-23:
Hi there, I would love to lend a hand on this if needed. I'm new to open source communities but I'm a fast learner and have an engineer/science background.