moveit/moveit #3442: how to configure moveit to generate a coarse resolution octomap and a fine resolution octomap simultaneously in moveit

dbdxnuliba, 2023-06-01:
How to configure MoveIt to generate octomap of different resolutions from two separate point cloud topics published by two depth cameras at the same time?
For example, generating a coarse resolution octomap and a fine resolution octomap simultaneously in moveit with two different pointcloud topics

simonschmeisser, 2023-06-01:
There is currently only support for one octomap, it's even shared between copies of PlanningScene.

What would be your use case? How should they be handled differently?

dbdxnuliba, 2023-06-10:
In the scenario, I first move to the vicinity of the target location using a low-resolution map (since the map is large, a coarse resolution is needed), and then use a high-resolution map (which improves collision detection accuracy and has a small map size, saving memory) for precise movement to the target position.

myjhvytdrt, 2024-07-05:
Have you solved the problem？