moveit/moveit #3446: Joint 'erver' not found in model 'ur5_robot' error in moveit when using compute_cartesian_path #640

devvaibhav455, 2023-06-12:
Hi,

System details: Ubuntu 20.04 on amd64 architecture with ROS noetic

I am trying to simulate a UR5 robot in gazebo and want the EE to follow a cartesian goal in real-time. I subscribe to a topic which continuously publishes 6-DOF pose and then I use `compute_cartesian_path`. It works most of the times but sometimes I get these errors.

Error # 1:
`SimpleSetup: Path simplification took 0.001818 seconds and changed from 3 to 2 states`
`[ INFO] [1686337081.111904271, 6693.901000000]: Attempting to follow 1 waypoints for link 'tool0' using a step of 0.050000 m and jump threshold 0.000000 (in global reference frame)`
`[ INFO] [1686337081.117172171, 6693.906000000]: Computed Cartesian path with 2 points (followed 100.000000% of requested trajectory)`
`[ INFO] [1686337081.117837855, 6693.907000000]: Execution request received`
`[ERROR] [1686337081.119268921, 6693.908000000]: Joint 'erver' not found in model 'ur5_robot'`
`[move_group-1] process has died [pid 1328127, exit code -11, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group /scaled_pos_joint_traj_controller/follow_joint_trajectory:=/eff_joint_traj_controller/follow_joint_trajectory __name:=move_group __log:=/home/dev/.ros/log/62d75aac-06e6-11ee-9fd9-b5f2b2fb5025/move_group-1.log].`
`log file: /home/dev/.ros/log/62d75aac-06e6-11ee-9fd9-b5f2b2fb5025/move_group-1*.log`

Error # 2:
`Execution request received`
`[ERROR] [1686340497.445277228, 10104.988000000]: Joint '??
                                                          ?' not found in model 'ur5_robot'`
`[move_group-1] process has died [pid 1802270, exit code -11, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group /scaled_pos_joint_traj_controller/follow_joint_trajectory:=/eff_joint_traj_controller/follow_joint_trajectory __name:=move_group __log:=/home/dev/.ros/log/62d75aac-06e6-11ee-9fd9-b5f2b2fb5025/move_group-1.log].`
`log file: /home/dev/.ros/log/62d75aac-06e6-11ee-9fd9-b5f2b2fb5025/move_group-1*.log`

I use the below commands to run the simulation and move it.

In terminal 1:
`roslaunch ur_gazebo ur5_bringup.launch`

in terminal 2:
`roslaunch ur5_moveit_config moveit_planning_execution.launch sim:=true`

Thanks in advance for your help.

rhaschke, 2023-06-12:
This looks like a stack or memory corruption issue, which in turn could be rooted in a concurrency issue. `compute_cartesian_path` was not designed for real-time application. Consider using [MoveIt Servo](https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html) instead.

devvaibhav455, 2023-06-12:
@rhaschke : I am using wait flag as `True` `self.group.execute(plan, wait=True)`. By doing this, can it still lead to this corruption issue?

devvaibhav455, 2023-06-13:
Hmmm... This is the code which I am using right now

```
#!/usr/bin/env python3

#
# Code adapted from https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py
# to UR5 robot
#
from math import pi
import sys
import copy
import rospy
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg
import message_filters
from sensor_msgs.msg import Joy
from math import pi
from std_msgs.msg import String, Header
import visualization_msgs.msg
from visualization_msgs.msg import Marker
from moveit_commander.conversions import pose_to_list
## END_SUB_TUTORIAL

def all_close(goal, actual, tolerance):
  """
  Convenience method for testing if a list of values are within a tolerance of their counterparts in another list
  @param: goal       A list of floats, a Pose or a PoseStamped
  @param: actual     A list of floats, a Pose or a PoseStamped
  @param: tolerance  A float
  @returns: bool
  """
  all_equal = True
  if type(goal) is list:
    for index in range(len(goal)):
      if abs(actual[index] - goal[index]) > tolerance:
        return False

  elif type(goal) is geometry_msgs.msg.PoseStamped:
    return all_close(goal.pose, actual.pose, tolerance)

  elif type(goal) is geometry_msgs.msg.Pose:
    return all_close(pose_to_list(goal), pose_to_list(actual), tolerance)

  return True

def print_log(message):
    header = Header(stamp=rospy.Time.now())
    epoch_time = header.stamp.to_sec()
    sim_time = rospy.get_time()
    log_string = f"{message}"
    rospy.loginfo(log_string)

class MoveGroupTutorial(object):
  """MoveGroupTutorial"""
  def __init__(self):
    super(MoveGroupTutorial, self).__init__()

    # First initialize `moveit_commander`_ and a `rospy`_ node:
    moveit_commander.roscpp_initialize(sys.argv)
    rospy.init_node('move_group_tutorial_ur5', anonymous=True)

    # Function to print logs with epoch and simulation time

# Example usage
    # print_log("This is a log message.")
    

    # Create a subscriber. In this example, we are subscribing to a topic called "chatter"
[... 192 lines cut ...]
    current_joints = self.group.get_current_joint_values()
    return all_close(joint_goal, current_joints, 0.01)
  

def main():
  
  try:
    tutorial = MoveGroupTutorial()
    tutorial.go_to_up_state()
    print("################## ROBOT SHOULD FOLLOW THE TRAJECTORY NOW ##################")
    rospy.spin()
    rospy.loginfo("################## USER DEBUG MESSAGE STARTED")
  

  
  except rospy.ROSInterruptException:
    return
  except KeyboardInterrupt:
    return

if __name__ == '__main__':
  main()
```

It simply subscribes to joystick button input and vive gripper Marker message which contains pose (HTC vive controller which gives pose). I use `message_filters.ApproximateTimeSynchronizer` to use a single callback function for two messages. Then, just plan the trajectory with one waypoint.